realsenseai/realsense-ros #2093: _d415 aluminum material is not loaded on rviz

tnajjar, 2021-09-29:
On branch foxy, the _d415 does not load the aluminum material

[image]

MartyG-RealSense, 2021-09-29:
Hi @tnajjar  Is this model from the models supplied with the RealSense ROS2 wrapper, please?

https://github.com/IntelRealSense/realsense-ros/tree/ros2/realsense2_description/meshes

tnajjar, 2021-09-29:
Hi, sorry I mispoke, I did not clone branch foxy, I have ros-foxy-realsense2-description. To reproduce simply install the package and run ros2 launch realsense2_description view_model.launch.py model:=test_d415_camera.urdf.xacro, you should get the red camera, meaning that the material is not loaded.
To answer your question, I'm not sure if ros-foxy-realsense2-description uses the meshes from the ros2 branch or the foxy branch but you probably know that.

MartyG-RealSense, 2021-09-29:
I will highlight your issue to **Doronhi** the RealSense ROS wrapper developer.  Thanks for the report!

doronhi, 2021-10-03:
Thanks for noticing and reporting. Fixed in #2098.

tnajjar, 2021-10-03:
Was just about to open a PR, thanks a lot!

tnajjar, 2021-10-11:
Oh sorry didn't know you were waiting for an approval. Anyway thanks a lot

closed 2021-10-11 by tnajjar