realsenseai/realsense-ros #2094: cannot launch d435i camera - Symbol lookup error

rythm07, 2021-09-29:
Hello, 

I was trying to launch the rs_d435i launch file but I'm consistently getting this error. I don't know what happened suddenly but they were working before.

`[ERROR] [realsense2_camera_node-1]: process has died [pid 9325, exit code 127, cmd '/home/jetson/dev2_ws/install/realsense2_camera/lib/realsense2_camera/realsense2_camera_node --ros-args --log-level info --ros-args -r __node:=d435i -r __ns:=/d435i --params-file /tmp/launch_params_1gbxh6q8 --params-file /home/jetson/dev2_ws/install/realsense2_camera/share/realsense2_camera/config/d435i.yaml'].`

Here is the screenshot of what appears in the terminal:
[image]

And here is the launch file and yaml file, respectively:

`"""Launch realsense2_camera node."""

import copy
from launch import LaunchDescription
from launch.actions import IncludeLaunchDescription
from launch.substitutions import LaunchConfiguration, ThisLaunchFileDir
from launch.launch_description_sources import PythonLaunchDescriptionSource
import sys
import pathlib
sys.path.append(str(pathlib.Path(__file__).parent.absolute()))
import rs_launch

local_parameters = [{'name': 'camera_name', 'default': 'd435i', 'description': 'camera unique name'},
                    {'name': 'device_type', 'default': 'd435i', 'description': 'choose device by type'},
                    {'name': 'config_file', 'default': "'/home/jetson/dev2_ws/install/realsense2_camera/share/realsense2_camera/config/d435i.yaml'", 'description': 'yaml config file of d435i camera'},
                    ]

def generate_launch_description():
    return LaunchDescription(
        rs_launch.declare_configurable_parameters(local_parameters) +
        [
            IncludeLaunchDescription(PythonLaunchDescriptionSource([ThisLaunchFileDir(), '/rs_launch.py']),
            launch_arguments=rs_launch.set_configurable_parameters(local_parameters).items(),
            ),
        ])`

`d435i:
    d435i:
      ros__parameters:
        device_type: 'd435i'
        serial_no: '048122072440'
        usb_port_id: ''

        base_frame_id: camera_link
        odom_frame_id: odom

        color_fps: 30.0
        color_height: 240
        color_width: 424
        color_qos: "SENSOR_DATA"

        depth_fps: 30.0
        depth_height: 240
        depth_width: 424
        depth_qos: "SENSOR_DATA"

        infra_height: 720
        infra_width: 1280
        infra_qos: "SENSOR_DATA"

        enable_accel: true
        enable_color: true
        enable_depth: true
        enable_gyro: true
        enable_infra1: false
        enable_infra2: false

        filters: 'pointcloud'

        pointcloud_texture_index: 0
        pointcloud_texture_stream: RS2_STREAM_COLOR

        stereo_module:
          emitter_enabled: 1
        
        unite_imu_method: 'linear_interpolation'

        publish_odom_tf: false`

Thanks a lot for your help!

MartyG-RealSense, 2021-09-29:
Hi @rythm07  It looks as though you are using the ROS2 wrapper.  Can you confirm this, please?

I have not encountered a launch file called **rs_d435i**.  Is this a custom launch file that you have made yourself?

rythm07, 2021-09-30:
Yes @MartyG-RealSense, sorry I didn't mention that. I am using the ROS2 Wrapper and I am using a custom launch file that is shown above in my first comment.

MartyG-RealSense, 2021-09-30:
Bearing in mind that your launch file previously worked for you, have you changed your ROS2 wrapper version recently?  The latest ROS2 wrapper at the time of writing this (3.2.2) made a change to the reading of .yaml config files.

[image]

rythm07, 2021-10-01:
Thanks for the fast answer! I haven't changed it, but I tried again and for some reason, it worked. I don't really know what happened but now is working!

Thanks a lot for your help and availability!

I will close the issue for the sake of control and organization of this repository :)

closed 2021-10-01 by rythm07