realsenseai/realsense-ros #2096: SLAM with realsense 435i

ale26030, 2021-09-30:
I am trying to start a 3d room scan with a realsense 435i. To do this I am following this guide (https://github.com/IntelRealSense/realsense-ros), but despite having executed all the commands indicated, and installed ros melodic (I have ubuntu 18.04), I cannot start the scan and run this command: roslaunch realsense2_camera rs_camera.launch. This is what appears to me on the terminal when I try to run the command
[image]
what can I do?

MartyG-RealSense, 2021-09-30:
Hi @ale26030  If you are receiving the control_transfer returned warning only a small number of times during launch and then they stop, or they occur one or two times a minute after launch, then this warning can be safely ignored.  You can check in RViz to see whether the topics are being published correctly despite the warnings.  If this warning is generating continuously though then it can indicate a serious communication problem with the device.

As a starting point in investigating your case, could you please try adding an **initial_reset:=true** command to your roslaunch instruction to reset the camera at launch:

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

ale26030, 2021-09-30:
I tried to execute the command you indicated, but nothing changes, I always have the same terminal outputs. Furthermore, trying to launch the command roslaunch realsense2_camera opensource_tracking.launch, which is indicated in the description of this video (https://www.youtube.com/watc
[image]
h?v=mguXtybmZv4), rviz opens, but inside it I should display images, but this does not happen, despite the fact that you see the image from the camera.
From the terminal I see that this error constantly appears:
[ERROR] (2021-09-30 12:40:08.177) Rtabmap.cpp:1156::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 2763 is ignored!

MartyG-RealSense, 2021-09-30:
Regarding the quoted video: https://www.youtube.com/watch?v=mguXtybmZv4 - they are apparently using Intel's D435i SLAM guide in the link below.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

The YouTube video is dated June 2020.  The following month in July 2020 though, in RealSense ROS wrapper **2.2.15**, changes in the wrapper's default settings were made where the IMU gyro and accel topics (and also the infra topics) were disabled by default.  

If your problem is that odometry is not appearing, could you try enabling the gyro and accel topics in your roslaunch instruction please:

**roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true**

ale26030, 2021-09-30:
With this command the odometry problem is solved. In any case, when opening rviz, the point clouds do not appear, and the situation I am in is that of the previous image. 

I followed that guide too but by executing the command to start the slam I get those warnings that I indicated in the first post, and no streaming starts; those warnings just keep popping up in an infinite loop.

closed 2021-10-14 by MartyG-RealSense