realsenseai/realsense-ros #2097: d435i tf  always maintained at 0 seconds

weizhi0304, 2021-10-02:
Hi everyone,
I want to read the tf from d435i ,I use this instruction
`rosrun tf tf_echo /camera_link /camera_color_frame`
The results show that:
```
At time 0.000
- Translation: [0.000, 0.015, 0.000]
- Rotation: in Quaternion [0.001, -0.002, 0.004, 1.000]
            in RPY (radian) [0.002, -0.004, 0.008]
            in RPY (degree) [0.102, -0.249, 0.432]
```
The time always maintained at 0 seconds.
Does anyone have an idea about it?

MartyG-RealSense, 2021-10-02:
Hi @weizhi0304  May I first confirm if you have performed roslaunch of the RealSense ROS wrapper first with an instruction such as **roslaunch realsense2_camera rs_camera.launch** and then performed rosrun secondly after launch has completed?

Also, if you have launched then do you have any **Ignoring transform for child_frame_id** errors listed in the log in the ROS terminal?  If so then performing a **gold reset** procedure on the camera to reset it (and its intrinsics and extrinsics) to their factory defaults has been known to fix problems related to using **rosrun tf tf_echo /camera_link /camera_color_frame**.  The gold reset procedure is described in https://github.com/IntelRealSense/librealsense/issues/6914#issuecomment-663413131

weizhi0304, 2021-10-03:
Hi @MartyG-RealSense I use `roslaunch realsense2_camera rs_camera.launch` first and than use `rosrun tf tf_echo /camera_link /camera_color_frame`.
Accelerometer information and gyroscop information are working normally.But it is no change after **gold reset**,tf is still maintained 0 seconds.

MartyG-RealSense, 2021-10-03:
My understanding of **tf echo** is that it prints information about the transformation between a source frame and target frame (in this case, between /camera link and /camera_color_frame).

There is a case at https://github.com/IntelRealSense/realsense-ros/issues/497#issue-371980865 where the RealSense ROS user in that case - who was comparing /camera_link and /camera_depth_frame (not camera_color_frame) was provided with a time if they launched with the **view_d435_model.launch** launch file but not if they launched with a live camera (as you are doing yourself with **rs_camera.launch**).

In another case at https://github.com/IntelRealSense/realsense-ros/issues/1449#issuecomment-711675699 **Doronhi** the RealSense ROS wrapper developer quoted a comparison of camera_depth_optical_frame and camera_gyro_optical_frame generated by **rosrun tf tf_echo camera_depth_optical_frame camera_gyro_optical_frame** that also had a time of 0.000.

In yet another case at https://github.com/IntelRealSense/realsense-ros/issues/772#issuecomment-493139646 a comparison of camera_pose_frame and camera_link first and then camera_fisheye1_frame and camera_gyro_frame both produced a 0.000 time result.

My knowledge of TF is admittedly limited.  But based on these past cases, 0.000 would seem to be an expected result when using RealSense cameras.

weizhi0304, 2021-10-04:
Ok ,thank you very much for your help.

weizhi0304, 2021-10-14:
No,thanks.