realsenseai/realsense-ros #2099: Auto exposure was setting itself from False to True

michalvi, 2021-10-04:
Hi,

Recently we've noticed some strange behaviour with the following setup:

**camera**: D415
**firmware**: 05.12.15.50
**LibRealSense**: v2.41.0
**RealSense ROS**: v2.2.21

Namely, after a restart, the auto exposure was setting itself from False to True.

The way we set it is by:
1. Setting a rosparam `/camera_source/rgb_camera/enable_auto_exposure` to False via a `rosparam set` command.
2. Launching the ros node using `rs_camera.launch` file.

The way we detected it is by seeing pictures being overexposed and then seeing also the parameter to be set to True by checking:

```
rosrun dynamic_reconfigure dynparam get /camera_source/rgb_camera
```

The issue doesn't happen very often, at most once per day. 

e can't yet reproduce the problem though. The last time it happened was on the 30th of September.

1. Do you have any intuition about what could've caused it?
2. Is there anything I should specifically check when this happens again? 

Kind regards,
Michał K.

PS. I attach logs from the last event of this kind: [camera_source_logs.txt](https://github.com/IntelRealSense/realsense-ros/files/7276545/camera_source_logs.txt)

MartyG-RealSense, 2021-10-04:
Hi @michalvi   The message **Failed to read modalias** would seem to be sourced from a section of the librealsense SDK's code that reacts if there is a problem in video4linux when using the SDK's **V4L2 backend**.  My understanding is that this backend would be used if librealsense was installed from packages using the instructions on the **distribution_linux.md** page in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

If librealsense and the RealSense ROS wrapper were installed together at the same time from the packages using **Method 1** then they would use the **RSUSB backend** instead.

https://github.com/IntelRealSense/realsense-ros#method-1-the-ros-distribution

I have not previously seen the two main errors in your ROS log in RealSense ROS cases before.

Failed to read modalias Last Error: No such file or directory
ERROR: Unable to communicate with master!

When debugging error messages, a useful rule to apply is to look at the first error that appears.  This is because subsequent errors may result from a cascade of faults caused by the first error that will disappear if the first error is corrected.  In your log, the first error to appear was **Failed to read modalias  Last Error: No such file or directory**, suggesting that the subsequent **ERROR: Unable to communicate with master!** may not be the real problem but a side-effect of the error before it.

From the log, it looks as though after the 'Unable to communicate with master' errors occurred, the master was re-found and an automatic re-launch occurred using **realsense2_camera/launch/rs_camera.launch**.  If you used 'rosparam set' to set auto-exposure to false in the original launch then if that rosparam is not inside the rs_camera launch file, it may have relaunched with the camera's default confguration, meaning that auto-exposure would be enabled by default.

michalvi, 2021-10-05:
Hi @MartyG-RealSense!

Thank you for your response.

Indeed, as you suggested in the last but one paragraph, the cause was that the 'rosparam set' was failing due to ros master sometimes not being up yet. **ERROR: Unable to communicate with master!** appeared exactly 7 times in the log because we were setting a set of 7 parameters this way. We managed to reproduce the problem and fix it by essentially waiting for ros master to appear first.

Thank you also for the comment about the modalias. We'll try to fix it as well to avoid any possible unknown issues.

MartyG-RealSense, 2021-10-05:
It's great to hear that you made positive progress, @michalvi - please do let me know if there is anything else that I can help you with.