realsenseai/realsense-ros #2103: errors when roslaunch with LR200

wenyan96, 2021-10-07:
Hi, recently i'm using LR200 to capture some rgb and depth images and want to use them in visual slam.
For this sensor LR200, i used librealsense-legacy and realsense-ros-indigo-devel. I tried mainly two method to build them.  
  
  
### Method 1
http://wiki.ros.org/realsense_camera/Tutorials/Building_librealsense_from_Sources
like the reference i tried to put them in the same workspace
  
### Method 2
https://github.com/IntelRealSense/realsense-ros/issues/386
i tried what  **doisyg** said in this issue
  
   
For both methods i can run the examples file of librealsense like ./cpp-capture successfully. BUT i met the same error as following when i roslaunch : `roslaunch realsense_camera r200_nodelet_rgbd.launch `

SUMMARY
========

PARAMETERS`
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/disparity_depth/max_range: 4.0
 * /camera/disparity_depth/min_range: 0.5
 * /camera/disparity_registered_sw/max_range: 4.0
 * /camera/disparity_registered_sw/min_range: 0.5
 * /camera/driver/base_frame_id: camera_link
 * /camera/driver/camera_type: R200
 * /camera/driver/color_fps: 30
 * /camera/driver/color_frame_id: camera_rgb_frame
 * /camera/driver/color_height: 480
 * /camera/driver/color_optical_frame_id: camera_rgb_optica...
 * /camera/driver/color_width: 640
 * /camera/driver/depth_fps: 30
 * /camera/driver/depth_frame_id: camera_depth_frame
 * /camera/driver/depth_height: 360
 * /camera/driver/depth_optical_frame_id: camera_depth_opti...
 * /camera/driver/depth_width: 480
 * /camera/driver/enable_color: True
 * /camera/driver/enable_depth: True
 * /camera/driver/enable_fisheye: False
 * /camera/driver/enable_imu: False
 * /camera/driver/enable_ir2: True
 * /camera/driver/enable_ir: True
 * /camera/driver/enable_pointcloud: False
 * /camera/driver/enable_tf: True
 * /camera/driver/fisheye_frame_id: camera_fisheye_frame
 * /camera/driver/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/driver/imu_frame_id: camera_imu_frame
 * /camera/driver/imu_optical_frame_id: camera_imu_optica...
 * /camera/driver/ir2_frame_id: camera_ir2_frame
 * /camera/driver/ir2_optical_frame_id: camera_ir2_optica...
 * /camera/driver/ir_frame_id: camera_ir_frame
 * /camera/driver/ir_optical_frame_id: camera_ir_optical...
 * /camera/driver/mode: manual
 * /camera/driver/serial_no: 
 * /camera/driver/usb_port_id: 
 * /camera/points_xyzrgb_sw_registered/queue_size: 100
 * /rosdistro: noetic
 * /rosversion: 1.15.13

NODES
  /camera/
    camera_nodelet_manager (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_points (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_sw_metric_rect (nodelet/nodelet)
    disparity_depth (nodelet/nodelet)
    disparity_registered_sw (nodelet/nodelet)
    driver (nodelet/nodelet)
    ir_rectify_ir (nodelet/nodelet)
    points_xyzrgb_sw_registered (nodelet/nodelet)
    register_depth_rgb (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

```
process[camera/camera_nodelet_manager-1]: started with pid [97411]
process[camera/driver-2]: started with pid [97412]
process[camera/ir_rectify_ir-3]: started with pid [97413]
process[camera/depth_rectify_depth-4]: started with pid [97414]
process[camera/depth_metric_rect-5]: started with pid [97419]
process[camera/depth_metric-6]: started with pid [97425]
process[camera/depth_points-7]: started with pid [97431]
[ INFO] [1633568091.946440709]: Initializing nodelet with 4 worker threads.
process[camera/register_depth_rgb-8]: started with pid [97443]
process[camera/points_xyzrgb_sw_registered-9]: started with pid [97450]
process[camera/depth_registered_sw_metric_rect-10]: started with pid [97455]
process[camera/disparity_depth-11]: started with pid [97463]
process[camera/disparity_registered_sw-12]: started with pid [97468]
[ERROR] [1633568092.319503431]: Failed to load nodelet [/camera/driver] of type [realsense_camera/R200Nodelet] even after refreshing the cache: Failed to load library /home/wenyan/Documents/rs_ros/devel/lib//librealsense_camera_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /home/wenyan/Documents/rs_ros/devel/lib//librealsense_camera_nodelet.so: undefined symbol: _ZN2cv6String10deallocateEv)
[ERROR] [1633568092.319575260]: The error before refreshing the cache was: Failed to load library /home/wenyan/Documents/rs_ros/devel/lib//librealsense_camera_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /home/wenyan/Documents/rs_ros/devel/lib//librealsense_camera_nodelet.so: undefined symbol: _ZN2cv6String10deallocateEv)
[FATAL] [1633568092.319764125]: Failed to load nodelet '/camera/driver` of type `realsense_camera/R200Nodelet` to manager `camera_nodelet_manager'
[camera/driver-2] process has died [pid 97412, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load realsense_camera/R200Nodelet camera_nodelet_manager depth:=depth color:=rgb rgb/image_raw:=rgb/image_rect_color ir:=ir ir2:=ir2 fisheye:=fisheye imu:=imu __name:=driver __log:=/home/wenyan/.ros/log/5b53a102-2694-11ec-87ac-270be475c52c/camera-driver-2.log].
log file: /home/wenyan/.ros/log/5b53a102-2694-11ec-87ac-270be475c52c/camera-driver-2*.log
```
  
I did find similar errors in some issues, but no one's error like **undefined symbol: _ZN2cv6String10deallocateEv)**, and librealsense_camera_nodelet.so exists in the folder... After check the difference between R200 and LR200, i'm not sure if the error is caused by wrong sensor type? i shouldn't use launch file of r200 but of lr200? For the sensor LR200, i found this issue https://github.com/IntelRealSense/realsense-ros/issues/245, but i'm still confused about how i should change the files in realsense-ros folder.. Could you give me more details? 

**OR** maybe it's not the reason, there are still other problems?

Really looking forward to any response and appreciate for your help in advance!!!

<h3>System Configuration</h3>

Version | Your Configuration
-- | --
Operating System | Ubuntu 20.04.3 LTS
Kernel | 5.11.0-37-generic
ROS | noetic
ROS RealSense | ?
librealsense | legacy
Camera Type-Firmware | Intel RealSense LR200, Camera FW: 2.0.71.18

MartyG-RealSense, 2021-10-07:
I will highlight your case to **Doronhi** the RealSense ROS wrapper developer.  Thanks very much for your patience!

doronhi, 2021-10-10:
Hi,
Just a thought: The error message you receive, **undefined symbol: _ZN2cv6String10deallocateEv)** relates to the opencv function **deallocateEv**. I think the issue has more to do with a mix of old ROS distributions and newer opencv. Searching around for this error and ROS I found this link: https://answers.ros.org/question/219640/reference-errors-after-opencv3-installation. 
It basically says to get rid of all installations of new opencv versions.

closed 2021-10-30 by MartyG-RealSense