realsenseai/realsense-ros #2106: How to get depth value from rostopic subscriber

xczzz, 2021-10-09:
My camera is D455. I tried to get depth from rostopic. Here is my code. Firstly I subscribe rostopic to get aligned depth.
`
ros::Subscriber aligned_depth_sub = nh.subscribe("/camera/aligned_depth_to_color/image_raw", 1, &depth_callback);
`
Then, I change the msg into cv::Mat in callback function like this.
`
Mat depth_image = Mat::zeros(Size(848, 480), CV_16UC1);
void depth_callback(const sensor_msgs::ImageConstPtr &msg) {
  depth_image = cv_bridge::toCvCopy(msg, sensor_msgs::image_encodings::TYPE_16UC1)->image;
  imshow("Depth_Image", depth_image);
  waitKey(1);
}
`
Finally I get the depth value of a specific pixel which is (pixel_x, pixel_y).
`
float real_z = depth_scale * depth_image.at<uint16_t>(pixel_x, pixel_y);
`
But the value is absolutely wrong. So I wonder how can I get the correct value when I subscribing rostopic.

MartyG-RealSense, 2021-10-10:
Great to hear that you found a solution!  Thanks very much for sharing it with the RealSense ROS community  :)

xczzz, 2021-10-10:
@MartyG-RealSense I can get colorized depth image when using roslaunch with "filter:=colorizer". But then I can not get depth value from the black and white depth image with "depth_image.at<uint16_t>(pixel_y, pixel_x)". I hope to get a colorized depth image and depth value at the same. How can I do that?

xczzz, 2021-10-12:
@MartyG-RealSense I think rosbag is not suitable for my work. So I wonder how to get a specific point's 3D coordinate from pointcloud corresponding to a pixel in color image.

MartyG-RealSense, 2021-10-19:
Hi @xczzz  Do you require further assistance with this case, please?  Thanks!

JohannaPrinz, 2022-07-27:
@xczzz do you know if there is a similar solution in python for that?
I also using D455 and need the depth data from a specific (x,y) coordinate.

Would be fantastic if anybody could help me with that =)
Thanks a lot

MartyG-RealSense, 2022-07-27:
Hi @JohannaPrinz  If you are only using Python (not ROS) then the SDK instruction **rs2_project_color_pixel_to_depth_pixel** will convert a specific color pixel to a depth pixel to obtain Z-depth for that XY coordinate, as described in the Python script at https://github.com/IntelRealSense/librealsense/issues/5603#issuecomment-574019008

If you wanted to launch a Python **node script** from ROS then the script **show_center_depth.py** is a good reference.

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/scripts/show_center_depth.py