realsenseai/realsense-ros #2107: Realsense2 with L515 unable to view any streams

seandburke99, 2021-10-09:
I have a realsense l515 that I am trying to use in ROS2 (foxy). I am able to use my camera normally in realsense-viewer and see the depth and image data with no issue.
When I run the rs_launch.py file, I am unable to view any data in rviz. I have set the fixed frame correctly but I cannot even view the 2D color or image data. When I launch the file with the parameter "filter:=pointcloud", the point cloud topic is created and I can echo the pointcloud topic and see taht data is being published but it is not viewable in rviz. The same issue persists when I try launching the demo_pointcloud_launch.py demo. I have tried lowering the fps and disabling the infra stream to just see the depth and color with no success. I have attached console logs of both me running the pointcloud demo and a log of the pointcloud topic.
console from launch
`
[INFO] [launch]: All log files can be found below /home/group14/.ros/log/2021-10-09-12-44-26-794696-group14-68102
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [68104]
[INFO] [rviz2-2]: process started with pid [68106]
[realsense2_camera_node-1] [0m[INFO] [1633794267.020621591] [RealSenseCameraNode]: RealSense ROS v3.2.2[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.020674050] [RealSenseCameraNode]: Built with LibRealSense v2.48.0[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.020683287] [RealSenseCameraNode]: Running with LibRealSense v2.48.0[0m
[realsense2_camera_node-1]  09/10 12:44:27,129 WARNING [140508059899648] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
[realsense2_camera_node-1] [0m[INFO] [1633794267.222955647] [RealSenseCameraNode]: Device with serial number f0271990 was found.
[realsense2_camera_node-1] [0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.222980546] [RealSenseCameraNode]: Device with physical ID 2-1-8 was found.[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.222985988] [RealSenseCameraNode]: Device with name Intel RealSense L515 was found.[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.223283864] [RealSenseCameraNode]: Device with port number 2-1 was found.[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.223299400] [RealSenseCameraNode]: Device USB type: 3.2[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.224880141] [RealSenseCameraNode]: getParameters...[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.227040826] [RealSenseCameraNode]: setupDevice...[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.227066731] [RealSenseCameraNode]: JSON file is not provided[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.227082456] [RealSenseCameraNode]: Device Name: Intel RealSense L515[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.227091757] [RealSenseCameraNode]: Device Serial No: f0271990[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.227100256] [RealSenseCameraNode]: Device physical port: 2-1-8[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.227108131] [RealSenseCameraNode]: Device FW version: 01.05.08.01[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.227116064] [RealSenseCameraNode]: Device Product ID: 0x0B64[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.227123627] [RealSenseCameraNode]: Enable PointCloud: On[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.227131141] [RealSenseCameraNode]: Align Depth: Off[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.227138674] [RealSenseCameraNode]: Sync Mode: On[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.227160318] [RealSenseCameraNode]: Device Sensors: [0m
[rviz2-2] [INFO] [1633794267.240073303] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [1633794267.240274091] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[realsense2_camera_node-1] [0m[INFO] [1633794267.242198489] [RealSenseCameraNode]: L500 Depth Sensor was found.[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.249633427] [RealSenseCameraNode]: RGB Camera was found.[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.249808784] [RealSenseCameraNode]: Motion Module was found.[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.249837733] [RealSenseCameraNode]: (Infrared, 1) sensor isn't supported by current device! -- Skipping...[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.249850620] [RealSenseCameraNode]: (Infrared, 2) sensor isn't supported by current device! -- Skipping...[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.249861769] [RealSenseCameraNode]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.249872174] [RealSenseCameraNode]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.249882320] [RealSenseCameraNode]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.249899629] [RealSenseCameraNode]: (Pose, 0) sensor isn't supported by current device! -- Skipping...[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.249912587] [RealSenseCameraNode]: Add Filter: pointcloud[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.250494611] [RealSenseCameraNode]: num_filters: 1[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.250513150] [RealSenseCameraNode]: Setting Dynamic reconfig parameters.[0m
[rviz2-2] [INFO] [1633794267.274800260] [rviz2]: Stereo is NOT SUPPORTED
[realsense2_camera_node-1] [33m[WARN] [1633794267.426957036] [RealSenseCameraNode]: Could not set param: pointcloud.stream_index_filter with 0Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {} doesn't comply with integer range.[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.427008218] [RealSenseCameraNode]: Done Setting Dynamic reconfig parameters.[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.427463629] [RealSenseCameraNode]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.427746336] [RealSenseCameraNode]: infra stream is enabled - width: 640, height: 480, fps: 30, Format: Y8[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.429618536] [RealSenseCameraNode]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.429717321] [RealSenseCameraNode]: setupPublishers...[0m
[realsense2_camera_node-1] [0m[INFO] [1633794267.434259141] [RealSenseCameraNode]: setupStreams...[0m
[rviz2-2] [INFO] [1633794267.446737932] [rviz2]: Stereo is NOT SUPPORTED
[realsense2_camera_node-1]  09/10 12:44:27,445 WARNING [140508059899648] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
[realsense2_camera_node-1]  09/10 12:44:27,540 WARNING [140508059899648] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[realsense2_camera_node-1] [0m[INFO] [1633794267.557363900] [RealSenseCameraNode]: SELECTED BASE:Depth, 0[0m
[realsense2_camera_node-1]  09/10 12:44:27,590 WARNING [140508059899648] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[realsense2_camera_node-1]  09/10 12:44:27,641 WARNING [140508059899648] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[realsense2_camera_node-1]  09/10 12:44:27,692 WARNING [140508059899648] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[realsense2_camera_node-1] [0m[INFO] [1633794267.742718609] [RealSenseCameraNode]: RealSense Node Is Up![0m
[realsense2_camera_node-1] [33m[WARN] [1633794267.747024564] [RealSenseCameraNode]: [0m
[realsense2_camera_node-1] [33m[WARN] [1633794267.747077428] [RealSenseCameraNode]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.[0m
[realsense2_camera_node-1]  09/10 12:44:28,246 WARNING [140508059899648] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[realsense2_camera_node-1]  09/10 12:44:28,592 WARNING [140508059899648] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
[realsense2_camera_node-1]  09/10 12:44:33,597 WARNING [140508059899648] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[realsense2_camera_node-1]  09/10 12:44:34,849 WARNING [140508059899648] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... 8096 lines cut ...]
- 191
- 74
- 12
- 44
- 64
- 0
- 0
- 0
- 0
- 51
- 67
- 57
- 0
- 197
- 229
- 167
- 191
- 163
- 71
- 146
- 191
- '...'
is_dense: true
---
`

MartyG-RealSense, 2021-10-13:
There are differences between programs that run directly in the RealSense SDK (such as the RealSense Viewer) and a launch in the RealSense ROS wrapper (which is a layer that interacts closely with librealsense but handles some processes differently in order to have compatibility with ROS).

seandburke99, 2021-10-15:
I have with both the "launch_demo_pointcloud" and "rs_launch" and many combinations of the parameter options.

MartyG-RealSense, 2021-10-15:
If librealsense is installed on its own from packages then the RealSense ROS2 wrapper should subsequently be installed from source code using **Method 2** in the ROS wrapper installation documentation  The librealsense SDK should be installed first and then the ROS wrapper installed secondly afterwards.

If librealsense was installed on its own from packages then ROS wrapper installation **Method 1** should not be used.  This is because this method installs both librealsense and the ROS wrapper together from packages.  So if you installed librealsense on its own first and then used Method 2, you would be installing librealsense twice.

If you do not have an Nvidia graphics chip in your computer and therefore do not need CUDA graphics acceleration support (which only works with Nvidia graphics hardware) then the simplest approach may be to install ROS Foxy from Debian packages and then install librealsense and the ROS wrapper together from packages using Method 1.

seandburke99, 2021-10-15:
I would like to give an update. I believe the issue did not stem from the Real sense library. I believe there was some sort of issue with the rviz2 installation. I have tried the camera on a different machine and although the errors are still firing away with almost every frame, I have successfully been able to get all of the feeds and even the pointcloud to show up in Rviz2 on the other machine. I think this issue can be closed unless you have anything further to add.

MartyG-RealSense, 2021-10-16:
I do not have anything further to add.  Thanks very much @seandburke99 for the update of your progress!  As suggested, I will close the issue. Thanks again   :)

closed 2021-10-16 by MartyG-RealSense