realsenseai/realsense-ros #2117: depthimage_to_laserscan package didn't work with rs_d400_and_t265_launch.py

giry-ankoro, 2021-10-13:
Hi !
I wanted to use the depthimage_to_laserscan package to get the /scan topic with rs_d400_and_t265_launch.py on real sense cameras, but it didn't work at all.

But interestingly, this package somehow worked in rs_launch.py.

I've started using ROS2(foxy) recently, so I'm not used to it yet.

I would be grateful if you could help me!!!

https://github.com/ros-perception/depthimage_to_laserscan/tree/foxy-devel

RealSenseSupport, 2021-11-02:
@giry-ankoro  Could you please post the logs if you still need support on this? Thanks!

RealSenseSupport, 2021-11-15:
@giry-ankoro Please note that this ticket will be closed  if we don't get update from you in another 7 days. Thanks!

RealSenseSupport, 2021-11-22:
@giry-ankoro Sorry we don't hear from you for weeks. This ticket will be closed accordingly. Please feel free to create new ticket if you still have any other questions or issues. Thanks!

closed 2021-11-22 by RealSenseSupport