realsenseai/realsense-ros #2119: RealSense ROS v2.3.1 / LibRealSense v2.49.0 / control_transfer returned error

HeitorDC, 2021-10-14:
Hi, I'm using RealSense ROS v2.3.1 / LibRealSense v2.49.0 / Ubuntu 20.04 and tried all the solutions that were already mentioned in the other questions regarding this topic but without success. 

I need help to understand why it is not working and is generating the errors that are being generated. RealSense viewer and Rviz are working correctly, I've checked as informed and advised to do in other issues. 

As for what is happening, a colleague in the lab is using with Ubuntu 18.04 and LibRealSense v2.47, this will be the problem then, the incompatibility of the two versions, the viewer and ros-realsense with the Ubuntu version?

heitor@heitorpc:~$ roslaunch realsense2_camera rs_camera.launch 
... logging to /home/heitor/.ros/log/b1b9441c-2d23-11ec-9fde-57c076c74a72/roslaunch-heitorpc-79085.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://heitorpc:37131/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
[... 78 lines cut ...]
[ INFO] [1634239188.275050563]: num_filters: 0
[ INFO] [1634239188.275068281]: Setting Dynamic reconfig parameters.
[ WARN] [1634239188.704851449]: Param '/camera/stereo_module/auto_exposure_limit' has value 0 that is not in range [1, 165000]. Removing this parameter from dynamic reconfigure options.
[ WARN] [1634239188.718328246]: Param '/camera/stereo_module/auto_gain_limit' has value 0 that is not in range [16, 248]. Removing this parameter from dynamic reconfigure options.
[ INFO] [1634239188.823959920]: Done Setting Dynamic reconfig parameters.
[ INFO] [1634239188.825193018]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1634239188.828206929]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1634239188.828262735]: setupPublishers...
[ INFO] [1634239188.833393257]: Expected frequency for depth = 30.00000
[ INFO] [1634239188.904023954]: Expected frequency for color = 30.00000
[ INFO] [1634239188.937552897]: setupStreams...
[ INFO] [1634239188.948787613]: insert Depth to Stereo Module
[ INFO] [1634239188.948900497]: insert Color to RGB Camera
[ INFO] [1634239189.013206353]: SELECTED BASE:Depth, 0
[ INFO] [1634239189.022778625]: RealSense Node Is Up!
[ WARN] [1634239189.133273478]: 
 14/10 16:19:49,134 WARNING [140039950427904] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 14/10 16:19:49,184 WARNING [140039950427904] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 14/10 16:19:49,236 WARNING [140039950427904] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 14/10 16:19:49,286 WARNING [140039950427904] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 14/10 16:19:49,465 WARNING [140039950427904] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 14/10 16:19:49,515 WARNING [140039950427904] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 14/10 16:19:49,668 WARNING [140039950427904] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 14/10 16:19:49,820 WARNING [140039950427904] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 14/10 16:19:51,678 WARNING [140039950427904] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

MartyG-RealSense, 2021-10-15:
Hi @HeitorDC  Ideally the ROS wrapper version should be matched as closely as possible with te librealsense SDK version listed in the release notes for a particular wrapper version.  It is recommended that wrapper 2.3.1 is matched with librealsense version **2.48.0**.  There is not a ROS wrapper version available for 2.49.0.  At the time of writing this, it is advised that the next wrapper version should be for SDK 2.50.0 that will be released in about 2 weeks from now.

In regard to the **control_transfer returned** warnings in the log, these can be ignored if they are only appearing at the start of launch or appearing about once a minute instead of generating continuously.  The topics may still be able to be correctly published in RViz even if these messages appear.  If the control_transfer returned warnings are generated continuously though then this may indicate a serious communication problem with the device.

Could you test please whether adding **initial_reset:=true** to your roslaunchh instruction to reset the camera at launch makes a positive difference, please?  For example:

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

HeitorDC, 2021-10-20:
Hi @MartyG-RealSense, thanks for your answer! I've tried the solution that you proposed but it didn't work. I decided to give up to use ROS. I have an another issue but I'll open a new with the correct subject.

Thanks again for your help.

closed 2021-10-20 by HeitorDC