realsenseai/realsense-ros #2144: SLAM with D435i tutorial： how to evaluate the localization accuracy？

ClarkWang1214, 2021-10-31:
Hi, 

I am new to VSLAM. I am following the instructions of "SLAM with D435i tutorial"
roslaunch realsense2_camera opensource_tracking.launch,

I have got the 2D occupancy map of my office successfully.

I want to know how to evaluate the localization accuracy without any GroundTruth? 

Could you give me some suggestion？

Thx.

[image]

[image]

MartyG-RealSense, 2021-10-31:
Hi @ClarkWang1214  At the link below, a RealSense user created an adaptation of Intel's D435i SLAM guide that added localization with RTABMAP and obtaining the relative position from a recorded pointcloud map.  Does this guide meet your needs, please?

https://shinkansan.github.io/2019-UGRP-DPoom/SLAM

ClarkWang1214, 2021-10-31:
> Hi @ClarkWang1214 At the link below, a RealSense user created an adaptation of Intel's D435i SLAM guide that added localization with RTABMAP and obtaining the relative position from a recorded [...]

Thanks for your reply. @MartyG-RealSense 

Yes, I have read the link you provided. 

In the "Post-processing Position Data from Matching" part, I don't understand the conclusion,  "The position data describes relative location from the origin, and the unit is meter. As result of comparision, error of the data is just few centimeters"，Could you give me some more narrative details or some links? 

Thank you~

MartyG-RealSense, 2021-10-31:
A mobile robot's measurements can drift in accuracy as the robot moves further from the start position.  The drift can be corrected by performing a procedure called **relocalization**.  https://github.com/IntelRealSense/realsense-ros/issues/1529#issuecomment-735299914 provides links with more details on this subject in regard to ROS.

MartyG-RealSense, 2021-11-07:
Hi @ClarkWang1214  Do you require further assistance with this case, please?  Thanks!

ClarkWang1214, 2021-11-10:
> Hi @ClarkWang1214 Do you require further assistance with this case, please? Thanks!

yes, i have met another problem. 

In my Jetson Xavier NX, i have installed the librealsense, realsense-ros, rtabmap 0.20.0(source code version),  rtabmap_ros 0.20.0

when i executed the command: "roslaunch realsense2_camera opensource_tracking.launch"

the rtabmap/rgbd_odometry did not work with the error: "[rtabmap/rgbd_odometry-5] process has died"

when using the binary version of the rtabmap, it works. 

could you help me to solve it? 

thanks

MartyG-RealSense, 2021-11-10:
If you are building rtabmap_ros from source code on Jetson, separate instructions for Jetson are provided in the link below.  Have you used the method in this link, please?

https://github.com/introlab/rtabmap_ros#build-from-source-for-nvidia-jetson

closed 2021-11-26 by MartyG-RealSense