realsenseai/realsense-ros #2145: Low Frame Rate Raspberry Pi

sergioar3143, 2021-11-03:
Hello again, I published https://github.com/IntelRealSense/realsense-ros/issues/2131 few days ago, everything is good but I want to have 30 fps. I read something about that in https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras but I need to job in ROS. How can I do it?

MartyG-RealSense, 2021-11-03:
Hi @sergioar3143   A RealSense ROS user in https://github.com/IntelRealSense/realsense-ros/issues/1808 modified the open source ethernet system to use ROS networking protocols instead, though FPS performance was not excellent at around 10 FPS.  I have not seen anybody else attempt such an adaptation since then.

When the networking white-paper mentions ROS, it has the number [10] beside it that refers to a RealSense ROS guide listed in the references at the bottom of the paper for using 3 cameras across 2 computers:

https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines

sergioar3143, 2021-11-05:
And only with the raspberry without computer, can I increase the fps?

MartyG-RealSense, 2021-11-05:
I believe that the networking method has more effect on the achieved FPS than the computer type used.  Of the cases that I have seen regarding networking ROS to RealSense cameras, my recollection is that low FPS values such as 10 are more likely to be encountered with a wireless network connection such as wi-fi or WLAN, whereas a wired ethernet connection has a chance of achieving 30 FPS.

sergioar3143, 2021-11-05:
Ok thank you

closed 2021-11-05 by sergioar3143