realsenseai/realsense-ros #2146: D435i Camera pointcloud coordinate frame keeps shifting (flickering) in ROS rviz

AnsunWu, 2021-11-05:
**System Configuration**
**System OS:** Ubuntu 16.04 with ROS Kinetic and rviz.
**Camera and firmware:**  Realsense D435i with firmware 05.12.14.50.
**LibealSense :** v2.45.0
**USB port** : USB3.2

**Problem Behavior**
Hello, I use the ' roslaunch rs_camera.launch ' command to open the Camera node, the output info from the same console is shown below. Everything looks normal. 

I tried to monitor the point cloud data in the ROS rviz and use the **'Axis color'** option to use different colors to represent the distance of points in the point cloud data. The screenshot is shown below:

[image]

****Problem**: the point cloud color is keep flickering in rviz, looks like the coordinate system in which the point cloud is located
keeps shifting.  I'm using the D435 camera to do a 3D reconstruction project, the coordinate frame flickering may be the reason for the bad fusing result...The point cloud flickering video can be checked via [https://youtu.be/FlCQLdsOxwI](url)**

 **Could you please help me check it? many thanks for your help.**

```
NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [23378]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3d4a9a00-3e43-11ec-85cf-907841da8257
process[rosout-1]: started with pid [23393]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [23410]
process[camera/realsense2_camera-3]: started with pid [23411]
[ INFO] [1636121906.614184026]: Initializing nodelet with 8 worker threads.
[ INFO] [1636121906.661287007]: RealSense ROS v2.3.0
[ INFO] [1636121906.661318687]: Built with LibRealSense v2.45.0
[ INFO] [1636121906.661332840]: Running with LibRealSense v2.45.0
[ INFO] [1636121906.691801875]:  
[ INFO] [1636121906.933805511]: Device with serial number 052622073834 was found.

[ INFO] [1636121906.933838630]: Device with physical ID 2-2-2 was found.
[ INFO] [1636121906.933850177]: Device with name Intel RealSense D435I was found.
[ INFO] [1636121906.934413718]: Device with port number 2-2 was found.
[ INFO] [1636121906.934441273]: Device USB type: 3.2
[ INFO] [1636121906.936570224]: getParameters...
[ INFO] [1636121907.027871507]: setupDevice...
[ INFO] [1636121907.734949990]: JSON file is loaded! (/opt/ros/kinetic/share/realsense2_camera/launch/includes/camera_json/MidDensityPreset.json)
[ INFO] [1636121907.735080828]: ROS Node Namespace: camera
[ INFO] [1636121907.735152132]: Device Name: Intel RealSense D435I
[ INFO] [1636121907.735206519]: Device Serial No: 052622073834
[ INFO] [1636121907.735282801]: Device physical port: 2-2-2
[ INFO] [1636121907.735334050]: Device FW version: 05.12.14.50
[ INFO] [1636121907.735395065]: Device Product ID: 0x0B3A
[ INFO] [1636121907.735463477]: Enable PointCloud: On
[ INFO] [1636121907.735533687]: Align Depth: Off
[ INFO] [1636121907.735586283]: Sync Mode: On
[ INFO] [1636121907.735727565]: Device Sensors: 
[ INFO] [1636121907.766994501]: Stereo Module was found.
[ INFO] [1636121907.792772642]: RGB Camera was found.
[ INFO] [1636121907.793073282]: Motion Module was found.
[ INFO] [1636121907.793111841]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1636121907.793147157]: Add Filter: pointcloud
[ INFO] [1636121907.793928540]: num_filters: 1
[ INFO] [1636121907.793953852]: Setting Dynamic reconfig parameters.
[ WARN] [1636121908.235221785]: Param '/camera/stereo_module/auto_exposure_limit' has value 0 that is not in range [1, 165000]. Removing this parameter from dynamic reconfigure options.
[ WARN] [1636121908.248604702]: Param '/camera/stereo_module/auto_gain_limit' has value 0 that is not in range [16, 248]. Removing this parameter from dynamic reconfigure options.
[ INFO] [1636121908.388774496]: Done Setting Dynamic reconfig parameters.
[ INFO] [1636121908.389686162]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1636121908.390024434]: infra1 stream is enabled - width: 848, height: 480, fps: 30, Format: Y8
[ INFO] [1636121908.390311216]: infra2 stream is enabled - width: 848, height: 480, fps: 30, Format: Y8
[ INFO] [1636121908.390340466]: setupPublishers...
[ INFO] [1636121908.394441650]: Expected frequency for depth = 30.00000
[ INFO] [1636121908.432941663]: Expected frequency for infra1 = 30.00000
[ INFO] [1636121908.466116148]: Expected frequency for infra2 = 30.00000
[ INFO] [1636121908.502190259]: setupStreams...
[ INFO] [1636121908.525031286]: insert Depth to Stereo Module
[ INFO] [1636121908.525177701]: insert Infrared to Stereo Module
[ INFO] [1636121908.525257214]: insert Infrared to Stereo Module
 05/11 11:48:28,590 WARNING [139912376477440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1636121908.640975105]: SELECTED BASE:Depth, 0
[ INFO] [1636121908.661963508]: RealSense Node Is Up!
[ WARN] [1636121908.706959522]: 
 05/11 11:48:28,707 WARNING [139912376477440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 05/11 11:48:28,758 WARNING [139912376477440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 05/11 11:48:28,910 WARNING [139912376477440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 05/11 11:48:28,961 WARNING [139912376477440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 05/11 11:48:29,113 WARNING [139912376477440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 05/11 11:48:29,163 WARNING [139912376477440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 05/11 11:48:29,314 WARNING [139912376477440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 05/11 11:48:29,364 WARNING [139912376477440] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1636121915.143632368]: publishing ordered pointcloud.

```

MartyG-RealSense, 2021-11-05:
Hi @AnsunWu   Could you first confirm the full roslaunch instruction that you are using, please?  I note that in your log, the pointcloud filter and ordered pointcloud are enabled.  These options would normally be disabled by default if only the basic **roslaunch realsense2_camera rs_camera.launch** instruction is used.

AnsunWu, 2021-11-05:
Hi @MartyG-RealSense , 

Thanks for your reply.

**1. The modified rs_camera.launch is shown below**
```
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="json_file_path"      default="$(find realsense2_camera)/launch/includes/camera_json/MidDensityPreset.json"/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>

  <arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="848"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="infra_width"        default="848"/>
  <arg name="infra_height"       default="480"/>
  <arg name="enable_infra1"       default="true"/>
  <arg name="enable_infra2"       default="true"/>

  <arg name="color_width"         default="848"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="250"/>

  <!--Do not use Oct.26-->
  <arg name="enable_gyro"         default="false"/>
  <arg name="enable_accel"        default="false"/>

  <arg name="enable_pointcloud"         default="true"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_ANY"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <!--Try to solve the unstructured point cloud issue Oct.7-->
  <arg name="ordered_pc"  default="true"/>

  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="true"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="unite_imu_method"          default=""/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>
  <arg name="allow_no_texture_points"   default="true"/>

  
  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="external_manager"         value="$(arg external_manager)"/>
      <arg name="manager"                  value="$(arg manager)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>
[... 35 lines cut ...]
      <arg name="enable_accel"             value="$(arg enable_accel)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
    </include>
  </group>
</launch>

```

The **structured (ordered)** point cloud is needed in our project, that is why I set the "ordered_pc" to true.
The resolution of the depth_width/height camera and infra_width/height is set to 848 x 480.

**2. I tried to use the default rs_rgbd.launch** and subscribe to the aligned structured point cloud. The flickering problem decreases a little bit.

**3. By the way. is there any other setting in the rs_camera.launch file to let the camera output structured point cloud data?** 

**4.  I'm wondering, Is there a reason for this problem that the camera node keeps restarting or resetting because of computer resource problems (ram or CPU usage)?**

MartyG-RealSense, 2021-11-05:
1.  Thanks very much for the details of the modified launch file that show how the pointcloud and ordered cloud settings were configured in the launch.

2.  Yes, using rs_rgbd.launch is a way to achieve an ordered point cloud without having to set ordered_pc to true, as the rgbd point cloud is ordered by default.  The point cloud of rs_camera is unordered by default, which is why ordered_pc has to be set to true on that particular launch file type to order its cloud.  

3.  ordered_pc = true is the only setting to generate an ordered cloud with rs_camera.launch.

4.  If your point cloud is based on depth aligned to color then you could reduce processing demand by setting infra1 and infra2 to false so that the infra streams are not published, as the camera is not dependent on these two streams to generate the depth stream and they are disabled in the default unmodified rs_camera.launch file.

[image]

closed 2021-11-18 by MartyG-RealSense