realsenseai/realsense-ros #2147: Help with jetson nano

OffWhite10, 2021-11-05:
Hi, I am using Jetson nano with D435i and ROS, when I run in the terminal roslaunch realsense2_camera rs_camera.launch and then I put the command rostopic hz /camera/colo/image_raw I have like 7 Hz, I can see the image in rviz.
So, I added depth_fps:=30 and color_fps:=30 to the command but I can't see anything when I do that. How  can I change the speed?

MartyG-RealSense, 2021-11-05:
Hi @OffWhite10   When defining a custom stream configuration in a roslaunch, three factors need to be provided for each stream type defined in the custom configuration.  These are width, height and FPS.  

If these three factors are not provided then the custom configuration is deemed invalid by the launch and the default stream configuration of the particular RealSense camera model being used is applied instead.

For example, to configure both depth and color at 640x480 resolution and 30 FPS:

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30**

OffWhite10, 2021-11-06:
Thank you, this instruction solved the problem but I have some doubts, how do you know the relationship between height, width and fps? For example if I want to change the fps to 15, is the same values in height and width?

MartyG-RealSense, 2021-11-06:
Changing the FPS does not require the height and width of the resolution to be altered.  You can combine a range of different resolutions and FPS values.

OffWhite10, 2021-11-07:
Ok, and one more question, when I add in the command some filters, such as decimation or pointcloud, it affect to the fps values of the camera?

MartyG-RealSense, 2021-11-07:
Post-processing filters are processed on the computer CPU and not the camera hardware and so applying filters can reduce performance if the particular computer / computing device being used has a low specification.  If you were only able to apply one filter because of computing limitations then Decimation would be a recommended choice.

Generating a pointcloud can also increase processing demand, but if you build the librealsense SDK with CUDA support then Nvidia Jetson boards such as Nano can use **CUDA** to offload pointcloud processing from the CPU to the graphics GPU to reduce CPU usage.

You can enable CUDA support by building librealsense from source code with CMake and including the build term **BUILD_WITH_CUDA=true** or installing the SDK from Debian packages using the Jetson instructions in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

CUDA support will not be included if librealsense and the RealSense ROS wrapper are built together from packages with the wrapper's **Method 1**.

OffWhite10, 2021-11-09:
Okay, thank you so much

closed 2021-11-09 by OffWhite10