realsenseai/realsense-ros #2149: D435i rs_camera.launch results in (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132 with no messages being published thereafter

Likhith-Sugganahalli, 2021-11-08:
rosdistro: melodic
RealSense ROS v2.3.1
Built with LibRealSense v2.48.0
Running with LibRealSense v2.48.0
Device USB type: 3.2
device running on is a nvidia jetson xavier nx
the camera is also plugged into a powered usb
[image]
after uvc streamer watchdog triggers on endpoint 132, no message is published.
realsense-viewer works with no issue
please suggest a solution to the issue

MartyG-RealSense, 2021-11-08:
Hi @Likhith-Sugganahalli   The **control_transfer returned** warnings can be ignored if they are only appearing at the start of launch or appearing about once a minute instead of generating continuously.  The topics may still be able to be correctly published in RViz even if these messages appear.  If the control_transfer returned warnings are generated continuously though then this may indicate a serious communication problem with the device.

As a starting point in investigating your case, could you test please whether adding **initial_reset:=true** to your roslaunchh instruction to reset the camera at launch makes a positive difference, please?  For example:

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

Likhith-Sugganahalli, 2021-11-08:
@MartyG-RealSense thanks a lot for the prompt reply, i did as you suggested, but the uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132 persists, 
[image]
even after adding the device reset, the nvc streamer triggers a couple of publishes in

MartyG-RealSense, 2021-11-08:
Thank you very much.  Next, could you try the launch instruction below with a custom stream configuration that only provides low-resolution depth at 640x480 and 15 FPS.  A custom stream configuration overrides the defaults that are applied during launch.  This should provide insights about whether or not the default stream configuration is causing problems with the launch.

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=15**

Likhith-Sugganahalli, 2021-11-08:
@MartyG-RealSense 
i ran the following command
rs_camera.launch initial_reset:=true color_width:=640 color_height:=480 color_fps:=15
while the default was already color stream is enabled - width: 848, height: 480, fps: 15, Format: RGB8, i had edited earlier,
and yes, it seems the camera ran for longer before the nvc streamer struck

MartyG-RealSense, 2021-11-09:
Thanks very much for the further information.  Are you using the official 1 meter USB cable supplied with the camera or a longer cable of your own choice?

Likhith-Sugganahalli, 2021-11-09:
@MartyG-RealSense 
i have a active usb extender attached to the offical cable, but this error only pops up on my jetson xavier, the roslaunch runs fine on the PC, also the realsense-viewer runs fine on the jetson

MartyG-RealSense, 2021-11-09:
The realsense-viewer program runs directly in the librealsense SDK, whilst the RealSense ROS wrapper runs as a compatibility layer on top of librealsense.  The ROS wrapper also handles some processes differently from a librealsense program in order to maintain compatibility with ROS.  So it is common for the camera to work fine with realsense-viewer even if it is having problems with launching in ROS.

Would it be possible to test the camera directly connected to the Jetson with the short official cable instead of using the extender in order to eliminate the extender cable as a cause of the problem on Jetson, please?

Likhith-Sugganahalli, 2021-11-09:
@MartyG-RealSense hey, thanks a lot for your help, it seems when i change both depth and color resolutions now with fps set to 15 it works, not sure if i hadnt tried this out before, thanks a lot for your help

closed 2021-11-09 by Likhith-Sugganahalli

MartyG-RealSense, 2021-11-09:
That's excellent news, @Likhith-Sugganahalli - thanks very much for the update   :)

sasilva1998, 2022-03-30:
Hi, I am having the exact same problem, here is my config:

````
<launch>
  <arg name="serial_no"             default=""/>
  <arg name="json_file_path"        default=""/>
  <arg name="camera"                default="d435"/>

  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="serial_no"         value="$(arg serial_no)"/>
      <arg name="json_file_path"    value="$(arg json_file_path)"/>
      <arg name="depth_width"       value="480"/>
      <arg name="depth_height"      value="270"/>
      <arg name="depth_fps"         value="15"/>
      <arg name="color_width"       value="640"/>
      <arg name="color_height"      value="480"/>
      <arg name="color_fps"         value="15"/>
      <arg name="enable_depth"      value="true"/>
      <arg name="enable_color"      value="true"/>
      <arg name="enable_infra1"     value="false"/>
      <arg name="enable_infra2"     value="false"/>
      <arg name="enable_fisheye"    value="false"/>
      <arg name="enable_gyro"       value="false"/>
      <arg name="enable_accel"      value="false"/>
      <arg name="enable_pointcloud" value="true"/>
      <arg name="enable_sync"       value="true"/>
      <arg name="tf_prefix"         value="$(arg camera)"/>
      <arg name="initial_reset"     value="true"/>
    </include>

  </group>
</launch>
````

can somebody help me?

MartyG-RealSense, 2022-03-31:
Hi @sasilva1998  Do you have an Nvidia Jetson like @Likhith-Sugganahalli did, please?  If you do have a Jetson, do the errors disappear if you set **enable_pointcloud** to False instead of true?  There are known issues with pointcloud generation on Jetson boards that do not occur if pointcloud is not enabled.  Further information about this can be found at https://github.com/IntelRealSense/realsense-ros/issues/1967

sasilva1998, 2022-03-31:
Just tried setting `enable_pointcloud=false` however it still did not work. I am using:
````
RealSense ROS v2.3.2
Built with LibRealSense v2.50.0
````

I get this output continously:

````
dex: 768, error: Resource temporarily unavailable, number: 11
 31/03 10:32:04,427 WARNING [546719125888] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/03 10:32:04,578 WARNING [546719125888] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/03 10:32:04,629 WARNING [546719125888] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ WARN] [1648740727.771514290]: Hardware Notification:USB CAM overflow,1.64874e+12,Error,Hardware Error
[ WARN] [1648740729.773241045]: Hardware Notification:USB CAM overflow,1.64874e+12,Error,Hardware Error
[ WARN] [1648740731.774821229]: Hardware Notification:USB CAM overflow,1.64874e+12,Error,Hardware Error
[ WARN] [1648740733.776166566]: Hardware Notification:USB CAM overflow,1.64874e+12,Error,Hardware Error
````
I will try an older version of sdk and ros wrapper which worked for me a year ago.

MartyG-RealSense, 2022-03-31:
Okay, thanks very much @sasilva1998 - I look forward to the results of your test of an older SDK and ROS wrapper.  It has previously been reported by a RealSense ROS Jetson user at https://github.com/IntelRealSense/realsense-ros/issues/1967#issuecomment-954814317 that a combination of SDK **2.43.0** and wrapper **2.2.23** resolved their Jetson pointcloud problem in their particular case.  Good luck!

anujpatel1608, 2022-12-12:
Hi @MartyG-RealSense  I'm getting this error. Here are the commands I ran to install ros wrapper.I'm using D435i.
sudo apt-get install ros-$ROS_DISTRO-realsense2-camera
cd /home/$USER
git clone https://github.com/IntelRealSense/librealsense
cd /home/$USER/librealsense
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules
sudo udevadm trigger

Is this correct?
If no could you please share a link to article which explains installation process for Rpi4 Ubuntu Mate 20?
Also please share link for Jetson Nano

MartyG-RealSense, 2022-12-12:
If you use **sudo apt-get install ros-$ROS_DISTRO-realsense2-camera** then this instruction installs both librealsense and the ROS wrapper from packages at the same time.  There is therefore no need to perform the instructions after this line as you would be installing librealsense a **second** time and risking a conflict between two separate librealsense versions on the same computer.

anujpatel1608, 2022-12-13:
HI @MartyG-RealSense I ran that command and referred another post where fps was being set to 6 and filters:=pointcloud was set along with initial reset. This worked for me.Thanks!

MartyG-RealSense, 2022-12-13:
That's great news that you were successful, @anujpatel1608 - thanks very much for the update  :)