realsenseai/realsense-ros #2151: Orientation of other objects observed from the camera

poornimajd, 2021-11-10:
Hello,
I have a intel realsense d435i.This camera is mounted on a bot(say bot A) and using this depth camera,I want to find the yaw of the other bots which are observed in the fov of the camera,like yaw of the bot B,bot C,etc .Do you have any suggestion for an application like this ,where intel realsense has been used before?

I can give some more details if required.
Thank you.

MartyG-RealSense, 2021-11-10:
Hi @poornimajd  If you wish to analyze the position and rotation of an observed robot (its "pose") then you could attach a 'fiducial marker' image tag called an **Aruco** to the body of the other robot.  

 If you need to use ROS, the link below has an example of a ROS Aruco detection project that has been tested with RealSense D435.

https://github.com/Jphartogi/ipa_marker_detection

poornimajd, 2021-11-10:
Thank you for the quick reply and the source.I will check out the link.

closed 2021-11-10 by poornimajd