realsenseai/realsense-ros #2152: No color topic in ros

zxl5344, 2021-11-10:
HI, I am using a L515 camera. I run **roslaunch realsense2_camera rs_camera.launch** in this repo and I only get topic in depth. 
[](ur
[image]
l)

How do I get all topic?

MartyG-RealSense, 2021-11-10:
Hi @zxl5344  Are you able to receive the color topic if you unplug the USB cable from the base of the camera, turn the micro-sized connector around the opposite way and re-insert it into the base of the camera (USB Type-C cables are two-way insertion on the micro sized end).

zxl5344, 2021-11-10:
No, I still only get topic depth. My camera should work well since I can see RGB and depth when I run **realsense-viewer**.

MartyG-RealSense, 2021-11-11:
In https://github.com/IntelRealSense/realsense-ros/issues/1523 a RealSense ROS user with an L515 had a similar problem with the color topic not being published.  

May I ask whether you have edited the rs_launch file to use a custom resolution for color instead of letting the default color resolution be chosen by the ROS launch? 

You also mentioned in your other L515 case in https://github.com/IntelRealSense/librealsense/issues/9931 that the camera was only being detected as USB 2.1 in the Viewer.  Could you also check whether the camera is being mis-detected as USB 2.1 in the ROS launch, please?  You can do this by finding a line called **Device USB Type** in the log.  If the camera is being detected as 2.1 then it will say '2.1' instead of '3.2'.

[image]

zxl5344, 2021-11-11:
No, I didn't edit it.

Yes, it is being detected as 2.1

zxl5344, 2021-11-11:
I checked my log and I found there is not "color stream is enabled" even "enable_color: True". 

... logging to /home/zxl5344/.ros/log/af44d6bc-42fc-11ec-9209-ef9a2ee492f5/roslaunch-E5-ME-L-KAFI11-177380.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://E5-ME-L-KAFI11:40449/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
[... 74 lines cut ...]
[ INFO] [1636641359.807964066]: L500 Depth Sensor was found.
[ INFO] [1636641359.812690918]: RGB Camera was found.
[ INFO] [1636641359.812780241]: Motion Module was found.
[ INFO] [1636641359.812803235]: num_filters: 0
[ INFO] [1636641359.812817640]: Setting Dynamic reconfig parameters.
[ INFO] [1636641360.023355174]: Done Setting Dynamic reconfig parameters.
[ INFO] [1636641360.023538228]: depth stream is enabled - width: 320, height: 240, fps: 30, Format: Z16
[ INFO] [1636641360.023580293]: setupPublishers...
[ INFO] [1636641360.025604311]: Expected frequency for depth = 30.00000
[ INFO] [1636641360.047690474]: setupStreams...
[ INFO] [1636641360.058582122]: insert Depth to L500 Depth Sensor
[ INFO] [1636641360.075958496]: SELECTED BASE:Depth, 0
[ INFO] [1636641360.076116352]: RealSense Node Is Up!
 11/11 09:36:00,076 WARNING [140081192064768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 11/11 09:36:00,126 WARNING [140081192064768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 11/11 09:36:00,178 WARNING [140081192064768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 11/11 09:36:00,481 WARNING [140081192064768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 11/11 09:36:00,735 WARNING [140081192064768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 11/11 09:36:01,698 WARNING [140081192064768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 11/11 09:36:02,383 WARNING [140081192064768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 11/11 09:36:03,635 WARNING [140081192064768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
^C 11/11 09:36:04,887 WARNING [140081192064768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

MartyG-RealSense, 2021-11-11:
Does the color stream become enabled if you enable it in the roslaunch instruction:

**roslaunch realsense2_camera rs_camera.launch enable_color:=true**

zxl5344, 2021-11-11:
No, "color stream" is still missing

MartyG-RealSense, 2021-11-11:
As you intend to buy a new USB cable as mentioned in https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-966474815 - shall we come back to this question after you have had the opportunity to see whether a cable change makes a difference?  Thanks!

zxl5344, 2021-11-11:
Thank you. I will update it once I got the new data cable

zxl5344, 2021-11-18:
Hi, I tried my new cable.  Although i still have "11:12:14 [Warn] /home/zxl5344/Realsense SDK/librealsense-2.49.0/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 768, error: No data available, number: 61", all camera function looks working well so far.

closed 2021-11-18 by zxl5344

MartyG-RealSense, 2021-11-18:
Excellent news, @zxl5344 - thanks for the update!

If you are receiving the **control_transfer returned** warning only a small number of times during launch and then they stop, or they occur one or two times a minute after launch, then this warning can be safely ignored.  Likewise, if the topics are being published correctly in RViz and performance does not seem to be degraded then the warnings can also be disregarded.

karanpandyaa, 2024-05-19:
I had the same issue, it got solved with: roslaunch realsense2_camera rs_aligned_depth.launch

hesic73, 2024-11-21:
> I had the same issue, it got solved with: **roslaunch realsense2_camera rs_aligned_depth.launch**

Thanks! I'm using a L515, and this works for me!