realsenseai/realsense-ros #2155: How can I get the stereo images with D455?

Turing47, 2021-11-13:
Hi, I use D455 on Ubuntu 18. I open the realsense_viewer and only have left images on the stereo module (including rgbd). 
Also in realsense2_camera, I also fail to find the stereo camera node. What should I do find them. I want to try stereo SLAM algorithms.

terminal:
/camera/accel/imu_info
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/extrinsics/depth_to_color
/camera/gyro/imu_info
/camera/imu
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static

[image]

[image]

MartyG-RealSense, 2021-11-13:
Hi @Turing47   Is your camera plugged into a USB 2 port, please?  If it is in a USB 2 port (or is in a USB 3 port but the connection is identified as USB 2) then Infrared 2 (the right infrared channel) will be unavailable.  this is because Infrared 2 is only supported on a USB 3 connection.

Turing47, 2021-11-13:
> Hi @Turing47 Is your camera plugged into a USB 2 port, please? If it is in a USB 2 port (or is in a USB 3 port but the connection is identified as USB 2) then Infrared 2 (the right infrared channel) will be unavailable. This is because Infrared 2 is only supported on a USB 3 connection.

Thank you. Probably you are right. I am using a USB 2.0 cable. I will buy a new one and reply later.

MartyG-RealSense, 2021-11-13:
Thanks very much, @Turing47 - I look forward to your next report.  Good luck!

Turing47, 2021-11-15:
> Thanks very much, @Turing47 - I look forward to your next report. Good luck!

Thank you. I just got a USB 3.0 cable and it works.

closed 2021-11-15 by Turing47

MartyG-RealSense, 2021-11-15:
Great news - thanks for the update   :)