realsenseai/realsense-ros #2168: Enabling "align_depth" causes FPS to drop, "aligned_depth_to_color" is only around 5 FPS (D435 camera, Jetson NX )

paulacvalle, 2021-11-17:
Hello!

I  have been having an issue with using the align depth feature on my Realsense D435 + Jetson NX (Jetpack 4.5) setup, and also with a Jetson Nano 2GB (same issue). 

When setting "align_depth" to "true", while hoping for 15 FPS for both color and depth, color reduces to 13 FPs, depth to 10 FPS and the "aligned_depth_to_color" to only around 5 FPS.  (I really need the aligned depth to have an FPS close to the color stream).

When setting "align_depth" to "false", both color and depth have a steady 15FPS... What could be the issue? Lack of processing power?  Compatibility with Jetson board?

I tested the same camera and launch file on my Linux PC and it worked fine! 

Any help is welcomed!

MartyG-RealSense, 2021-11-17:
Hi @paulacvalle  Using depth-color alignment has a processing overhead that can cause slowdown.  On Jetson boards you can accelerate pointcloud processing by having **CUDA** support enabled in the librealsense SDK in order to offload the processing from the CPU onto the Nvidia graphics GPU.

If librealsense was built from packages or from source code with the CMake build term **DBUILD_WITH_CUDA=true** included in the CMake build instruction then CUDA support should be included.  However, CUDA support will not be included if the RealSense ROS wrapper was built from packages with **Method 1** which installs librealsense and the wrapper together.

paulacvalle, 2021-11-19:
@MartyG-RealSense  Thank you for your quick reply. I did build librealsense from source and it should have the CUDA support, but since I am still having the issue maybe I did something wrong or it did not include CUDA. 

I will try again and see what can be done, but it is good to know that is an processing issue and could be solved using the CUDA acceleration. Now I know what to look for, thank you!

closed 2021-11-19 by paulacvalle

MartyG-RealSense, 2021-11-19:
That's great to hear, @paulacvalle - thanks very much for the update   :)