realsenseai/realsense-ros #2171: Cannot connect to multiple cameras after updating to v3.2.3

ParthasarathyBana, 2021-11-19:
This issue is related to `D415` cameras and `ros2`. 
Our robot was set up with two D415 cameras and both the cameras were streaming image feed correctly with the following config:
* `realsense_ros` - `v3.2.2`
* `librealsense2` - `v2.49.0`
* `firmware` - `v5.12.15.50`

After updating to the recent versions, as follows, one of the cameras is not detected when we run the following launch script.
* `realsense_ros` - `v3.2.3`
* `librealsense2` - `v2.50.0` 
* `firmware` - `v5.13.00.50`

`ros2 launch realsense2_camera rs_multi_camera_launch.py serial_no_camera1:=135322062929 serial_no_camera2:=135322061686`

Only feed from camera with `serial_no_camera1` is detected and displayed on `rviz2`. 

The following is the output from the terminal.
```
demo@cyndaquil-agent:~$ ros2 launch realsense2_camera rs_multi_camera_launch.py serial_no_camera1:=135322062929 serial_no_camera2:=135322061686
[INFO] [launch]: All log files can be found below /home/demo/.ros/log/2021-11-19-15-52-03-976932-cyndaquil-agent-15711
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [15713]
[INFO] [realsense2_camera_node-2]: process started with pid [15715]
[INFO] [static_transform_publisher-3]: process started with pid [15717]
[static_transform_publisher-3] [WARN] [1637362324.284061171] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-3] [INFO] [1637362324.289728125] [static_transform_publisher_06YUbyfCLQoaMgx6]: Spinning until stopped - publishing transform
[static_transform_publisher-3] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-3] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-3] from 'camera1_link' to 'camera2_link'
[realsense2_camera_node-2] [INFO] [1637362324.511387202] [RealSenseCameraNode]: RealSense ROS v3.2.3
[realsense2_camera_node-2] [INFO] [1637362324.511496198] [RealSenseCameraNode]: Built with LibRealSense v2.50.0
[realsense2_camera_node-2] [INFO] [1637362324.511527115] [RealSenseCameraNode]: Running with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1637362324.511804622] [RealSenseCameraNode]: RealSense ROS v3.2.3
[realsense2_camera_node-1] [INFO] [1637362324.511905031] [RealSenseCameraNode]: Built with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1637362324.511935096] [RealSenseCameraNode]: Running with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1637362324.528548797] [RealSenseCameraNode]: Device with serial number 135322062929 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1637362324.528577015] [RealSenseCameraNode]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1637362324.528581798] [RealSenseCameraNode]: Device with name Intel RealSense D415 was found.
[realsense2_camera_node-1] [INFO] [1637362324.528805570] [RealSenseCameraNode]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1637362324.528812732] [RealSenseCameraNode]: Device USB type: 3.2
[realsense2_camera_node-2] [INFO] [1637362324.529125906] [RealSenseCameraNode]: Device with serial number 135322062929 was found.
[realsense2_camera_node-2] 
[realsense2_camera_node-2] [INFO] [1637362324.529140277] [RealSenseCameraNode]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[realsense2_camera_node-2] [INFO] [1637362324.529146391] [RealSenseCameraNode]: Device with name Intel RealSense D415 was found.
[realsense2_camera_node-2] [INFO] [1637362324.529314912] [RealSenseCameraNode]: Device with port number 2-1 was found.
[realsense2_camera_node-2] [INFO] [1637362324.529328643] [RealSenseCameraNode]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1637362324.530509241] [RealSenseCameraNode]: getParameters...
[realsense2_camera_node-2] [INFO] [1637362324.530897559] [RealSenseCameraNode]: getParameters...
[realsense2_camera_node-1] [INFO] [1637362324.534063152] [RealSenseCameraNode]: setupDevice...
[realsense2_camera_node-1] [INFO] [1637362324.534076435] [RealSenseCameraNode]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1637362324.534089407] [RealSenseCameraNode]: Device Name: Intel RealSense D415
[realsense2_camera_node-1] [INFO] [1637362324.534098702] [RealSenseCameraNode]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1637362324.534107615] [RealSenseCameraNode]: Device FW version: 05.13.00.50
[realsense2_camera_node-1] [INFO] [1637362324.534115626] [RealSenseCameraNode]: Device Product ID: 0x0AD3
[realsense2_camera_node-1] [INFO] [1637362324.534122869] [RealSenseCameraNode]: Enable PointCloud: Off
[realsense2_camera_node-1] [INFO] [1637362324.534129783] [RealSenseCameraNode]: Align Depth: Off
[realsense2_camera_node-1] [INFO] [1637362324.534137050] [RealSenseCameraNode]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1637362324.534155421] [RealSenseCameraNode]: Device Sensors: 
[realsense2_camera_node-2] [INFO] [1637362324.534532348] [RealSenseCameraNode]: setupDevice...
[realsense2_camera_node-2] [INFO] [1637362324.534544266] [RealSenseCameraNode]: JSON file is not provided
[realsense2_camera_node-2] [INFO] [1637362324.534552630] [RealSenseCameraNode]: Device Name: Intel RealSense D415
[realsense2_camera_node-2] [INFO] [1637362324.534561872] [RealSenseCameraNode]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0
[realsense2_camera_node-2] [INFO] [1637362324.534570704] [RealSenseCameraNode]: Device FW version: 05.13.00.50
[realsense2_camera_node-2] [INFO] [1637362324.534579064] [RealSenseCameraNode]: Device Product ID: 0x0AD3
[realsense2_camera_node-2] [INFO] [1637362324.534586872] [RealSenseCameraNode]: Enable PointCloud: Off
[realsense2_camera_node-2] [INFO] [1637362324.534594552] [RealSenseCameraNode]: Align Depth: Off
[realsense2_camera_node-2] [INFO] [1637362324.534601860] [RealSenseCameraNode]: Sync Mode: Off
[realsense2_camera_node-2] [INFO] [1637362324.534615519] [RealSenseCameraNode]: Device Sensors: 
[realsense2_camera_node-1]  19/11 15:52:04,534 WARNING [140216002123520] (backend-v4l2.cpp:1387) Pixel format 20303157-0000-0010-8000-00aa003 likely requires patch for fourcc code W10 !
[realsense2_camera_node-2]  19/11 15:52:04,534 WARNING [140428754007808] (backend-v4l2.cpp:1387) Pixel format 20303157-0000-0010-8000-00aa003 likely requires patch for fourcc code W10 !
[... 184 lines cut ...]
[realsense2_camera_node-1]  19/11 15:52:07,101 WARNING [140216010516224] (backend-v4l2.cpp:1387) Pixel format 20303157-0000-0010-8000-00aa003 likely requires patch for fourcc code W10 !
[realsense2_camera_node-1] [INFO] [1637362327.111857752] [RealSenseCameraNode]: Stereo Module was found.
[realsense2_camera_node-1]  19/11 15:52:07,111 WARNING [140216010516224] (backend-v4l2.cpp:1387) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[realsense2_camera_node-1] [INFO] [1637362327.120403312] [RealSenseCameraNode]: RGB Camera was found.
[realsense2_camera_node-1] [INFO] [1637362327.120417042] [RealSenseCameraNode]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1637362327.120421968] [RealSenseCameraNode]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1637362327.120425562] [RealSenseCameraNode]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1637362327.120429106] [RealSenseCameraNode]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1637362327.120433104] [RealSenseCameraNode]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1637362327.120438131] [RealSenseCameraNode]: num_filters: 0
[realsense2_camera_node-1] [INFO] [1637362327.120442119] [RealSenseCameraNode]: Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [WARN] [1637362327.132160071] [RealSenseCameraNode]: Could not set param: rgb_camera.power_line_frequency with 3Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.
[... cut ...]
[realsense2_camera_node-1]  19/11 15:52:07,564 ERROR [140215490430720] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
[realsense2_camera_node-1]  19/11 15:52:07,564 ERROR [140215490430720] (librealsense-exception.h:52) xioctl(VIDIOC_REQBUFS) failed Last Error: No such device
[realsense2_camera_node-1]  19/11 15:52:07,564 ERROR [140215490430720] (sensor.cpp:596) release_power failed: xioctl(VIDIOC_REQBUFS) failed Last Error: No such device

```

MartyG-RealSense, 2021-11-20:
Hi @ParthasarathyBana  Is the second camera detectable if the two cameras are launched in separate ROS terminals instead of with the multiple camera ROS2 launch file?

**Terminal 1**
ros2 launch realsense2_camera rs_launch.py camera_name:=my_d415 serial_no:=135322062929 

**Terminal 2**
ros2 launch realsense2_camera rs_launch.py camera_name:=my_d415 serial_no:=135322061686

ParthasarathyBana, 2021-11-20:
When launched in separate terminals , it looks like both the cameras are detected. However, one of the cameras produces a hardware stream error as shown below:
'''
demo@cyndaquil-agent:~$ ros2 launch realsense2_camera rs_launch.py camera_name:=my_d415 serial_no:=_135322061686
[INFO] [launch]: All log files can be found below /home/demo/.ros/log/2021-11-20-12-01-55-842116-cyndaquil-agent-24653
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [24655]
[realsense2_camera_node-1] [INFO] [1637434916.222196583] [RealSenseCameraNode]: RealSense ROS v3.2.3
[realsense2_camera_node-1] [INFO] [1637434916.222309815] [RealSenseCameraNode]: Built with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1637434916.222344611] [RealSenseCameraNode]: Running with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1637434916.237751754] [RealSenseCameraNode]: Device with serial number 135322062929 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1637434916.237782624] [RealSenseCameraNode]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video12 was found.
[realsense2_camera_node-1] [INFO] [1637434916.237788205] [RealSenseCameraNode]: Device with name Intel RealSense D415 was found.
[realsense2_camera_node-1] [INFO] [1637434916.238035431] [RealSenseCameraNode]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1637434916.352834012] [RealSenseCameraNode]: Device with serial number 135322061686 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1637434916.352902727] [RealSenseCameraNode]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/video4linux/video6 was found.
[realsense2_camera_node-1] [INFO] [1637434916.352924995] [RealSenseCameraNode]: Device with name Intel RealSense D415 was found.
[realsense2_camera_node-1] [INFO] [1637434916.353602632] [RealSenseCameraNode]: Device with port number 2-4 was found.
[realsense2_camera_node-1] [INFO] [1637434916.353638222] [RealSenseCameraNode]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1637434916.359121424] [RealSenseCameraNode]: getParameters...
[realsense2_camera_node-1] [INFO] [1637434916.364049983] [RealSenseCameraNode]: setupDevice...
[realsense2_camera_node-1] [INFO] [1637434916.364086858] [RealSenseCameraNode]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1637434916.364109114] [RealSenseCameraNode]: Device Name: Intel RealSense D415
[realsense2_camera_node-1] [INFO] [1637434916.364127959] [RealSenseCameraNode]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/video4linux/video6
[realsense2_camera_node-1] [INFO] [1637434916.364149763] [RealSenseCameraNode]: Device FW version: 05.13.00.50
[realsense2_camera_node-1] [INFO] [1637434916.364170300] [RealSenseCameraNode]: Device Product ID: 0x0AD3
[realsense2_camera_node-1] [INFO] [1637434916.364189197] [RealSenseCameraNode]: Enable PointCloud: Off
[realsense2_camera_node-1] [INFO] [1637434916.364206502] [RealSenseCameraNode]: Align Depth: Off
[realsense2_camera_node-1] [INFO] [1637434916.364224308] [RealSenseCameraNode]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1637434916.364248169] [RealSenseCameraNode]: Device Sensors: 
[realsense2_camera_node-1]  20/11 12:01:56,365 WARNING [139934732121856] (backend-v4l2.cpp:1387) Pixel format 20303157-0000-0010-8000-00aa003 likely requires patch for fourcc code W10 !
[realsense2_camera_node-1] [INFO] [1637434916.380052025] [RealSenseCameraNode]: Stereo Module was found.
[realsense2_camera_node-1]  20/11 12:01:56,380 WARNING [139934732121856] (backend-v4l2.cpp:1387) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[realsense2_camera_node-1] [INFO] [1637434916.388367249] [RealSenseCameraNode]: RGB Camera was found.
[realsense2_camera_node-1] [INFO] [1637434916.388387995] [RealSenseCameraNode]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1637434916.388395873] [RealSenseCameraNode]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1637434916.388401916] [RealSenseCameraNode]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1637434916.388408555] [RealSenseCameraNode]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1637434916.388416512] [RealSenseCameraNode]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1637434916.388425366] [RealSenseCameraNode]: num_filters: 0
[realsense2_camera_node-1] [INFO] [1637434916.388431804] [RealSenseCameraNode]: Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [WARN] [1637434916.398964557] [RealSenseCameraNode]: Could not set param: rgb_camera.power_line_frequency with 3Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1637434916.399109804] [RealSenseCameraNode]: Done Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [INFO] [1637434916.399861004] [RealSenseCameraNode]: depth stream is enabled - width: 1280, height: 720, fps: 30, Format: Z16
[... cut ...]
[realsense2_camera_node-1] [INFO] [1637434916.438943409] [RealSenseCameraNode]: Device Serial No: 135322061686
[realsense2_camera_node-1] [INFO] [1637434916.438993542] [RealSenseCameraNode]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1637434917.980218602] [RealSenseCameraNode]: 
[realsense2_camera_node-1] [WARN] [1637434918.400257582] [RealSenseCameraNode]: Hardware Notification:Depth stream start failure,1.63743e+12,Error,Hardware Error
'''
Image msgs are seen via 'ros2 topic echo /my_d415/color/image_raw' but cannot be visualized on 'rviz2'

MartyG-RealSense, 2021-11-20:
Is the affected camera able to successfully stream depth in the RealSense Viewer tool?

Does the depth stream start failure still occur if you add **initial_reset:=true** to the roslaunch instruction of the affected camera to reset it at launch?

**ros2 launch realsense2_camera rs_launch.py camera_name:=my_d415 serial_no:=135322062929 initial_reset:=true**

ParthasarathyBana, 2021-11-22:
Yes, I am able to see the color and depth feeds from both cameras using the `realsense-viewer`.

As for the `initial_reset:=true`, that does not work. It produces the same hardware stream error.

MartyG-RealSense, 2021-11-23:
Which ROS2 version are you using, please?  (Foxy, Galactic, etc)

If you are using Galactic, advice relating to the **Depth stream start failure** hardware error is offered in https://github.com/IntelRealSense/librealsense/issues/9829#issuecomment-936146158

In that same discussion in https://github.com/IntelRealSense/librealsense/issues/9829#issuecomment-942878846 the RealSense ROS user in that case shared the details of what enabled them to view the image in RViz2.

ParthasarathyBana, 2021-11-23:
We are using `ros2-rolling` and I followed the discussion about this issue in #2068. We are using a frame rate = 6Hz and the cameras are plugged into USB 3.2 ports. We have `realsense_ros` installed from source and `librealsense2` installed from debian. 

We previously had both cameras working using the package specs as mentioned above [#2171 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/2171#issue-1058985739). Therefore, we rolled back to the previous specs and it is now working. 

I am not sure why this does not work for `ros2-rolling`, `realsense_ros = 3.2.3`, `librealsense2 = 2.50` and `firmware = 5.13.00`. 

For now we will continue with the previous versions and look forward to getting an update in the future that would solve this issue. Besides, I can only help you with testing this on our systems and don't think we can help in developing support for `ros2-rolling` If you would like to close this issue, feel free to close it.

MartyG-RealSense, 2021-11-23:
The ROS2 wrapper - including 3.2.3 - already supports Rolling.  I will close this case though as you suggest, but please feel free to return to the subject on tis support forum at a future date if you need to.  Good luck!

closed 2021-11-23 by MartyG-RealSense

ParthasarathyBana, 2021-11-23:
Thanks for your help! I just wanted to highlight that you say it is working for `ros2-rolling` and `realsense_ros v3.2.3` but this case is just telling you that it is broken and it needs a fix. I will definitely get back on the forum in the future.

doronhi, 2021-11-24:
I am not aware of any changes in the last release to introduce these errors. 
Regarding the running of multiple cameras, you had the wrong command line.
You should run: `ros2 launch realsense2_camera rs_multi_camera_launch.py serial_no1:=_135322062929 serial_no2:=_135322061686`
Not `ros2 launch realsense2_camera rs_multi_camera_launch.py serial_no_camera1:=135322062929 serial_no_camera2:=135322061686`

I am not sure what's wrong with watching with RViz but I suggest checking that the "Fixed Frame" is set to something valid like "camera1_link" maybe. Alternatively, you can test using rqt_image_view.