realsenseai/realsense-ros #2173: rosdep install errors: "Ignoring environment", "cannot locate..."

JHPHELAN, 2021-11-21:
@doronhi @MartyG-RealSense 
Having prev. installed the latest Librealsense 2.50 successfully from source via:
[https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md](url)
and with successful realsense-viewer
I'm attempting to install the ROS_wrapper working from:
[https://github.com/IntelRealSense/realsense-ros/tree/ros2](url)
I get to Step 4: Install dependencies:
`rosdep install -i --from-path src --rosdistro $_ros_dist --skip-keys=librealsense2 -y`
then get these errors:

> WARNING: given --rosdistro --skip-keys=librealsense2 but ROS_DISTRO is "foxy". Ignoring environment.
> WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3.  
> ERROR: the following packages/stacks could not have their rosdep keys resolved
> to system dependencies:
> realsense2_description: Cannot locate rosdep definition for [xacro]
> realsense2_camera: Cannot locate rosdep definition for [diagnostic_updater]

What's happening here?
Jim P.
[2021.11.21 rosdep install FAIL .pdf](https://github.com/IntelRealSense/realsense-ros/files/7577380/2021.11.21.rosdep.install.FAIL.pdf)

doronhi, 2021-11-22:
It seems like a bug in the documentation. Please try replace `$_ros_dist` with `$ROS_DISTRO`. i.e.
Try out the following command: `rosdep install -i --from-path src --rosdistro $ROS_DISTRO --skip-keys=librealsense2 -y`

MartyG-RealSense, 2021-11-22:
@doronhi   Thanks very much for your help for @JHPHELAN   :)

JHPHELAN, 2021-11-23:
@doronhi @MartyG-RealSense 
I made the correction above and install progressed ok.
We may  have finally stumbled on a workable path to:
Intel RealSense Depth Camera D455 / ROS2_Wrapper / RealSense 2.50 via libuvc / ROS2 foxy / Ubuntu 20.04 / Raspberry Pi 4b 4G
There are still some issues, see attached, but this is real progress!
Thank you both!
Jim P
[2021.11.22 ROS_Wrapper success.pdf](https://github.com/IntelRealSense/realsense-ros/files/7585624/2021.11.22.ROS_Wrapper.success.pdf)

MartyG-RealSense, 2021-11-23:
Excellent news @JHPHELAN - thanks very much for the update   :)

MartyG-RealSense, 2021-11-29:
Hi @JHPHELAN  Do you require further assistance with this case, please?  Thanks!

JHPHELAN, 2021-11-29:
@MartyG-RealSense 
Not on this, it's good.  Thanks!
JHP

MartyG-RealSense, 2021-11-30:
Thanks very much @JHPHELAN - as you are satisfied with the outcome, I will close this case.  Thanks again!

closed 2021-11-30 by MartyG-RealSense

lifangyan16, 2021-12-28:
> @doronhi @MartyG-RealSense Having prev. installed the latest Librealsense 2.50 successfully from source via: [...]

use ros instead of ros2!!!

JHPHELAN, 2021-12-29:
@lifangyan16 
I don't think using ROS for a ROS2 environment will work.
I believe I got it a bit further along, although never perfectly.  See attached.
[2021.11.22 ROS_Wrapper success.pdf](https://github.com/IntelRealSense/realsense-ros/files/7787111/2021.11.22.ROS_Wrapper.success.pdf)