realsenseai/realsense-ros #2174: realsense d455 unsupported device!

xixi0226, 2021-11-22:
There is a problem with realsense d455 on Jetson TX2 .Realsense d455 on the Realsense_viewer is OK,but realsense-ros bag is not be launched  .How can I solve this problem? Thanks!

`* /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet

auto-starting new master
process[master]: started with pid [9312]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 935f9834-4b75-11ec-982d-48b02d0f9292
process[rosout-1]: started with pid [9323]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [9330]
process[camera/realsense2_camera-3]: started with pid [9331]
[ INFO] [1637572892.316092784]: Initializing nodelet with 6 worker threads.
[ INFO] [1637572892.492312377]: RealSense ROS v2.2.14
[ INFO] [1637572892.492432151]: Running with LibRealSense v2.35.2
[ INFO] [1637572892.591953950]:  
[ INFO] [1637572892.991110798]: Device with serial number 053122250619 was found.

[ INFO] [1637572892.991197389]: Device with physical ID 2-3.1-7 was found.
[ INFO] [1637572892.991255756]: Device with name Intel RealSense D455 was found.
[ INFO] [1637572892.992943317]: Device with port number 2-3.1 was found.
[FATAL] [1637572892.993142610]: **Unsupported device! Product ID: 0x0B5C**
[camera/realsense2_camera_manager-2] process has died [pid 9330, exit code 1, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/kumar/.ros/log/935f9834-4b75-11ec-982d-48b02d0f9292/camera-realsense2_camera_manager-2.log].
log file: /home/kumar/.ros/log/935f9834-4b75-11ec-982d-48b02d0f9292/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has finished cleanly
log file: /home/kumar/.ros/log/935f9834-4b75-11ec-982d-48b02d0f9292/camera-realsense2_camera-3*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

MartyG-RealSense, 2021-11-22:
Hi @xixi0226  Support for the D455 camera model was not added in the RealSense ROS wrapper until wrapper version **2.2.16**, which matches to librealsense SDK 2.37.0.

xixi0226, 2021-11-22:
ok,I will follow your suggestion , but I also found that realsense d435i can not be launched on the same version of RealSense ROS wrapper and librealsense SDK. How can I solve this problem? .Thanks!
`[ INFO] [1637580822.031550656]: Initializing nodelet with 6 worker threads.
[ INFO] [1637580822.233846272]: RealSense ROS v2.2.14
[ INFO] [1637580822.233931040]: Running with LibRealSense v2.35.2
[ INFO] [1637580822.267211424]:  
[ INFO] [1637580822.539069920]: Device with serial number 043322075379 was found.

[ INFO] [1637580822.539216608]: Device with physical ID 2-1.1-3 was found.
[ INFO] [1637580822.539264960]: Device with name Intel RealSense D435I was found.
[ INFO] [1637580822.540316928]: Device with port number 2-1.1 was found.
[ INFO] [1637580822.544881568]: getParameters...
[ INFO] [1637580822.668450720]: setupDevice...
[ INFO] [1637580822.668575136]: JSON file is not provided
[ INFO] [1637580822.668657408]: ROS Node Namespace: camera
[ INFO] [1637580822.668759456]: Device Name: Intel RealSense D435I
[ INFO] [1637580822.668833056]: Device Serial No: 043322075379
[ INFO] [1637580822.668887104]: Device physical port: 2-1.1-3
[ INFO] [1637580822.668975072]: Device FW version: 05.12.14.50
[ INFO] [1637580822.669033216]: Device Product ID: 0x0B3A
[ INFO] [1637580822.669137472]: Enable PointCloud: Off
[ INFO] [1637580822.669211200]: Align Depth: Off
[ INFO] [1637580822.669265824]: Sync Mode: Off
[ INFO] [1637580822.669472864]: Device Sensors: 
[ INFO] [1637580822.718049024]: Stereo Module was found.
[ INFO] [1637580822.754469760]: RGB Camera was found.
[ INFO] [1637580822.754937984]: Motion Module was found.
[ INFO] [1637580822.755069696]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1637580822.755149376]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1637580822.755226720]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1637580822.755359424]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1637580822.755476864]: num_filters: 0
[ INFO] [1637580822.755556672]: Setting Dynamic reconfig parameters.
 22/11 19:33:43,229 WARNING [547579650432] (types.cpp:49) hwmon command 0x7d failed.
Error type: No data to return (-21).
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  hwmon command 0x7d failed.
Error type: No data to return (-21).
[camera/realsense2_camera_manager-2] process has died [pid 8127, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/kumar/.ros/log/09aac632-4b88-11ec-9a0d-48b02d0f9292/camera-realsense2_camera_manager-2.log].
log file: /home/kumar/.ros/log/09aac632-4b88-11ec-9a0d-48b02d0f9292/camera-realsense2_camera_manager-2*.log
^C[camera/realsense2_camera-3] killing on exit
[ WARN] [1637580831.217336064]: Couldn't find service realsense2_camera_manager/unload_nodelet, perhaps the manager is already shut down
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

MartyG-RealSense, 2021-11-22:
The firmware driver version that you are using requires a minimum of SDK 2.39.0 due to internal firmware changes introduced from firmware 5.12.8.200 onwards.  For SDK versions older than 2.39.0, the older firmware **5.12.7.100** is recommended.

closed 2021-12-05 by MartyG-RealSense