realsenseai/realsense-ros #2176: roslaunch crashes

alex-ssom, 2021-11-22:
Hi,

I'm having an issue with realsense-ros. I've built the librealsense v2.50.0 and RealSense ROS v2.3.2

I'm having this issue on a Jetson Nano
```bash
 - NVIDIA Jetson Nano (Developer Kit Version)
   * Jetpack 4.6 [L4T 32.6.1]
   * NV Power Mode: MAXN - Type: 0
   * jetson_stats.service: active
 - Board info:
   * Type: Nano (Developer Kit Version)
   * SOC Family: tegra210 - ID:33
   * Module: P3448-0000 - Board: P3449-0000
   * Code Name: porg
   * Boardids: 3448
   * CUDA GPU architecture (ARCH_BIN): 5.3
   * Serial Number: 1422919050898
 - Libraries:
   * CUDA: 10.2.300
   * cuDNN: 8.2.1.32
   * TensorRT: 8.0.1.6
   * Visionworks: 1.6.0.501
   * OpenCV: 4.5.4 compiled CUDA: YES
   * VPI: ii libnvvpi1 1.1.12 arm64 NVIDIA Vision Programming Interface library
   * Vulkan: 1.2.70
 - jetson-stats:
   * Version 3.1.1
   * Works on Python 2.7.17
```

When I try to launch `roslaunch realsense2_camera rs_camera.launch` I get a linked library error.:
```bash
[ INFO] [1637599595.802010830]: RealSense ROS v2.3.2
[ INFO] [1637599595.802103384]: Built with LibRealSense v2.50.0
[ INFO] [1637599595.802162917]: Running with LibRealSense v2.50.0
[ INFO] [1637599595.921690008]:  
[ INFO] [1637599595.942851116]: Device with serial number 947522072275 was found.
[ INFO] [1637599595.943414668]: Device with physical ID /sys/devices/70090000.xusb/usb2/2-1/2-1.2/2-1.2:1.0/video4linux/video1 was found.
[ INFO] [1637599595.943776029]: Device with name Intel RealSense D435I was found.
```
```bash
[ INFO] [1637599596.350533943]: Done Setting Dynamic reconfig parameters.
/opt/ros/melodic/lib/nodelet/nodelet: symbol lookup error: /home/jetbot/Code/realsense_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN2cv3MatC1Ev
[camera/realsense2_camera_manager-2] process has died [pid 13040, exit code 127, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/jetbot/.ros/log/bfae4564-4bb3-11ec-a8b5-00044be6f694/camera-realsense2_camera_manager-2.log].
log file: /home/jetbot/.ros/log/bfae4564-4bb3-11ec-a8b5-00044be6f694/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has finished cleanly
```

The `.so` file is located at `/home/jetbot/Code/realsense_ws/devel/lib/librealsense2_camera.so`. I suspect that the issue is caused by the double `//` but I don't know how to fix it.

MartyG-RealSense, 2021-11-22:
Hi @alex-ssom  There was an occurrence of the **_ZN2cv3MatC1Ev** error in https://github.com/IntelRealSense/realsense-ros/issues/2078#issuecomment-924877996

In that case, the RealSense ROS user who experienced the error resolved the problem with a combination of **sudo apt-get install ros-melodic-realsense2-camera** and a clean catkin_make of the workspace.

alex-ssom, 2021-11-22:
Hi, 

What you're suggesting isn't going to fix my issue. I don't wan't to install the pre-compiled version of the package because I wan't `librealsense2` to use the jetson nano GPU. If i install **ros-melodic-realsense2-camera** my processor is overloaded, the GPU isn't used and the framerate drops. I can use `realsense-viewer` and get 30 fps for both depth and color. Therefore, the issue isn't with `librealsense2` it with the ros wrapper.

MartyG-RealSense, 2021-11-22:
I'm inclined to think that it was more likely the clean catkin make that resolved the problem than installing the SDK and wrapper together from packages with **Method 1**.

What method are you using to build the librealsense SDK on Nano, please?

alex-ssom, 2021-11-22:
I used the [building-from-source-using-native-backend](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend) method. I followed every step and validated the install by running `realsense-viewer` and updating the camera firmware. 

I then installed `realsense-ros` by following these instructions: [step-2-install-intel-realsense-ros-from-sources](https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources)

MartyG-RealSense, 2021-11-22:
Did you include **DBUILD_WITH_CUDA=true** in the CMake build instruction when building the SDK from source to enable CUDA support?

Has dynamic_reconfigure been installed before you performed **catkin_init_workspace**.  If it has not yet been installed, an easy way to install it on Melodic is with the method below.

```
sudo apt-get update
sudo apt-get install ros-melodic-ddynamic-reconfigure
```

alex-ssom, 2021-11-22:
Yes `DBUILD_WITH_CUDA=true`
Yes `ros-melodic-ddynamic-reconfigure` was installed before compiling

`catkin_init_workspace` is not necessary since calling `catkin_make` already does that. http://wiki.ros.org/catkin/commands/catkin_make

MartyG-RealSense, 2021-11-22:
Yes. I know about that catkin_make documentation.  I would recommend performing the **catkin_init_workspace**  command anyway even if it seems like a redundant action just for the sake of confirming that the instructions have been followed to the letter.

alex-ssom, 2021-11-22:
I tried it, but there's no improvement. From what I can figure, the error might be caused by multiple OpenCV version installed side by side.

alex-ssom, 2021-11-22:
Is the ROS wrapper compatible with OpenCV4?

MartyG-RealSense, 2021-11-22:
In https://github.com/IntelRealSense/realsense-ros/issues/1772 a RealSense ROS user with a Nano and Melodic describes the changes they made to get their system working with OpenCV4.

alex-ssom, 2021-11-23:
I retried using the latest stock image of the Jetson Nano. I was able to compile with success. 

From all the tests I ran, it looks like the issue only happens when OpenCV is compiled with CUDA support.

```bash
 - NVIDIA Jetson Nano (Developer Kit Version)
   * Jetpack 4.6 [L4T 32.6.1]
   * NV Power Mode: MAXN - Type: 0
   * jetson_stats.service: active
 - Board info:
   * Type: Nano (Developer Kit Version)
   * SOC Family: tegra210 - ID:33
   * Module: P3448-0000 - Board: P3449-0000
   * Code Name: porg
   * Boardids: 3448
   * CUDA GPU architecture (ARCH_BIN): 5.3
   * Serial Number: 1422919050898
 - Libraries:
   * CUDA: 10.2.300
   * cuDNN: 8.2.1.32
   * TensorRT: 8.0.1.6
   * Visionworks: 1.6.0.501
   * OpenCV: 4.1.1 compiled CUDA: NO
   * VPI: ii libnvvpi1 1.1.12 arm64 NVIDIA Vision Programming Interface library
   * Vulkan: 1.2.70
 - jetson-stats:
   * Version 3.1.1
   * Works on Python 2.7.17
```

MartyG-RealSense, 2021-11-23:
It's excellent to hear that you made progress.  How does the framerate of the RealSense ROS wrapper perform now on your Nano?

alex-ssom, 2021-11-23:
I get 26 fps for both `/camera/color/image_raw/compressed` and `/camera/aligned_depth_to_color/image_raw/compressedDepth` using the RVL algorithm for the compressedDepth image_transport. That's over a wired ethernet connection. I had to compile from source [image_transport_plugins/noetic-devel](https://github.com/ros-perception/image_transport_plugins) since RVL isn't included in the pre-compiled melodic package. This is a fast enough framerate for our application.

MartyG-RealSense, 2021-11-23:
Great, thanks very much @alex-ssom for the confirmation that you achieved a satisfactory level of performance!

MartyG-RealSense, 2021-11-29:
Case closed due to satisfactory performance achieved and no further comments received.

closed 2021-11-29 by MartyG-RealSense

Camilochiang, 2022-04-21:
Hei @alex-ssom , sorry for bother but I have a couple of questions for you.
I compile a version of opencv from source without cuda support with:
```
cmake -D CMAKE_BUILD_TYPE=RELEASE \
-D CMAKE_INSTALL_PREFIX=/usr/local \
-D PYTHON_EXECUTABLE=/usr/bin/python3.6 \
-D PYTHON_DEFAULT_EXECUTABLE=$(which python3) \
-D WITH_CUDA=OFF ..
```

 what can be confirmed here: 
```
aspen@aspen-tool:~/ASPEN/Software/ROS$ python3 -c 'import cv2; print(cv2); print(cv2.getBuildInformation())'
<module 'cv2' from '/usr/local/lib/python3.6/dist-packages/cv2/python-3.6/cv2.cpython-36m-aarch64-linux-gnu.so'>

General configuration for OpenCV 3.4.16 =====================================
  Version control:               unknown

  Platform:
    Timestamp:                   2022-04-20T11:15:27Z
    Host:                        Linux 4.9.253-tegra aarch64
    CMake:                       3.23.0
    CMake generator:             Unix Makefiles
    CMake build tool:            /usr/bin/make
    Configuration:               RELEASE

  CPU/HW features:
    Baseline:                    NEON FP16

  C/C++:
    Built as dynamic libs?:      YES
    C++11:                       YES
    C++ Compiler:                /usr/bin/c++  (ver 7.5.0)
    C++ flags (Release):         -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Wsuggest-override -Wno-delete-non-virtual-dtor -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -fvisibility-inlines-hidden -O3 -DNDEBUG  -DNDEBUG
    C++ flags (Debug):           -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Wsuggest-override -Wno-delete-non-virtual-dtor -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -fvisibility-inlines-hidden -g  -O0 -DDEBUG -D_DEBUG
    C Compiler:                  /usr/bin/cc
    C flags (Release):           -fsigned-char -W -Wall -Werror=return-type -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -O3 -DNDEBUG  -DNDEBUG
    C flags (Debug):             -fsigned-char -W -Wall -Werror=return-type -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -g  -O0 -DDEBUG -D_DEBUG
    Linker flags (Release):      -Wl,--gc-sections -Wl,--as-needed  
    Linker flags (Debug):        -Wl,--gc-sections -Wl,--as-needed  
    ccache:                      NO
    Precompiled headers:         NO
    Extra dependencies:          dl m pthread rt
    3rdparty dependencies:

  OpenCV modules:
    To be built:                 calib3d core dnn features2d flann highgui imgcodecs imgproc ml objdetect photo python2 python3 shape stitching superres ts video videoio videostab viz
    Disabled:                    world
    Disabled by dependency:      -
    Unavailable:                 cudaarithm cudabgsegm cudacodec cudafeatures2d cudafilters cudaimgproc cudalegacy cudaobjdetect cudaoptflow cudastereo cudawarping cudev java
    Applications:                tests perf_tests apps
    Documentation:               NO
    Non-free algorithms:         NO

  GUI: 
    GTK+:                        YES (ver 2.24.32)
      GThread :                  YES (ver 2.56.4)
      GtkGlExt:                  NO
    VTK support:                 YES (ver 6.3.0)

  Media I/O: 
    ZLib:                        /usr/lib/aarch64-linux-gnu/libz.so (ver 1.2.11)
    JPEG:                        /usr/lib/aarch64-linux-gnu/libjpeg.so (ver 80)
    WEBP:                        /usr/lib/aarch64-linux-gnu/libwebp.so (ver encoder: 0x020e)
    PNG:                         /usr/lib/aarch64-linux-gnu/libpng.so (ver 1.6.34)
    TIFF:                        /usr/lib/aarch64-linux-gnu/libtiff.so (ver 42 / 4.0.9)
    JPEG 2000:                   build (ver 1.900.1)
    OpenEXR:                     build (ver 2.3.0)
    HDR:                         YES
    SUNRASTER:                   YES
[... 61 lines cut ...]
```
[ INFO] [1650519391.228970362]: RealSense ROS v2.3.2
[ INFO] [1650519391.229146712]: Built with LibRealSense v2.50.0
[ INFO] [1650519391.229402358]: Running with LibRealSense v2.50.0
[ INFO] [1650519391.613187332]: Device with name Intel RealSense D415 was found.
[ INFO] [1650519391.968558037]: Device FW version: 05.13.00.50
....
Package: nvidia-jetpack
Version: 4.6-b197
Architecture: arm64
Maintainer: NVIDIA Corporation
Installed-Size: 194
[... cut ...] n-multimedia-api (>> 32.6-0), nvidia-l4t-jetson-multimedia-api (<< 32.7-0)
Homepage: http://developer.nvidia.com/jetson
Priority: standard
Section: metapackages
Filename: pool/main/n/nvidia-jetpack/nvidia-jetpack_4.6-b197_arm64.deb
Size: 29372
SHA256: acec83ad0c1ef05caf9b8ccc6a975c4fb2a7f7830cbe63bbcf7b196a6c1f350e
SHA1: 3e11456cf0ec6b3a40d81b80ca1e14cebafa65ff
MD5sum: 72b2b7b280793bd4abdabe0d38b08535
Description: NVIDIA Jetpack Meta Package
Description-md5: ad1462289bdbc54909ae109d1d32c0a8

```