realsenseai/realsense-ros #2179: I have  running with a different librealsense version  [ WARN] [1638100292.566477829]: than the one the wrapper was compiled with!   error please solve this :(

BamBamBap, 2021-11-28:
My spec is below text

OS : ubuntu 18.08 
CPU : Intel® Core™ i5-9400F CPU @ 2.90GHz × 6 
GPU : GTX 1050 
Camera : realsense L515
USB : 3.2 

Two warnings appear when the code is executed. Please solve it.

Below is the full code

kim@kim:~$ roslaunch realsense2_camera rs_camera.launch 
... logging to /home/kim/.ros/log/8762b40a-5041-11ec-bcb2-40b076dd2030/roslaunch-kim-28393.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://kim:41495/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
[... 84 lines cut ...]
[ INFO] [1638100292.992031791]: Sync Mode: Off
[ INFO] [1638100292.992069633]: Device Sensors: 
[ INFO] [1638100292.992498855]: L500 Depth Sensor was found.
[ INFO] [1638100292.994329362]: RGB Camera was found.
[ INFO] [1638100292.994455687]: Motion Module was found.
[ INFO] [1638100292.994484483]: num_filters: 0
[ INFO] [1638100292.994503169]: Setting Dynamic reconfig parameters.
[ INFO] [1638100293.125993687]: Done Setting Dynamic reconfig parameters.
[ INFO] [1638100293.126720811]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1638100293.128958697]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1638100293.129012897]: setupPublishers...
[ INFO] [1638100293.132673219]: Expected frequency for depth = 30.00000
[ INFO] [1638100293.179301007]: Expected frequency for color = 30.00000
[ INFO] [1638100293.210955546]: setupStreams...
[ INFO] [1638100293.284895868]: SELECTED BASE:Depth, 0
[ INFO] [1638100293.482309817]: RealSense Node Is Up!
[ WARN] [1638100293.487623468]: 
**[ WARN] [1638100293.487746927]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.**
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
^C^C^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

MartyG-RealSense, 2021-11-28:
Hi @BamBamBap   When you change your librealsense SDK version and you have built the RealSense ROS wrapper from source code, you must then build the wrapper again after librealsense has been updated.  Rebuilding wrapper 2.3.2 should make the warning disappear, since that wrapper version is the correct match for librealsense 2.50.0.

BamBamBap, 2021-11-28:
Yes I solved! but...
How do I fix this error? 

"frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically."

can you solve it?

MartyG-RealSense, 2021-11-28:
This warning can occur because the **global timestamp** is disabled by default on the L515 camera model.  **Doronhi** the RealSense ROS wrapper developer offers code in https://github.com/IntelRealSense/realsense-ros/issues/1607#issuecomment-760128960 that you can add to the launch file to enable global time.

Alternatively, you could try enabling global time by adding the command **global_time_enabled:=true** to your roslaunch instruction.  For example:

**roslaunch realsense2_camera rs_camera.launch global_time_enabled:=true**

BamBamBap, 2021-11-28:
roslaunch realsense2_camera rs_camera.launch global_time_enabled:=true
Unfortunately this solution didn't work for me..
I'll try #1607

BamBamBap, 2021-11-29:
I tried #1607.
Error is remove but can't not open camera..
Below is my code

kim@kim:~$ roslaunch realsense2_camera rs_camera.launch 
... logging to /home/kim/.ros/log/b54683b8-50ca-11ec-900f-40b076dd2030/roslaunch-kim-10202.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://kim:34945/

SUMMARY
========

PARAMETERS
 * /camera/l500_depth_sensor/global_time_enabled: True
 * /camera/motion_module/global_time_enabled: True
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
[... 69 lines cut ...]
[ INFO] [1638159211.049708077]: Device Serial No: f0220517
[ INFO] [1638159211.049724532]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-5/2-5:1.0/video4linux/video2
[ INFO] [1638159211.049744274]: Device FW version: 01.05.08.01
[ INFO] [1638159211.049762528]: Device Product ID: 0x0B64
[ INFO] [1638159211.049775681]: Enable PointCloud: Off
[ INFO] [1638159211.049800698]: Align Depth: Off
[ INFO] [1638159211.049821031]: Sync Mode: Off
[ INFO] [1638159211.049861823]: Device Sensors: 
[ INFO] [1638159211.050403442]: L500 Depth Sensor was found.
[ INFO] [1638159211.052968718]: RGB Camera was found.
[ INFO] [1638159211.053143660]: Motion Module was found.
[ INFO] [1638159211.053183292]: num_filters: 0
[ INFO] [1638159211.053204601]: Setting Dynamic reconfig parameters.
[ INFO] [1638159211.102423243]: Done Setting Dynamic reconfig parameters.
[ INFO] [1638159211.103002134]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1638159211.104781835]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1638159211.104824125]: setupPublishers...
[ INFO] [1638159211.107759886]: Expected frequency for depth = 30.00000
[ INFO] [1638159211.142701598]: Expected frequency for color = 30.00000
[ INFO] [1638159211.166188832]: setupStreams...
[ INFO] [1638159211.270347675]: SELECTED BASE:Depth, 0
[ INFO] [1638159211.418952266]: RealSense Node Is Up!
[ WARN] [1638159211.445907806]: 

Can you solved it?

MartyG-RealSense, 2021-11-29:
I cannot see any problems in the log.  Have you run RViz with the instruction **rosrun rviz rviz** to visually view the camera data, please?

BamBamBap, 2021-11-30:
I was wrong! Works great when the camera is working! Thank you, it seems to have worked out, thank you very much

MartyG-RealSense, 2021-11-30:
That's great to hear that your camera is working correctly, @BamBamBap - thanks very much for the update!

MartyG-RealSense, 2021-12-05:
Case closed due to successful outcome achieved and no further comments received.

closed 2021-12-05 by MartyG-RealSense