realsenseai/realsense-ros #2182: Can not subscribe the /camera/color/image_raw topic

qin53027, 2021-12-01:
Hi,

My real robot is the Dingo-D from clearpathRobotics which uses the D435i.

1. first of all,  when I run the ROS simulation, I hooked up the D435i with the laptop, I can subscribe the image_raw topic and can see the very nice pictures  -----> it means D435i is working

2. Since Dingo-D uses the /front_realsense/color/image_raw (from the rostopic list and no /camera/color/image_raw), I subscribe Dingo's one. But the image_callbacks function never get called --->  /front_realsense/color/image_raw  topic never get published.

3. But if I chance the  /front_realsense/color/image_raw  into  /front_realsense/color/image_rect_raw (depth image view), then I can see the depth images very well. 

Also I have done: sudo apt-get install ros-melodic-realsense2-camera

Can you please give me an advise how to fix the problem?  Or can you let me know how to publish the image_raw topic in the launch file?

Thanks so much
Jimmy

MartyG-RealSense, 2021-12-02:
Hi @qin53027  The link below has information about using the RealSense camera of the Dingo-D robot, which describes how to roslaunch the real robot using the **dingo_viz** package.

http://wiki.ros.org/dingo_viz

You may therefore be able to subscribe to the Dingo-D 's topics by opening a separate ROS terminal and using the roslaunch command for a real robot that is provided in the above link.

**roslaunch dingo_viz view_robot.launch**

The contents of the **view_robot.launch** launch file are listed here:

https://github.com/dingo-cpr/dingo_desktop/blob/master/dingo_viz/launch/view_robot.launch

qin53027, 2021-12-02:
Hi Marty,

Thanks for your quick response.

Yes I do use: roslaunch dingo_viz view_robot.launch to get real Dingo-D robot working. 

Currently the Dingo-D robot works except that I can not get the /color/image_raw for RGB images.

Currently I did not use the SSH to launch the dingo on board since SSH can not display the RVIZ for my case.

My real question is why I can get the color/image_rect_raw but not the image_raw from the D435i?

If I use the dataset from image_rect_raw (depth image) rather the image_raw (RGB image) for the training for the path navigation, will depth images are still useful for training?

Anyway, I will use the SSH to do the test again to see...

Thanks a lot

MartyG-RealSense, 2021-12-03:
The RealSense ROS wrapper is able to publish **/color/image_raw** but there have been cases where that topic has been missing.  It can be published when using the **roslaunch realsense2_camera rs_camera.launch** roslaunch, as shown in the list of topics in the Intel ROS wrapper documentation pages in the links below.

https://github.com/IntelRealSense/realsense-ros#published-topics
https://dev.intelrealsense.com/docs/ros-wrapper#section-d-400-stereo-depth-cameras

[image]

[image]

I cannot find a reference to a topic called 'color/image_rect_raw' that is published by the RealSense ROS wrapper, just **depth/image_rect_raw**.  The **view_robot.launch** file can apparently publish **/realsense/color/image_raw** but I cannot find a reference to a 'realsense/color/image_rect_raw' being published by it.

I have researched your question extensively but do not have advice about your dataset question that I can offer unfortunately.

qin53027, 2021-12-03:
Hi Marty, 

Here is the updates:

1. This morning, I have hooked up the D435i with my laptop (Ubuntu 18.04), then run: roslaunch realsense2_camera rs_camera.launch

From the rostopic, I can see all the camera/color/* and camera/color/image_raw shows me very nice images ---> thanks!

2. When I do the SSH to the Dingo-D board which is the Nvidia Jetson NX, then run: roslaunch realsense2_camera rs_camera.launch, then I got errors:

[ WARN] [1638545372.658277514]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1638545372.658389418]: The requested device with  is NOT found. Will Try again.
 03/12 09:29:32,656 ERROR [547205673344] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 03/12 09:29:32,657 ERROR [547429478784] (sensor.cpp:517) acquire_power failed: failed to set power state
 03/12 09:29:32,658 WARNING [547429478784] (rs.cpp:306) null pointer passed for argument "device"

I think that error is due to Jetson NX does not have the realsense SDK. Currently Jetson NX is in JetPack4.4 which uses CUDA-10.2. but after I build the realsense SDK on python2.7 on Jetson NX,  it requires CUDA-10.0. So I can not get the realsense SDK work on the Jetson NX for the Python2.7.  I am waiting n JetPack5.0 to be released in Jan. which will support the Ubuntu 20.04 where the Python3 will be in default. I think that using the JetPack5.0, I can build the realsense SDK on Jetson NX for Python3. Then I can run the  roslaunch realsense2_camera rs_camera.launch on board without any problem.

3. Currently I will use the /front_realsense/depth/image_rect_raw to get the depth image to build the dataset since it works. I will see if I can use that dataset to train the network for the images based robot navigation

4. But I do want to report one thing:  this morning I have tried to Subscribe the /front_realsense/color/image_raw (RGB images),
    it gets into the callback function and display 4 images. After that it hung.....
    Do you have any idea on this?   Otherwise I will report the problem to Clearpath for advise 

Thanks so much for your help!
Jimmy

MartyG-RealSense, 2021-12-04:
My knowledge of using ssh in ROS with two computers is admittedly limited.  Intel provide a RealSense ROS tutorial in the link below for using ssh across 2 computers to remotely connect to a camera on the second computer though.

https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines

The guide takes the approach of ssh'ing to the second computer from a ROS terminal on the first computer and then preceding the roslaunch instruction with an **export ROS_MASTER_URI** command containing the http address of the first computer.

https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines#terminal-4--used-to-operate-the-second-machine-perclnx217-in-my-case

In that guide, as all ROS terminals are open on computer 1 and the roslaunches are being initiated from there, the impression that I get (not certainty) that it gives me is that the RealSense ROS wrapper does not need to be installed on the second computer that the remote camera is attached to.

Clearpath should be able to provide specialist advice on this subject, since they are experienced with using RealSense with robots such as Dingo-D.

qin53027, 2021-12-08:
Hi Marty,

Currently I have done following:

1.  I can install/build Librealsense SDK on Laptop, Nvidia Jetson device on Ubuntu 18.04  and Python3.6. The Realsense images works very well.

2. I also can get the Librealsense SDK build on the Jetson device on Ubuntu 18.04 and Python2.7. But since Nvidia stops to support the tensorflow on Python2.7 for Jetson devices, so there is no need to build Librealsense SDK on that since the tensorflow does not work on Python2.7 anymore on the Jetson device.

3. I will wait Nvidia JetPack5.0 to be released on 1Q of 2022 which supports the Ubuntu 20.04. Then I can install the ROS on jetson device in which the Python3 is in default. Then I will build the Librealsense SDK on jetson devices which will work for sure for Python3 base.

For short solution, I am using the USB camera for now since it works for both Python2.7 and Python3.6. 

I like the Intel D435i very much and am looking forward to integrate that into my Robot in 1Q of 2022.

I really appreciate your help
Jimmy

MartyG-RealSense, 2021-12-08:
Thanks so much @qin53027 for the detailed information!

There is an official TensorFlow wrapper for librealsense that uses Python 3.6 and the **pyrealsense2** RealSense compatibility wrapper for Python.  Might it be useful for your work?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/tensorflow

Also, if Python 3 is installed on a computer when the pyrealsense2 Python wrapper is built for librealsense then pyrealsense2 should automatically choose to link to Python 3 instead of Python 2.7 if it can find Python 3.

The librealsense SDK and the pyrealsense2 wrapper can both be compiled at the same time if building librealsense from source code with CMake.

MartyG-RealSense, 2021-12-14:
Hi @qin53027  Do you require further assistance with this case, please?  Thanks!

qin53027, 2021-12-14:
thanks for the help! close this issue