realsenseai/realsense-ros #2186: Error occurs when T265 turns on the fisheye camera

RoZhong, 2021-12-05:
After installing the SDK, there is no problem running the realsense viewer, then after installing realsense2_camera, running rs_t265.launchrs is also no problem, there is no image theme at this time, but once I change it
  <arg name="enable_fisheye1" default="true"/>
  <arg name="enable_fisheye2" default="true"/>
Run rs_t265.launch again, and the following error will appear.

PARAMETERS
 * /camera/tracking_module/enable_pose_jumping: False
 * /camera/tracking_module/enable_relocalization: True
 * /rosdistro: melodic
 * /rosversion: 1.14.12
 * /t265/realsense2_camera/accel_fps: 62
 * /t265/realsense2_camera/accel_frame_id: t265_accel_frame
 * /t265/realsense2_camera/accel_optical_frame_id: t265_accel_optica...
 * /t265/realsense2_camera/align_depth: False
 * /t265/realsense2_camera/aligned_depth_to_color_frame_id: t265_aligned_dept...
 * /t265/realsense2_camera/aligned_depth_to_fisheye1_frame_id: t265_aligned_dept...
 * /t265/realsense2_camera/aligned_depth_to_fisheye2_frame_id: t265_aligned_dept...
 * /t265/realsense2_camera/aligned_depth_to_fisheye_frame_id: t265_aligned_dept...
 * /t265/realsense2_camera/aligned_depth_to_infra1_frame_id: t265_aligned_dept...
 * /t265/realsense2_camera/aligned_depth_to_infra2_frame_id: t265_aligned_dept...
 * /t265/realsense2_camera/allow_no_texture_points: False
 * /t265/realsense2_camera/base_frame_id: t265_link
 * /t265/realsense2_camera/calib_odom_file: 
 * /t265/realsense2_camera/clip_distance: -1.0
 * /t265/realsense2_camera/color_fps: 30
 * /t265/realsense2_camera/color_frame_id: t265_color_frame
 * /t265/realsense2_camera/color_height: 480
 * /t265/realsense2_camera/color_optical_frame_id: t265_color_optica...
 * /t265/realsense2_camera/color_width: 640
 * /t265/realsense2_camera/confidence_fps: 30
 * /t265/realsense2_camera/confidence_height: 480
 * /t265/realsense2_camera/confidence_width: 640
 * /t265/realsense2_camera/depth_fps: 30
 * /t265/realsense2_camera/depth_frame_id: t265_depth_frame
 * /t265/realsense2_camera/depth_height: 480
 * /t265/realsense2_camera/depth_optical_frame_id: t265_depth_optica...
 * /t265/realsense2_camera/depth_width: 640
 * /t265/realsense2_camera/device_type: t265
 * /t265/realsense2_camera/enable_accel: True
 * /t265/realsense2_camera/enable_color: True
 * /t265/realsense2_camera/enable_confidence: True
 * /t265/realsense2_camera/enable_depth: True
 * /t265/realsense2_camera/enable_fisheye1: True
 * /t265/realsense2_camera/enable_fisheye2: True
 * /t265/realsense2_camera/enable_fisheye: False
 * /t265/realsense2_camera/enable_gyro: True
 * /t265/realsense2_camera/enable_infra1: False
 * /t265/realsense2_camera/enable_infra2: False
 * /t265/realsense2_camera/enable_infra: False
 * /t265/realsense2_camera/enable_pointcloud: False
 * /t265/realsense2_camera/enable_pose: False
 * /t265/realsense2_camera/enable_sync: False
 * /t265/realsense2_camera/filters: 
 * /t265/realsense2_camera/fisheye1_frame_id: t265_fisheye1_frame
 * /t265/realsense2_camera/fisheye1_optical_frame_id: t265_fisheye1_opt...
 * /t265/realsense2_camera/fisheye2_frame_id: t265_fisheye2_frame
 * /t265/realsense2_camera/fisheye2_optical_frame_id: t265_fisheye2_opt...
 * /t265/realsense2_camera/fisheye_fps: 30
 * /t265/realsense2_camera/fisheye_frame_id: t265_fisheye_frame
 * /t265/realsense2_camera/fisheye_height: 800
 * /t265/realsense2_camera/fisheye_optical_frame_id: t265_fisheye_opti...
 * /t265/realsense2_camera/fisheye_width: 848
 * /t265/realsense2_camera/gyro_fps: 200
 * /t265/realsense2_camera/gyro_frame_id: t265_gyro_frame
 * /t265/realsense2_camera/gyro_optical_frame_id: t265_gyro_optical...
 * /t265/realsense2_camera/imu_optical_frame_id: t265_imu_optical_...
 * /t265/realsense2_camera/infra1_frame_id: t265_infra1_frame
 * /t265/realsense2_camera/infra1_optical_frame_id: t265_infra1_optic...
 * /t265/realsense2_camera/infra2_frame_id: t265_infra2_frame
 * /t265/realsense2_camera/infra2_optical_frame_id: t265_infra2_optic...
 * /t265/realsense2_camera/infra_fps: 30
 * /t265/realsense2_camera/infra_height: 480
 * /t265/realsense2_camera/infra_rgb: False
 * /t265/realsense2_camera/infra_width: 640
 * /t265/realsense2_camera/initial_reset: False
 * /t265/realsense2_camera/json_file_path: 
 * /t265/realsense2_camera/linear_accel_cov: 0.01
 * /t265/realsense2_camera/odom_frame_id: t265_odom_frame
 * /t265/realsense2_camera/ordered_pc: False
 * /t265/realsense2_camera/pointcloud_texture_index: 0
 * /t265/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /t265/realsense2_camera/pose_frame_id: t265_pose_frame
 * /t265/realsense2_camera/pose_optical_frame_id: t265_pose_optical...
 * /t265/realsense2_camera/publish_odom_tf: True
 * /t265/realsense2_camera/publish_tf: True
 * /t265/realsense2_camera/reconnect_timeout: 6.0
 * /t265/realsense2_camera/rosbag_filename: 
 * /t265/realsense2_camera/serial_no: 
 * /t265/realsense2_camera/stereo_module/exposure/1: 7500
 * /t265/realsense2_camera/stereo_module/exposure/2: 1
 * /t265/realsense2_camera/stereo_module/gain/1: 16
 * /t265/realsense2_camera/stereo_module/gain/2: 16
 * /t265/realsense2_camera/tf_publish_rate: 0.0
 * /t265/realsense2_camera/topic_odom_in: t265/odom_in
 * /t265/realsense2_camera/unite_imu_method: 
 * /t265/realsense2_camera/usb_port_id: 
 * /t265/realsense2_camera/wait_for_device_timeout: -1.0

NODES
  /
    rviz (rviz/rviz)
  /t265/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [27788]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to c46da644-556d-11ec-80c2-f48e38e7fd27
process[rosout-1]: started with pid [27806]
started core service [/rosout]
process[t265/realsense2_camera_manager-2]: started with pid [27813]
process[t265/realsense2_camera-3]: started with pid [27814]
process[rviz-4]: started with pid [27815]
[ INFO] [1638669048.711854497]: Initializing nodelet with 8 worker threads.
[ INFO] [1638669048.947057789]: RealSense ROS v2.3.2
[ INFO] [1638669048.947112709]: Built with LibRealSense v2.50.0
[ INFO] [1638669048.947134145]: Running with LibRealSense v2.50.0
[ INFO] [1638669048.967897170]:  
[ INFO] [1638669048.974737706]: Device with serial number 908412110344 was found.

[ INFO] [1638669048.974791531]: Device with physical ID 2-1.3-4 was found.
[ INFO] [1638669048.974810828]: Device with name Intel RealSense T265 was found.
[ INFO] [1638669048.975490237]: Device with port number 2-1.3 was found.
[ INFO] [1638669048.975568333]: Device USB type: 3.1
[ INFO] [1638669048.981779927]: No calib_odom_file. No input odometry accepted.
[ INFO] [1638669048.982006501]: getParameters...
[ INFO] [1638669049.021829380]: setupDevice...
[ INFO] [1638669049.021861624]: JSON file is not provided
[ INFO] [1638669049.021880263]: ROS Node Namespace: t265
[ INFO] [1638669049.021917208]: Device Name: Intel RealSense T265
[ INFO] [1638669049.021938078]: Device Serial No: 908412110344
[ INFO] [1638669049.021961540]: Device physical port: 2-1.3-4
[ INFO] [1638669049.021984595]: Device FW version: 0.2.0.951
[ INFO] [1638669049.022006258]: Device Product ID: 0x0B37
[ INFO] [1638669049.022028621]: Enable PointCloud: Off
[ INFO] [1638669049.022044113]: Align Depth: Off
[ INFO] [1638669049.022065377]: Sync Mode: Off
[ INFO] [1638669049.022096527]: Device Sensors: 
[ INFO] [1638669049.022247850]: Tracking Module was found.
[ INFO] [1638669049.022287661]: (Depth, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1638669049.022307217]: (Color, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1638669049.022329899]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1638669049.022365956]: num_filters: 0
[ INFO] [1638669049.022391324]: Setting Dynamic reconfig parameters.
[ WARN] [1638669049.024862637]: Param '/t265/tracking_module/frames_queue_size' has value 256 that is not in range [0, 32]. Removing this parameter from dynamic reconfigure options.
[ INFO] [1638669049.030180619]: Done Setting Dynamic reconfig parameters.
/opt/ros/melodic/lib/nodelet/nodelet: symbol lookup error: /home/zyf/catkin_ws2/devel/lib//librealsense2_camera.so: undefined symbol: _ZN2cv3MatC1EiiiRKNS_7Scalar_IdEE

I don’t know why this happens, and I also found
  * /t265/realsense2_camera/enable_fisheye1: True
  * /t265/realsense2_camera/enable_fisheye2: True
  * /t265/realsense2_camera/enable_fisheye: False
In the end, why is this still false? Could it be the problem?

liaozihang, 2022-03-09:
In my case,
change dir to yours `ros_ws` working directory.

type `vim src/realsense-ros/realsense2_camera/CMakeLists.txt`

 add the following code link to OpenCV

`find_package(OpenCV 4 REQUIRED)
include_directories(${OpenCV_INCLUDE_DIRS)
target_link_libraries(${PROJECT_NAME}
     ${OpenCV_LIBS}                                                                                                                                                                  
     )
`

yyy8Bb, 2022-05-19:
> In my case, change dir to yours `ros_ws` working directory.
> 
> type `vim src/realsense-ros/realsense2_camera/CMakeLists.txt`
> 
> add the following code link to OpenCV
> 
> `find_package(OpenCV 4 REQUIRED) include_directories(${OpenCV_INCLUDE_DIRS) target_link_libraries(${PROJECT_NAME} ${OpenCV_LIBS} ) `

I meet the same error.Having add opencv to cmakelist.
but still:

[ INFO] [1652953871.787076391]: Done Setting Dynamic reconfig parameters.
/opt/ros/melodic/lib/nodelet/nodelet: symbol lookup error: /home/young8/mobile_manipulater/real_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN2cv3MatC1Ev

closed 2026-10-23 by MartyG-RealSense