realsenseai/realsense-ros #2187: How to align color and depth images like rs_camera, convert them to opencv format, and put them in orb-slam2？

zma69650, 2021-12-05:
The camera I use is realsense d455. In ROS, I use the rs_camera package provided by realsense. When I use it for orb-slam2, there is no problem. But when I got rid of ros, I found that the trajectory of orb-slam2 was abnormal.
  When I use rs_camera, I let the camera rotate 90 degrees in place, it is normal at this time, the following is a screenshot of the normal time：
[image]
  Then I wrote a program to read the camera data myself. I referred to the example provided by realsense, aligned the color and depth, and converted them to opencv format, and finally put them in the function of orbslam2. The same is true when the camera is rotated 90 degrees in place, and an abnormality is found. ORB-SLAM2 is no problem, and the configuration file is the same. The following is a screenshot of the exception:
[image]

  My procedure is as follows:
‘#include <iostream>
#include "librealsense2/rs.hpp"
#include "System.h"
#include "opencv2/core/core.hpp"

int main(int argc,char** argv) {

    rs2::pipeline pipe;
    rs2::align aligned_color(RS2_STREAM_COLOR);
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_COLOR,640,480,RS2_FORMAT_BGR8,30);
    cfg.enable_stream(RS2_STREAM_DEPTH,640,480,RS2_FORMAT_Z16,30);
    pipe.start(cfg);

    ORB_SLAM2::System SLAM("./ORBvoc.txt","./realsensed455.yaml",ORB_SLAM2::System::RGBD, true);

    while(1)
    {
        rs2::frameset frames = pipe.wait_for_frames();
        frames = aligned_color.process(frames);
        rs2::video_frame depth = frames.get_depth_frame();
        rs2::video_frame color = frames.get_color_frame();

        //opencv
        cv::Mat depth_mat(cv::Size(640,480),CV_16U,(void *)depth.get_data(),cv::Mat::AUTO_STEP);
        cv::Mat color_mat(cv::Size(640,480),CV_8UC3,(void *)color.get_data(),cv::Mat::AUTO_STEP);

        size_t timestamp = time(NULL);

        //orbslam
        SLAM.TrackRGBD(color_mat,depth_mat,timestamp);

    }

    SLAM.Shutdown();
	return 0;

}’

MartyG-RealSense, 2021-12-05:
Hi @zma69650   Is the image produced for ORB-SLAM2 by your OpenCV script correct up until the point when you rotate the camera 90 degrees and then the image goes wrong?

zma69650, 2021-12-06:
thank you for your reply. @MartyG-RealSense The image looks normal, and I use the imshow function to display it.

zma69650, 2021-12-06:
The problem has been solved! ! I'm careless! I wrote the internal reference in depth, it should be in color. thank you for your reply! ! I will close this question.

closed 2021-12-06 by zma69650