realsenseai/realsense-ros #2189: control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

mintnguyenn, 2021-12-08:
Hello, I am running ' roslaunch realsense2_camera rs_camera.launch ' with Realsense D435, and I face the error like

started roslaunch server http://hpomen15:42523/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
[... 79 lines cut ...]
[ INFO] [1638932608.627963035]: Expected frequency for depth = 30.00000
[ INFO] [1638932608.658138438]: Expected frequency for color = 30.00000
[ INFO] [1638932608.674053411]: setupStreams...
 08/12 14:03:28,954 WARNING [139669442389760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1638932609.004707118]: SELECTED BASE:Depth, 0
[ INFO] [1638932609.015824615]: RealSense Node Is Up!
[ WARN] [1638932609.072511401]: 
 08/12 14:03:29,073 WARNING [139669442389760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/12 14:03:29,123 WARNING [139669442389760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/12 14:03:29,173 WARNING [139669442389760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/12 14:03:29,306 WARNING [139669442389760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/12 14:03:29,356 WARNING [139669442389760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/12 14:03:29,406 WARNING [139669442389760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/12 14:04:36,627 WARNING [139669442389760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/12 14:04:41,681 WARNING [139669442389760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/12 14:04:42,254 ERROR [139668747962112] (ds5-options.cpp:88) Asic Temperature value is not valid!
 08/12 14:04:43,257 ERROR [139668747962112] (ds5-options.cpp:88) Asic Temperature value is not valid!
 08/12 14:04:44,260 ERROR [139668747962112] (ds5-options.cpp:88) Asic Temperature value is not valid!
 08/12 14:05:18,368 ERROR [139668747962112] (ds5-options.cpp:88) Asic Temperature value is not valid!
 08/12 14:05:19,371 ERROR [139668747962112] (ds5-options.cpp:88) Asic Temperature value is not valid!
 08/12 14:05:20,374 ERROR [139668747962112] (ds5-options.cpp:88) Asic Temperature value is not valid!
 08/12 14:24:27,609 WARNING [139669442389760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/12 14:24:27,659 WARNING [139669442389760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

Thanks for your help!

MartyG-RealSense, 2021-12-08:
Hi @mintnguyenn  If you are receiving the **control_transfer returned** warning only several times during launch and then they stop, or they occur one or two times a minute after launch, then this warning can be safely ignored. You can check in RViz to see whether the topics are being published correctly despite the warnings.  If this warning is generating continuously though then it can indicate a serious communication problem with the device.

Please try using the custom launch instruction below to set depth and color both to 640x480 and 15 FPS to see whether this reduces the frequency of the warnings.  

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=15 color_width:=640 color_height:=480 color_fps:=15**

mintnguyenn, 2021-12-09:
Hi @MartyG-RealSense , the WARNING occur pretty much, but it does not affect the quality of rostopic. I think I still can capture the image via callback. But when I run with realsense-viewer, the stream quality is pretty bad, like Full HD with 6fps, and a couple of errors occurred. I think the problem is the SDK version, or something like that, because the rviz can stream pretty well.

MartyG-RealSense, 2021-12-09:
Ah yes, I see in the log that you have a mismatch between the librealsense version and wrapper version.  

[image]

It looks as though you first installed librealsense 2.48.0 and built ROS wrapper version 2.3.2.  Then librealsense was updated to 2.50.0, which is the correct SDK version to match to wrapper 2.50.0.  This is because a particular wrapper version should ideally be matched as closely as possible with a specific librealsense version listed in the wrapper release notes.

[image]

[image]

However, once the librealsense version has been changed (for example, upgrading from 2.48.0 to 2.50.0) then the ROS wrapper has to be built again afterwards.  Then the log message about the mismatch will disappear and it should display the following:

```
RealSense ROS v2.3.2
Built with LibRealSense v2.50.0
Running with LibRealSense v2.50.0
```

closed 2021-12-21 by MartyG-RealSense