realsenseai/realsense-ros #2190: Camera orientation of D435i in ROS

EhrazImam, 2021-12-08:
Hi I am trying to run Robot localization but when i am moving upward the i am getting values in -y while it should be on z...How can i change the orientation of camera so i can get the exact position of robot.

MartyG-RealSense, 2021-12-08:
Hi @EhrazImam   The D435i coordinate axes are described in the link below, where Z (depth) is forward, Y+ is downward and Y- is upward.

https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md#sensor-origin-and-coordinate-system

I note that you have already seen https://github.com/IntelRealSense/realsense-ros/issues/1176 that discusses this issue.

If performing a static transform as a RealSense user in https://github.com/IntelRealSense/realsense-ros/issues/1176 suggested does not work for you, the link below suggests an alternative approach of changing the camera position / orientation in **RViz** and saving the configuration.

https://answers.ros.org/question/339109/how-to-change-the-cameras-view-in-rviz-in-the-begin/

EhrazImam, 2021-12-08:
Hi @MartyG-RealSense Thanx for your quick reply..
Already tried it but still same issue

MartyG-RealSense, 2021-12-08:
https://github.com/IntelRealSense/realsense-ros/issues/1099#issuecomment-658101736 may be relevant to your situation.  In it, **Doronhi** the RealSense ROS wrapper developer suggests applying a multiplication to convert to standard ROS frame_id (X-forward, Y-left, Z-up).

EhrazImam, 2021-12-09:
> [#1099 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/1099#issuecomment-658101736) may be relevant to your situation. In it, **Doronhi** the RealSense ROS wrapper developer suggests applying a multiplication to convert to standard ROS frame_id (X-forward, Y-left, Z-up).

The multiplication @doronhi he's talking about is already in the code but nothing helped...please let me know any other method of changing the orientation of camera

MartyG-RealSense, 2021-12-09:
I do not personally have a further suggestion to make, unfortunately.

Wehn you say that you are doing robot localization, do you mean that you are literally using the ROS **robot_localization** pakcage, please?

bhaskar-glitch, 2021-12-09:
> I do not personally have a further suggestion to make, unfortunately.
> 
> Wehn you say that you are doing robot localization, do you mean that you are literally using the ROS **robot_localization** package, please?
> 
> http://wiki.ros.org/robot_localization

No! I'm using **VINS_FUSION**(https://github.com/pjrambo/VINS-Fusion-gpu) with d435i and I'm only using stereo camera of d435i not the imu of it!

MartyG-RealSense, 2021-12-09:
I researched your question extensively in regard to using z as the up-down position (non IMU) in VINS-FUSION instead of y but could find no guidance unfortunately.

MartyG-RealSense, 2021-12-15:
Hi @bhaskar-glitch  Do you require further assistance with this case, please?  Thanks!

bhaskar-glitch, 2021-12-15:
> Hi @bhaskar-glitch Do you require further assistance with this case, please? Thanks!

How can I extract imu values of realsense d435i externally using GPIO pins...??

MartyG-RealSense, 2021-12-15:
Hi @bhaskar-glitch  You will not be able to obtain stream data from the GPIO pins, as described in https://github.com/IntelRealSense/librealsense/issues/10049

closed 2021-12-27 by MartyG-RealSense