realsenseai/realsense-ros #2191: uvc streamer watchdog triggered on endpoint

Zkjoker, 2021-12-08:
My camera is D435i,  and can't use filters:=pointcloud. 
If I use:
`roslaunch realsense_camera2 rs_camera.launch`, it will be ok , and every thing goes well, no matter in the laptop or Tx2.
But when I use:
`roslaunch realsense_camera2 rs_camera.launch filters:=pointcloud`, in my laptop,  I can see the pointcloud in the rviz. However,   in my TX2,  it raise an error, "uvc streamer watchdog triggered on endpoint". I have tried many solutions in the issues, but the error never changed. （for example changed the fps to 15）
I  wonder if you have identified the problem and what solutions are available？？

MartyG-RealSense, 2021-12-08:
Hi @Zkjoker  Could you check whether you have the ability to change **Color Transformer** options in RViz when the pointcloud is enabled, as described in https://github.com/IntelRealSense/realsense-ros/issues/1967#issuecomment-873375954

Zkjoker, 2021-12-08:
No,there is nothing. In fact, `rostopic echo /camera/depth/color/points`  can't output anything.

I have used the method: `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud depth_width:=640 depth_height:=480 depth_fps:=15` ,  in this way I can get the image and pointcloud, however, there is such a big delay that it can't be used at all. The program is still not working properly.

And I found another method: ` roslaunch realsense2_camera rs_camera.launch filters:=pointcloud pointcloud_texture_stream:=RS2_STREAM_ANY` ,  in this way , I can get a white pointcloud without any delay in rviz, but in this way I can't get image any more.

I found that many people have this problem in the issues. Have you ever bought an ARM computer such as TX2 and Xavier for testing? It seems to be a bug, and not because of the computing performance of the platform.

Zkjoker, 2021-12-08:
@MartyG-RealSense

MartyG-RealSense, 2021-12-08:
There is another case at https://github.com/IntelRealSense/realsense-ros/issues/2087 regarding a RealSense ROS user with a Jetson TX2 who had problems when the pointcloud filter was enabled.

In addition to TX2, there have been a small number of issues reported for other Jetson models such as Xavier when generating pointclouds in the ROS wrapper.  It is not clear why this phenomenon occurs on Jetson board models.

Zkjoker, 2021-12-09:
@MartyG-RealSense Thank you for your relpy. I have searched the problem in the issues, and tried different method.
`initial_reset:= true` doesn't have positive effect. 
I have tried : `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud depth_width:=640 depth_height:=480 depth_fps:=15` it can work but the delay is too long. 

In #2078 , the command is:
`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud depth_width:=848 depth_height:=480 depth_fps:=15 color_width:=1280 color_height:=720 color_fps:=15 `
but his camera is D455,  my camera is D435i,  could you please tell me which number of depth_width/ height, color_width/height should i use? Thank you!

In fact I also don't want to use the Jetson model,  many programs that work on AMD computers do not work here. But for robot,  it is a common host computer option.  I still hope you can pay attention to it because if the ROS driver cannot be used on the robot, this driver will become meaningless. 

Really thank you for your reply!

MartyG-RealSense, 2021-12-15:
Hi @Zkjoker  Do you require further assistance with this case, please?  Thanks!

Zkjoker, 2021-12-15:
of course, I still hope you can find a real way to solve the problem in tx2 or Xavier.

MartyG-RealSense, 2021-12-15:
A RealSense ROS user at https://github.com/IntelRealSense/realsense-ros/issues/1964 who had a Jetson Xavier AGX had very high CPU usage when the pointcloud was enabled.  They reduced CPU usage significantly by applying a **Decimation Filter**, at the cost of a reduction in resolution.

MartyG-RealSense, 2021-12-21:
Hi @Zkjoker  Do you have an update about this case that you can provide, please?  Was the comment above about applying a Decimation Filter helpful?  Thanks!

Zkjoker, 2021-12-23:
Recently, I can't debug due to my business trip. I'm very sorry. I can't tell you whether this method is effective.

MartyG-RealSense, 2021-12-23:
Okay, that is fine.  Please do update when you are able to do debugging work again.  Thanks!

closed 2022-03-19 by MartyG-RealSense