realsenseai/realsense-ros #2192: issues in setting resolution with camera D435

Tom127, 2021-12-08:
@doronhi Hi, I‘m using D435 in ubuntu16.04, when I use the [ros driver](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_camera.launch), although I modified the [line 26](https://github.com/IntelRealSense/realsense-ros/blob/f400d682beee6c216052a419f419e95b797255ad/realsense2_camera/launch/rs_camera.launch#L26) and [line 27](https://github.com/IntelRealSense/realsense-ros/blob/f400d682beee6c216052a419f419e95b797255ad/realsense2_camera/launch/rs_camera.launch#L27), I found that its resolution is always 640x480.but I need a resolution of 1280x720, how could I reset it?

Best regards,
Tom

MartyG-RealSense, 2021-12-08:
Hi @Tom127   If you have defined a custom configuration for Infrared with three factors provided for that stream type - Width 1280, Height 720 and FPS 30 - but the launch is applying 640x480, I wonder whether you are using the camera in a USB 2 port instead of a USB 3 port, or the camera is in a USB 3 port but the camera connection is being mis-detected as USB 2.  

1280x720 at 30 FPS would not be supported on a limited-capability USB 2 connection, as the only FPS speed available on USB 2 with a D435 using 1280x720 is 6 FPS. 

You can check in the roslaunch log in the ROS terminal how the USB connection is being detected by finding the **Device USB type** line.  If it is detected as USB 2 then the value on this line will be '2.1', whilst if it is detected as USB 3 then it should say '3.2'.

Tom127, 2021-12-09:
@MartyG-RealSense Thank you for your reply, the problem has been solved!

closed 2021-12-09 by Tom127

MartyG-RealSense, 2021-12-09:
Great to hear that you achieved a solution, @Tom127 - thanks very much for the update!