realsenseai/realsense-ros #2193: Cmake error starting with version 2.2.5

canberkgurel, 2021-12-09:
I'm trying to install realsense-ros on Nvidia Jetson XavierNX (Ubuntu 18.04 & ROS Melodic).

I build the workspace with `~/catkin_ws_three$ catkin build --make-args -j4 --cmake-args -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release`

I tried to build 2.2.24 (the latest version at the time of writing this) and it didn't work.

So, I tried finding a version that builds without an error. Version 2.2.24 is the latest version that builds OK. However, there's a cmake error that starts with version 2.2.5.Here is the error:

```
Errors     << realsense2_camera:make /home/xaviernx/catkin_ws_three/logs/realsense2_camera/build.make.008.log                                        
In file included from /home/xaviernx/catkin_ws_three/src/realsense-ros/realsense2_camera/include/base_realsense_node.h:7:0,
                 from /home/xaviernx/catkin_ws_three/src/realsense-ros/realsense2_camera/src/../include/t265_realsense_node.h:3,
                 from /home/xaviernx/catkin_ws_three/src/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:1:
/home/xaviernx/catkin_ws/src/realsense-ros/ddynamic_reconfigure/include/ddynamic_reconfigure/ddynamic_reconfigure.h:213:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
           #pragma clang diagnostic push // CLion suppressor
 
/home/xaviernx/catkin_ws/src/realsense-ros/ddynamic_reconfigure/include/ddynamic_reconfigure/ddynamic_reconfigure.h:214:0: warning: ignoring #pragma ide diagnostic [-Wunknown-pragmas]
           #pragma ide diagnostic ignored "OCUnusedGlobalDeclarationInspection"
 
/home/xaviernx/catkin_ws/src/realsense-ros/ddynamic_reconfigure/include/ddynamic_reconfigure/ddynamic_reconfigure.h:220:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
           #pragma clang diagnostic pop
 
In file included from /home/xaviernx/catkin_ws_three/src/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:7:0,
                 from /home/xaviernx/catkin_ws_three/src/realsense-ros/realsense2_camera/src/realsense_node_factory.cpp:5:
/home/xaviernx/catkin_ws/src/realsense-ros/ddynamic_reconfigure/include/ddynamic_reconfigure/ddynamic_reconfigure.h:213:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
           #pragma clang diagnostic push // CLion suppressor
 
/home/xaviernx/catkin_ws/src/realsense-ros/ddynamic_reconfigure/include/ddynamic_reconfigure/ddynamic_reconfigure.h:214:0: warning: ignoring #pragma ide diagnostic [-Wunknown-pragmas]
           #pragma ide diagnostic ignored "OCUnusedGlobalDeclarationInspection"
 
/home/xaviernx/catkin_ws/src/realsense-ros/ddynamic_reconfigure/include/ddynamic_reconfigure/ddynamic_reconfigure.h:220:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
           #pragma clang diagnostic pop
 
In file included from /home/xaviernx/catkin_ws_three/src/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:7:0,
                 from /home/xaviernx/catkin_ws_three/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1:
/home/xaviernx/catkin_ws/src/realsense-ros/ddynamic_reconfigure/include/ddynamic_reconfigure/ddynamic_reconfigure.h:213:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
           #pragma clang diagnostic push // CLion suppressor
 
/home/xaviernx/catkin_ws/src/realsense-ros/ddynamic_reconfigure/include/ddynamic_reconfigure/ddynamic_reconfigure.h:214:0: warning: ignoring #pragma ide diagnostic [-Wunknown-pragmas]
           #pragma ide diagnostic ignored "OCUnusedGlobalDeclarationInspection"
 
/home/xaviernx/catkin_ws/src/realsense-ros/ddynamic_reconfigure/include/ddynamic_reconfigure/ddynamic_reconfigure.h:220:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
           #pragma clang diagnostic pop
 
/home/xaviernx/catkin_ws_three/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp: In member function ‘void realsense2_camera::BaseRealSenseNode::registerDynamicOption(ros::NodeHandle&, rs2::options, std::__cxx11::string&)’:
/home/xaviernx/catkin_ws_three/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:281:22: error: ‘using element_type = class ddynamic_reconfigure::DDynamicReconfigure {aka class ddynamic_reconfigure::DDynamicReconfigure}’ has no member named ‘registerVariable’
             ddynrec->registerVariable<bool>(
                      ^~~~~~~~~~~~~~~~
/home/xaviernx/catkin_ws_three/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:281:39: error: expected primary-expression before ‘bool’
             ddynrec->registerVariable<bool>(
                                       ^~~~
/home/xaviernx/catkin_ws_three/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:283:89: error: expected primary-expression before ‘,’ token
               [option, sensor](bool new_value) { sensor.set_option(option, new_value); },
                                                                                         ^
/home/xaviernx/catkin_ws_three/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:309:24: error: ‘using element_type = class ddynamic_reconfigure::DDynamicReconfigure {aka class ddynamic_reconfigure::DDynamicReconfigure}’ has no member named ‘registerVariable’
               ddynrec->registerVariable<int>(
                        ^~~~~~~~~~~~~~~~
/home/xaviernx/catkin_ws_three/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:309:41: error: expected primary-expression before ‘int’
               ddynrec->registerVariable<int>(
                                         ^~~
/home/xaviernx/catkin_ws_three/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:311:92: error: expected primary-expression before ‘,’ token
                   [option, sensor](int new_value) { sensor.set_option(option, new_value); },
                                                                                            ^
/home/xaviernx/catkin_ws_three/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:329:28: error: ‘using element_type = class ddynamic_reconfigure::DDynamicReconfigure {aka class ddynamic_reconfigure::DDynamicReconfigure}’ has no member named ‘registerVariable’
                   ddynrec->registerVariable<double>(
                            ^~~~~~~~~~~~~~~~
/home/xaviernx/catkin_ws_three/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:329:45: error: expected primary-expression before ‘double’
                   ddynrec->registerVariable<double>(
                                             ^~~~~~
/home/xaviernx/catkin_ws_three/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:331:99: error: expected primary-expression before ‘,’ token
[... 4 lines cut ...]
                      ^~~~~~~~~~~~~~~~~~~~
/home/xaviernx/catkin_ws_three/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:357:43: error: expected primary-expression before ‘int’
             ddynrec->registerEnumVariable<int>(
                                           ^~~
/home/xaviernx/catkin_ws_three/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:359:90: error: expected primary-expression before ‘,’ token
                 [option, sensor](int new_value) { sensor.set_option(option, new_value); },
                                                                                          ^
/home/xaviernx/catkin_ws_three/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:363:14: error: ‘using element_type = class ddynamic_reconfigure::DDynamicReconfigure {aka class ddynamic_reconfigure::DDynamicReconfigure}’ has no member named ‘publishServicesTopics’
     ddynrec->publishServicesTopics();
              ^~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/realsense2_camera.dir/all] Error 2
make: *** [all] Error 2
cd /home/xaviernx/catkin_ws_three/build/realsense2_camera; catkin build --get-env realsense2_camera | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.....................................................................................................................................................
Failed     << realsense2_camera:make                [ Exited with code 2 ]                                                                           
[... cut ...]
[build]   Ignored:   None.                                                                                                                           
[build]   Warnings:  None.                                                                                                                           
[build]   Abandoned: None.                                                                                                                           
[build]   Failed:    1 packages failed.                                                                                                              
[build] Runtime: 37.4 seconds total.   
```

canberkgurel, 2021-12-10:
@MartyG-RealSense Yes, your analysis is correct.

I installed the librealsense via apt-get. So, I have 2.50.0 on my NVIDIA computer.

I have two questions. 

If I need different librealsense versions, I can go ahead an builf from source. Is there a table that I can use to match the librealsense version with realsense-ros version?

I'm testing three different open-source algorithms, they each use D435i, but may require different realsense-ros versions. So, I created three separate catkin workspaces. And I git cloned 3 different versions of realsense-ros. Is it possible to pass the directory of librealsense that I what the compiler to use as a cmake argument?

MartyG-RealSense, 2021-12-10:
No, there isn't a table comparing wrappers against recommended librealsense versions to match to.  The easiest way to learn the correct match is to look up a particular wrapper version on the wrapper releases page and check that version's release notes for which librealsense version to use with it.

I would highly recommend matching one version to one librealsense version at a time, and then cleanly remove those versions and install different versions for the next set of tests.  Otherwise I can foresee that you could end up with **API mismatch** errors if you have multiple librealsense and wrapper versions on the same computer that have the potential to conflict with each other.

closed 2021-12-16 by canberkgurel