realsenseai/realsense-ros #2194: L515 Short-range mode on ROS how do I set it without opening realsense-viewer

sprenkle, 2021-12-11:
I am using ROS and want to change it to the short-range detection.  How do I do this?  I know how to do it with realsense-viewer but not by using ros msgs or use of the launch file.  This is NOT a duplicate, someone asked the same kind of question and the answer was "You can open the dynamic reconfigure and select the mode in the drop down menu", open what?.   That tells me nothing.  Can someone please tell me how this can be done in more detail that a novice might be able to figure it out.

MartyG-RealSense, 2021-12-11:
Hi @sprenkle   The quote is referring to the ability to alter parameters during runtime after launch has completed.  This can be performed in ROS1 (Kinetic, Melodic, Noetic) with a package called **ddynamic_reconfigure** (also known as rqt_reconfigure).  ddynamic_reconfigure can be controlled through a visual interface that is opened by inputting the command below into the ROS terminal after roslaunch has completed.

**rosrun rqt_reconfigure rqt_reconfigure**

Further information is provided in the RealSense ROS wrapper documentation at the link below.

https://github.com/IntelRealSense/realsense-ros#set-camera-controls-using-dynamic-reconfigure-params

sprenkle, 2021-12-11:
Thank you for the quick answer, if I could just have one more question relating to this.  Is that setting for visualizing only, will my robot using the sensor have the same accuracy in the measurement at low distances do matter what this setting is?

MartyG-RealSense, 2021-12-11:
A preset (also known as a **json** camera configuration) has a direct effect on the camera data by reconfiguring the values of camera settings.

sprenkle, 2021-12-11:
Thanks again.

closed 2021-12-11 by sprenkle

JuliusSustarevas, 2021-12-17:
Is there a way to do this on launch? or make the preset persistent? 
For some reason setting it as json is not working for me. I also saw an issue about initial  hardware reset potentially causing this
https://github.com/IntelRealSense/realsense-ros/issues/2199

MartyG-RealSense, 2021-12-18:
Hi @JuliusSustarevas  Your question has been replied to in your case at https://github.com/IntelRealSense/realsense-ros/issues/2199 - thanks very much!