realsenseai/realsense-ros #2195: How to get transformation of d455 from its IMU?

victorll998, 2021-12-13:
Hello there,

I have been noticing that d455 has the same IMU unit as d435i. I would like to know if there is any interface can generate d455's transformation in ROS.

I tried to set unite_imu_method to copy or linear_interpolation so as to get fused IMU data and publish odom tf frame in ROS. But little did I proceed to get a new topic with IMU data or transformation. 

Any suggestion about it would be helpful

MartyG-RealSense, 2021-12-13:
Hi @victorll998  Does the information provided in https://github.com/IntelRealSense/realsense-ros/issues/1449#issuecomment-711675699 about obtaining the gyro transform with **rosrun tf tf_echo camera_depth_optical_frame camera_gyro_optical_frame** answer your question, please?

victorll998, 2021-12-13:
Yes, that's what I'm looking for at the moment.

Could you also have a look if the launch parameter unite_imu_method works for your machine?

MartyG-RealSense, 2021-12-13:
In the rs_camera launch file, the IMU 'gyro' and 'accel' topics are disabled by default.  Have you added the commands below to your roslaunch instruction to enable publishing of the IMU topics, please?

**enable_gyro:=true enable_accel:=true**

For example:

**roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation**

victorll998, 2021-12-13:
I found my mistake of wrong substitution args. 

I fixed it and now I can see the new IMU topic, but there is no new IMU topic message. I am not sure if it is working.

--- 
SUMMARY
========

PARAMETERS
 * /d455/passthrough_x/filter_field_name: x
 * /d455/passthrough_x/filter_limit_max: 2
 * /d455/passthrough_x/filter_limit_min: -2
 * /d455/passthrough_x/filter_limit_negative: False
 * /d455/passthrough_z/filter_field_name: z
 * /d455/passthrough_z/filter_limit_max: 6
 * /d455/passthrough_z/filter_limit_min: 0.4
 * /d455/passthrough_z/filter_limit_negative: False
 * /d455/realsense2_camera/accel_fps: 0
 * /d455/realsense2_camera/accel_frame_id: d455_accel_frame
 * /d455/realsense2_camera/accel_optical_frame_id: d455_accel_optica...
 * /d455/realsense2_camera/align_depth: True
 * /d455/realsense2_camera/aligned_depth_to_color_frame_id: d455_aligned_dept...
 * /d455/realsense2_camera/aligned_depth_to_fisheye1_frame_id: d455_aligned_dept...
 * /d455/realsense2_camera/aligned_depth_to_fisheye2_frame_id: d455_aligned_dept...
 * /d455/realsense2_camera/aligned_depth_to_fisheye_frame_id: d455_aligned_dept...
 * /d455/realsense2_camera/aligned_depth_to_infra1_frame_id: d455_aligned_dept...
 * /d455/realsense2_camera/aligned_depth_to_infra2_frame_id: d455_aligned_dept...
 * /d455/realsense2_camera/allow_no_texture_points: False
 * /d455/realsense2_camera/base_frame_id: d455_link
 * /d455/realsense2_camera/calib_odom_file: 
 * /d455/realsense2_camera/clip_distance: -1.0
 * /d455/realsense2_camera/color_fps: 15
 * /d455/realsense2_camera/color_frame_id: d455_color_frame
 * /d455/realsense2_camera/color_height: 720
 * /d455/realsense2_camera/color_optical_frame_id: d455_color_optica...
 * /d455/realsense2_camera/color_width: 1280
 * /d455/realsense2_camera/confidence_fps: 30
 * /d455/realsense2_camera/confidence_height: 480
 * /d455/realsense2_camera/confidence_width: 640
 * /d455/realsense2_camera/depth_fps: 15
 * /d455/realsense2_camera/depth_frame_id: d455_depth_frame
 * /d455/realsense2_camera/depth_height: 720
 * /d455/realsense2_camera/depth_optical_frame_id: d455_depth_optica...
 * /d455/realsense2_camera/depth_width: 1280
 * /d455/realsense2_camera/device_type: d455
 * /d455/realsense2_camera/enable_accel: True
 * /d455/realsense2_camera/enable_color: True
 * /d455/realsense2_camera/enable_confidence: True
 * /d455/realsense2_camera/enable_depth: True
 * /d455/realsense2_camera/enable_fisheye1: False
 * /d455/realsense2_camera/enable_fisheye2: False
 * /d455/realsense2_camera/enable_fisheye: False
 * /d455/realsense2_camera/enable_gyro: True
 * /d455/realsense2_camera/enable_infra1: False
 * /d455/realsense2_camera/enable_infra2: False
 * /d455/realsense2_camera/enable_infra: False
 * /d455/realsense2_camera/enable_pointcloud: False
 * /d455/realsense2_camera/enable_pose: False
 * /d455/realsense2_camera/enable_sync: False
 * /d455/realsense2_camera/filters: pointcloud
 * /d455/realsense2_camera/fisheye1_frame_id: d455_fisheye1_frame
 * /d455/realsense2_camera/fisheye1_optical_frame_id: d455_fisheye1_opt...
 * /d455/realsense2_camera/fisheye2_frame_id: d455_fisheye2_frame
 * /d455/realsense2_camera/fisheye2_optical_frame_id: d455_fisheye2_opt...
 * /d455/realsense2_camera/fisheye_fps: 30
 * /d455/realsense2_camera/fisheye_frame_id: d455_fisheye_frame
 * /d455/realsense2_camera/fisheye_height: 0
 * /d455/realsense2_camera/fisheye_optical_frame_id: d455_fisheye_opti...
 * /d455/realsense2_camera/fisheye_width: 0
 * /d455/realsense2_camera/gyro_fps: 0
[... 111 lines cut ...]
[ INFO] [2021-12-13 23:40:08]: setupPublishers...
[ INFO] [2021-12-13 23:40:08]: Expected frequency for depth = 15.00000
[ INFO] [2021-12-13 23:40:08]: Expected frequency for color = 15.00000
[ INFO] [2021-12-13 23:40:08]: Expected frequency for aligned_depth_to_color = 15.00000
[ INFO] [2021-12-13 23:40:08]: Start publisher IMU
[ INFO] [2021-12-13 23:40:08]: setupStreams...
[ INFO] [2021-12-13 23:40:09]: insert Depth to Stereo Module
[ INFO] [2021-12-13 23:40:09]: insert Color to RGB Camera
[ INFO] [2021-12-13 23:40:09]: insert Gyro to Motion Module
[ INFO] [2021-12-13 23:40:09]: insert Accel to Motion Module
 14/12 10:40:09,022 WARNING [140342611404544] (sensor.cpp:991) HID timestamp not found, switching to Host timestamps.
[ WARN] [2021-12-13 23:40:09]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 14/12 10:40:09,043 WARNING [140342587229952] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 14/12 10:40:09,043 WARNING [140342587229952] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 14/12 10:40:09,043 WARNING [140342587229952] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
[ INFO] [2021-12-13 23:40:09]: SELECTED BASE:Depth, 0
 14/12 10:40:09,043 WARNING [140342587229952] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 14/12 10:40:09,063 WARNING [140342571448064] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 14/12 10:40:09,063 WARNING [140342571448064] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 14/12 10:40:09,063 WARNING [140342571448064] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
[ INFO] [2021-12-13 23:40:09]: RealSense Node Is Up!
 14/12 10:40:09,063 WARNING [140342571448064] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
[ WARN] [2021-12-13 23:40:10]: Hardware Notification:Depth stream start failure,1.63944e+12,Error,Hardware Error
[ WARN] [2021-12-13 23:40:13]: This node/nodelet subscribes topics only when subscribed.
[ WARN] [2021-12-13 23:40:13]: This node/nodelet subscribes topics only when subscribed.

MartyG-RealSense, 2021-12-14:
The log states that the IMU topics are being published, but you are experiencing the **Invalid md size: bytes used = 0 ,start offset=10** warning.  In https://github.com/IntelRealSense/realsense-ros/issues/1734 a couple of RealSense ROS users found that building librealsense from source code with the **V4L backend** (using the CMake build term **-DFORCE_LIBUVC=OFF**) resolved this warning.

closed 2021-12-27 by MartyG-RealSense