realsenseai/realsense-ros #2196: D435i IMU: Orientation values stay at 0

roderik3000, 2021-12-16:
Camera model: D435i
OS: Ubuntu 18.04
Platform: Nvidia Jetson 4GB

I'm trying to obtain orientation data from the D435i using the topic ``camera/imu`` while running the launch command ``ros2 launch realsense2_camera rs_launch.py enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation``

Listening to the topic I recieve:
```
---
header:
  stamp:
    sec: 1639644350
    nanosec: 624004352
  frame_id: camera_imu_optical_frame
orientation:
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity:
  x: -4.234506195643917e-05
  y: 0.0017455772031098604
  z: 2.141096229024697e-05
angular_velocity_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
linear_acceleration:
  x: 0.11746231202614794
  y: -9.720884680561024
  z: -0.20704855840722697
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
---
```

I have found this issue: https://github.com/IntelRealSense/realsense-ros/issues/1210 it describes the same problem. Reading through the issue I understand that someone was able to get it to work by listening to the ``/imu/data`` topic.

But listening to ``/imu/data`` will never return anything and when subscribing to the topic in a node it never gets called. Is there still a way to recieve orientation data from the IMU?

MartyG-RealSense, 2021-12-16:
Hi @roderik3000   In the https://github.com/IntelRealSense/realsense-ros/issues/1210 case that you linked to, **Doronhi** the RealSense ROS wrapper developer stated in https://github.com/IntelRealSense/realsense-ros/issues/1210#issuecomment-634465607 that the D435i does not publish orientation information but you could use a SLAM related roslaunch such as **opensource_tracking.launch** instead of rs_camera to obtain orientation.  My understanding is that the RealSense ROS user in https://github.com/IntelRealSense/realsense-ros/issues/1210 was able to echo the **/imu/data** topic after launching opensouce_tracking.launch

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/opensource_tracking.launch

opensource_tracking.launch is a ROS1 launch file though.  I am not aware of an equivalent SLAM-equipped launch file for ROS2.

MartyG-RealSense, 2021-12-24:
Hi @roderik3000  Do you require further assistance with this case, please?  Thanks!

roderik3000, 2021-12-24:
I do not, thank you.

closed 2021-12-24 by roderik3000