realsenseai/realsense-ros #2197: How to Build the Librealsense2 SDK on the Jetson NX based on the Python2.7?

qin53027, 2021-12-17:
Hi Marty and others:

Do you have a reference to share on how to build the SDK on Python2.7 since for my case I need to use the Python2.7 on the D435i camera?

Thanks
Jimmy

MartyG-RealSense, 2021-12-17:
Hi @qin53027  The RealSense Python wrapper should use Python 2.7 if it cannot use a Python 3 version on your Jetson.  If you need to have both Python 2.7 and Python 3 on your Jetson though then you can specify for the wrapper to use Python 2.7 by building from source code with CMake and including the CMake flag below in the build instruction.

`-DPYTHON_EXECUTABLE=/usr/bin/python2.7`

There is an installation guide shared by a RealSense user at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 for building librealsense and the Python wrapper together from source code at the same time.  Change references in the instructions about '3.6' to '2.7'.

qin53027, 2021-12-17:
Hi Marty,

After I followed that link, I can build the Librealsense2 SDK with Python2.7 on Jetson NX. For example, in the python, I can do:
import pyrealsense2.

But if I run:  realsense-viewer:   it shows me the RS2_USB_STATUS_BUSY error.

If I run python program using the Realsense camera,  it shows No device connected error and failed to set power state error.

Then I did following, but still have the same error.

 sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/99-realsense-libusb.rules && sudo udevadm control --reload-rules && udevadm trigger.

Now I  am ordering a USB 3 hub since D435i uses the USB-3 port and will try that.

When I did regular install on SDK 2.40 version with Python3.6 on Jetson NX (which has no USB-3 port) and Jetson Xavier (which has the USB-3 port),   for both cases, the Intel D435i works very well. 

Just for the Python2.7 installation gives me a lot of trouble.

Any suggestions?

Thanks
Jimmy

qin53027, 2021-12-17:
Hi Marty,

Never mind for this now since I have figured out the problem.

Actually the D435i camera is connected to Robot MCU first and then back to the Jetson NX.  When I install the Librealsense SDK on Jetson NX, I only connect the external power supply to NX. In other words, the D435i did not get the power when I do the D435i test.

After turn on the Robot power button which also supply the power to Jetson NX, then the D435i camera works very well.

So  for both Python2.7 and Python3.6, the Librealsense2 SDK (version 2.40) works for all Jetson Xavier and NX.

Close this issue.

Thanks
Jimmy

closed 2021-12-17 by qin53027

MartyG-RealSense, 2021-12-18:
That's great to hear that you found a solution - thanks very much for the update and sharing the details of your solution with the RealSense ROS community   :)

qin53027, 2022-07-23:
> @MartyG-RealSense <https://github.com/MartyG-RealSense> That did the
> trick.
>
> For future reference, here is a complete list of steps I took to get the
> IntelRealsense Camara D455 to work with python3 on the Jetson Nano.
>
>    1.
>
>    Update CMake using this method
>    (You need to have curl installed
>    <https://stackoverflow.com/questions/56941778/cmake-use-system-curl-is-on-but-a-curl-is-not-found> for
>    this to work. )
>    2.
>
>    Download the zip file from
>    https://github.com/IntelRealSense/librealsense/releases/. I am using
>    version 2.38.1
>    3.
>
>    Extract the file, cd into the extracted file .
>    4.
>
>    Create a dir called build and cd into it.
>    5.
>
>    Run the CMake command to test to see if the build will work.
>    cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -
>    DPYTHON_EXECUTABLE=/usr/bin/python3.6 -DCMAKE_BUILD_TYPE=release
>    -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
>    -DBUILD_WITH_CUDA:bool=true
>
> If you have issues with the above step, check out this
> <https://github.com/jetsonhacks/buildLibrealsense2TX/issues/13>, this
>  and this
> <https://askubuntu.com/questions/620601/cant-install-libsdl2-dev-or-cmake-glfw-libxinerama-dev-problem>
>
>    1.
>
>    Still in the build dir. Run make -j4 and then sudo make install.
>    2.
>
>    Add these to the end of your .bashrc file
>
> export PATH=$PATH:~/.local/bin
> export PYTHONPATH=$PYTHONPATH:/usr/local/lib
> export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.6/pyrealsense2
>
>
>    1.
>
>    source your .bashrc file

For future reference, here is a complete list of steps I took to get the
IntelRealsense Camara D455 to work with python3 on the Jetson Nano.

   1.

   Update CMake using this method
   <https://github.com/IntelRealSense/librealsense/issues/6980#issuecomment-666858977>
   (You need to have curl installed
   <https://stackoverflow.com/questions/56941778/cmake-use-system-curl-is-on-but-a-curl-is-not-found>
for
   this to work. )
   2.

   Download the zip file from
   https://github.com/IntelRealSense/librealsense/releases/. I am using
   version 2.38.1
   3.

   Extract the file, cd into the extracted file .
   4.

   Create a dir called build and cd into it.
   5.

   Run the CMake command to test to see if the build will work.
   cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true
   -DPYTHON_EXECUTABLE=/usr/bin/python2.7 -DCMAKE_BUILD_TYPE=release
   -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
   -DBUILD_WITH_CUDA:bool=true

If you have issues with the above step, check out this
<https://github.com/jetsonhacks/buildLibrealsense2TX/issues/13>, this
<https://github.com/IntelRealSense/librealsense/issues/6449#issuecomment-650793097>
 and this
<https://askubuntu.com/questions/620601/cant-install-libsdl2-dev-or-cmake-glfw-libxinerama-dev-problem>

   1.

   Still in the build dir. Run make -j4 and then sudo make install.
   2.

   Add these to the end of your .bashrc file

export PATH=$PATH:~/.local/bin
export PYTHONPATH=$PYTHONPATH:/usr/local/lib
export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.6/pyrealsense2

MartyG-RealSense, 2022-07-23:
Thanks again @qin53027  :)

qin53027, 2022-10-11:
@MartyG-RealSense <https://github.com/MartyG-RealSense> That did the trick.

For future reference, here is a complete list of steps I took to get the
IntelRealsense Camara D455 to work with python3 on the Jetson Nano.

   1.

   Update CMake using this method
   <https://github.com/IntelRealSense/librealsense/issues/6980#issuecomment-666858977>
   (You need to have curl installed
   <https://stackoverflow.com/questions/56941778/cmake-use-system-curl-is-on-but-a-curl-is-not-found>
for
   this to work. )
   2.

   Download the zip file from
   https://github.com/IntelRealSense/librealsense/releases/. I am using
   version 2.38.1
   3.

   Extract the file, cd into the extracted file .
   4.

   Create a dir called build and cd into it.
   5.

   Run the CMake command to test to see if the build will work.
   cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -
   DPYTHON_EXECUTABLE=/usr/bin/python3.6 -DCMAKE_BUILD_TYPE=release
   -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
   -DBUILD_WITH_CUDA:bool=true

If you have issues with the above step, check out this
<https://github.com/jetsonhacks/buildLibrealsense2TX/issues/13>, this
<https://github.com/IntelRealSense/librealsense/issues/6449#issuecomment-650793097>
 and this
<https://askubuntu.com/questions/620601/cant-install-libsdl2-dev-or-cmake-glfw-libxinerama-dev-problem>

   1.

   Still in the build dir. Run make -j4 and then sudo make install.
   2.

   Add these to the end of your .bashrc file

export PATH=$PATH:~/.local/bin
export PYTHONPATH=$PYTHONPATH:/usr/local/lib
export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.6/pyrealsense2

   1.

   source your .bashrc file

MartyG-RealSense, 2022-10-11:
Hi @qin53027  Do you have a new question about this case, please?  Thanks!

qin53027, 2022-10-11:
Hi Marty, Problem is solved. thanks

MartyG-RealSense, 2022-10-11:
Thanks for the confirmation!