realsenseai/realsense-ros #2198: Point Cloud orientation

bhaskar-glitch, 2021-12-17:
Hi,
Im facing an issue with the point cloud of realsense of topic /camera/depth/color/points
in orientation of pointcloud is -y is upward...the problem is -y is facing toward z axis as in this picture
[image]

Please help

MartyG-RealSense, 2021-12-17:
Hi @bhaskar-glitch   Is your question a continuation of the earlier case https://github.com/IntelRealSense/realsense-ros/issues/2190 about orientation that you commented on, please?

bhaskar-glitch, 2021-12-17:
> Hi @bhaskar-glitch Is your question a continuation of the earlier case #2190 about orientation that you commented on, please?

Yah! Orientation problem is same but issue is different

MartyG-RealSense, 2021-12-18:
I researched this issue extensively again but like in https://github.com/IntelRealSense/realsense-ros/issues/2190 the answer still seems to be to use a static transform.  You can see an example of this in a guide at the link below regarding the RealSense T265 Tracking camera, where the T265's axes in its 'pose' coordinate system are Y up and Z forward, and there is a static transform performed between the pose_frame_id and the base_frame_id of the camera.

https://www.ross-robotics.co.uk/news/intel-realsense-t265-tracking-camera-for-mobile-robotics-first-impressions

MartyG-RealSense, 2021-12-24:
Hi @bhaskar-glitch  Do you require further assistance with this case, please?  Thanks!

bhaskar-glitch, 2021-12-24:
> Hi @bhaskar-glitch Do you require further assistance with this case, please? Thanks!

Yah! I haven't got any solution yet..let this issue opened so that anyone who knows the solution..will assist me

MartyG-RealSense, 2021-12-24:
Okay, I will leave this case open for a further time period so that people can see it.  Thanks for the update.

closed 2022-03-19 by MartyG-RealSense