realsenseai/realsense-ros #2199: L515 range differe between viewer and ros

JuliusSustarevas, 2021-12-17:
Hi,
Apologies if this is a silly question: 
I am using L515,ubuntu 18 and realsense2_camera rs_camera.launch
 
- Using the realsense_viewer I have configured the cam to short range mode and saved the configs to JSON. In realsense_viewer the short range looks fine I'm getting about 25-20cm minimal depth. 
- I then used the realsense2_camera rs_camera.launch and set  json_file_path to the short range json. The output says JSON is loaded successfully. 

But when viewing on rviz - it definitely is not short range. I am not getting the same depth.

Am I doing something wrong?

MartyG-RealSense, 2021-12-18:
Hi @JuliusSustarevas  If you are seeing a minimum depth sensing distance of 20 to 25 cm then this is likely because of the built-in default minimum depth range of the L515 camera model rather than a setting applied by the Short Range preset.  The information in the image below is extracted from the listing for the L515 in the official RealSense Store.

[image]

The primary effects that the Short Range preset has compared to other presets is to reduce the values of the **Laser Power** and the **Receiver Gain** settings of the L515 camera.  The changes to these settings improves short-range sensing by helping to prevent close-range objects from over-saturating the camera's receiver.

Could you provide information please about the ways in which the RViz depth image is different to the one that you experience in the RealSense Viewer, please?  Thanks!

JuliusSustarevas, 2021-12-18:
I still get ~20-25 cm with rviz if I use dynamic reconfigure to set short range preset. But if i use json_file_path it doesn't seem to work.  I'll try to take some screenshots if I can.

MartyG-RealSense, 2021-12-18:
Thanks very much - I look forward to receiving your images if it is possible to provide them.

JuliusSustarevas, 2021-12-18:
Hi So here is the rviz view:
1. call rs_camera like this:
```   <include file="$(find realsense2_camera)/launch/rs_camera.launch" >
      <arg name="device_type"         default="l515"/>
      <arg name="json_file_path"      default="$(find as_sensors)/config/l515_short_distance.json"/>
      <arg name="initial_reset"             default="true"/>
      <arg name="align_depth"             default="true"/>
      <arg name="filters"             default="pointcloud"/>
    </include>```
2. gif of aligned_depth_to_color:
![pre](https://user-images.githubusercontent.com/33488780/146646191-79b9ee9d-0ed5-4647-abcf-237af0f508f6.gif)

3.  use dynamic reconfigure GUI. When opening GUI says preset is max range.  Change to short range. 
4. git post change:
![post](https://user-images.githubusercontent.com/33488780/146646200-024c09ef-272f-4b09-bc6c-c02f61441efb.gif)

MartyG-RealSense, 2021-12-18:
How did you create the **l515_short_distance.json** file?  Did you select the Short Range preset in the RealSense Viewer and then export the configuration to a custom file called l515_short_distance.json using the json save option near the top of the Viewer's options side-panel?

JuliusSustarevas, 2021-12-18:
yes exactly like that

JuliusSustarevas, 2021-12-18:
could it be todo with inital hardware reset? I'll give it a try

closed 2021-12-30 by MartyG-RealSense