realsenseai/realsense-ros #2200: Aligned Depth Topic Rate is Underperforming

hakunaMatataHub, 2021-12-18:
I want to use this realsense ros wrapper with D435i on jetson Xavier nx. I have enabled aligned depth images to true and also using spatial, temporal and decimation filters with queue size 15 for my application. I have set the depth and color rgb streams to 15 fps. I always get the aligned depth stream at about 3fps. I tried changing queue size for filters but that didn't help either.

Any help is appreciated.

MartyG-RealSense, 2021-12-18:
Hi @hakunaMatataHub  Applying post-processing filters can be processing-intensive because they are processed on the CPU and not on the camera hardware.  Of the three filters mentioned, my understanding is that the **spatial** filter is the most demanding whilst not providing much additional benefit, so you could try leaving that filter out to see how it improves your FPS.

Alignment is also processing-intensive, though a Jetson board can make use of librealsense's CUDA support to offload the alignment processing from the CPU to the Nvidia graphics GPU.  The librealsense SDK has to be built with CUDA support included though.  

If you build librealsense from the SDK's Jetson packages and the RealSense ROS wrapper from source code with **Method 2** of the wrapper instructions then CUDA support should be included in the packages.  

If librealsense is built from source code then CUDA support is added to the build by including the flag **-BUILD_WITH_CUDA=true** in the CMake build instruction.

If the RealSense ROS wrapper and librealsense are built together from packages with **Method 1** of the wrapper instructions then CUDA support is **not** included in the packages.

*************

If you find that the depth and RGB streams do not both have the same FPS and one of them is lagging then having auto-exposure enabled and an RGB option called **auto-exposure priority** disabled can enforce a constant FPS instead of allowing the FPS to drift.  Instructions for disabling auto-exposure priority in the ROS wrapper can be found at https://github.com/IntelRealSense/librealsense/issues/10051#issuecomment-992740825

hakunaMatataHub, 2021-12-18:
@MartyG-RealSense  Thank you for swift response. I will definitely try out these suggestions. 

Also an update on the issue:: I did try turning all the post processing filters off except the align depth to color thingy. But still got 3fps in aligned depth stream?

MartyG-RealSense, 2021-12-18:
Thanks very much @hakunaMatataHub for the update - I look forward to your next report after testing.  Good luck!

hakunaMatataHub, 2021-12-19:
@MartyG-RealSense  Hey I built the library with CUDA but the fps with aligned depth stream did not improve
I tested it with decimation , spatial and temporal filters on with queue size of 15 and the fps for aligned depth stream was less than 2 fps.
In another test I turned off spatial and temporal but not the decimation and the aligned depth fps was slightly better approx 4 but not better as compared to running on cpu

MartyG-RealSense, 2021-12-19:
If you are referring to setting a filter queue size, Intel suggest setting it to '5000' in a script in the post-processing documentation linked to below.

https://dev.intelrealsense.com/docs/post-processing-filters#section-using-filters-in-application-code

[image]

hakunaMatataHub, 2021-12-19:
@MartyG-RealSense 
I am setting these parameters thorough a config file,

/camera/temporal/filter_smooth_alpha: 0.1
/camera/temporal/filter_smooth_delta: 100
/camera/temporal/frames_queue_size: 15
/camera/temporal/steam_filter: 1
/camera/spatial/frames_queue_size: 15
/camera/spatial/filter_smooth_alpha: 0.28
/camera/spatial/filter_smooth_delta: 17
/camera/decimation/frames_queue_size: 15
/camera/decimation/filter_magnitude: 2
/camera/disparity/frames_queue_size: 15
/camera/stereo_module/enable_auto_exposure: true

I am not sure the parameter that is 5000 does the same thing.

hakunaMatataHub, 2021-12-19:
@MartyG-RealSense  Also the same configuration for the camera and filters work on a cpu with 32 cores without any drop in fps.

MartyG-RealSense, 2021-12-19:
If **BUILD_WITH_OPENMP=true** is added to the CMake build instruction for building librealsense from source code then the SDK can take advantage of multiple CPU cores for depth-color alignment.  This reduces latency, though at the cost of higher CPU percentage usage.

I would not discount the possibility that it is a Jetson-specific issue though, as there are previous cases in the RealSense ROS wrapper where performance drops on Jetson when enabling alignment or the pointcloud filter but the same does not occur on desktop / laptop PCs running the same project.

A good example of pointcloud-related Jetson issues is https://github.com/IntelRealSense/realsense-ros/issues/1967 where a RealSense ROS user of Jetson found that a solution was to use the earlier librealsense version 2.43.0 

You mention that you are using librealsense 2.48.0.  Have you matched it with RealSense ROS wrapper **2.3.1** for ROS1 (Kinetic, Melodic, Noetic) or wrapper **3.2.2** for ROS2?  It is important to match specific librealsense and wrapper version combinations together.

hakunaMatataHub, 2021-12-20:
@MartyG-RealSense  yeah I verfied I am using ROS wrapper 2.3.1 with librealsense 2.48.0
Also I realised I am using pointcloud filters as well. 
I ran it with pointcloud, align depth and decimation turned on it gives me 3 fps in aligned depth stream. Is this expected?
 While if I turn off my pointcloud and only use decimation and align depth, the fps improves to 13 for aligned depth stream while the expected fps is 15 hz. 
 
 Does it mean that pointcloud filters can never output  high fps?

Also for your note I have built the lib realsense from source with CUDA.

MartyG-RealSense, 2021-12-25:
Hi @hakunaMatataHub  Do you require further assistance with this case, please?  Thanks!

hakunaMatataHub, 2021-12-27:
@MartyG-RealSense yeah I verfied I am using ROS wrapper 2.3.1 with librealsense 2.48.0
Also I realised I am using pointcloud filters as well.
I ran it with pointcloud, align depth and decimation turned on it gives me 3 fps in aligned depth stream. Is this expected?
While if I turn off my pointcloud and only use decimation and align depth, the fps improves to 13 for aligned depth stream while the expected fps is 15 hz.
Does it mean that pointcloud filters can never output high fps?
Also another update: I ran the same setup on a cpu with 32 cores with five filters, i.e pointcloud, decimation, spatial, temporal and align depth and the final aligned depth fps is as expected close to 15fps.
Also for your note I have built the lib realsense from source with CUDA.

MartyG-RealSense, 2021-12-27:
If CUDA support is enabled then CUDA should be able to provide acceleration for pointcloud and depth-color align operations.  It would not accelerate the decimation filter though.

As mentioned earlier in this discussion at https://github.com/IntelRealSense/realsense-ros/issues/2200#issuecomment-997378858 - Jetson board models are known for having problems such as slowdown in the RealSense ROS wrapper when the pointcloud filter is enabled, whilst working normally if pointcloud is disabled.  The pointcloud can function normally when generated on laptop and desktop PCs, as you found when trying the same setup on a computer with 32 cores.

Have you tried disabling **auto_exposure_priority** to see whether it improves FPS, as suggested earlier in this discussion?  If you haven't then you can test quickly whether it would make a difference by disabling it in the dynamic reconfigure interface after launch using the instructions at https://github.com/IntelRealSense/realsense-ros/issues/2142#issuecomment-953669299

closed 2022-01-08 by MartyG-RealSense