realsenseai/realsense-ros #2201: L515 can't publish pointcloud on Jetson AGX Xavier.

Divelix, 2021-12-22:
I've installed `realsence-ros` on Xavier via **Method 1** and have the following launch file:
```xml
<launch>
  <node pkg="tf" name="base_link_tf" type="static_transform_publisher" args="0 0 0 0 0 0 map base_link 100" />
  <node pkg="tf" name="frontal_camera_tf" type="static_transform_publisher" args="0.3751 0 0.1453 0 0 -1.5708 base_link frontal_camera_link 100" />

  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="camera" value="frontal_camera"/>
    <arg name="device_type" value="l515"/>
    <arg name="align_depth" value="true"/>
    
    <arg name="enable_color" value="true"/>
    <arg name="enable_depth" value="true"/>
    <arg name="filters" value="pointcloud"/>
  <rosparam>
      /frontal_camera/motion_module/global_time_enabled: true
      /frontal_camera/l500_depth_sensor/global_time_enabled: true
      /frontal_camera/rgb_camera/global_time_enabled: true
  </rosparam>
</launch>
```
When I launch it, I get full list of topics and services, but there are no messages published to topics I need (**RGB, depth, pointcloud**).
[image]

But, when I comment out line ``` <arg name="filters" value="pointcloud"/>``` - RGB and Depth topics are published well.

Version from logs:
[image]

**What I tried:**
- Plugged L515 into my laptop and launched file above -> **all 3 topics were published well**.
- Launched on Xavier with various resolutions and sensors -> everything works until I uncomment pointcloud filter.
- Run `realsense-viewer` on Xavier -> all sensors displayed well in 30 fps on all resolutions, loading CPU less than 40% on each core ~equally.

**What I noticed:**
- Once a century (mostly after robot reboot) on first launch all 3 topics are published well, but at low fps (~1-3 fps).
- Always, when I use filter pointcloud, Xavier CPU has 1 arbitrary core loaded at 100% (while `realsense-viewer` loads all cores equally)  (changes every few seconds)
[image]

MartyG-RealSense, 2021-12-22:
Hi @Divelix  I don't think that you have done anything wrong.  Your configuration looks fine and I do not believe that it is related to using an L515 camera model.  There is a known pattern of the RealSense ROS wrapper working fine on Jetson boards when the pointcloud filter is not applied but not providing the pointcloud when the filter is enabled.

An Xavier NX user who experienced this problem reported in https://github.com/IntelRealSense/realsense-ros/issues/1967#issuecomment-954814317 that they could generate a pointcloud if they used an earlier librealsense SDK version (2.43.0) and an earlier ROS wrapper version (2.2.23).  They also built librealsense with CUDA graphics acceleration support included.  

If librealsense and the ROS wrapper are built together with 'Method 1' then CUDA support is **not** included.  CUDA support can be added to librealsense on Jetson by one of the methods below:

*  Build librealsense first from Jetson packages using instructions in the link below and build the ROS wrapper secondly from source code.

*  Build librealsense first from source code and then build the ROS wrapper secondly from source code.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

Divelix, 2021-12-23:
I tried to follow your suggestion and install older version of SDK, but when I run `librealsense-2.43.0/scripts/patch-realsense-ubuntu-L4T.sh`, I get error: `Unsipported JetPack revision 32.5.2 aborting script`.

If I pass patching step and try to build SDK with `cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=false -DBUILD_WITH_CUDA=true`, I get error:
```
-- Could NOT find apriltag (missing: APRILTAG_INC APRILTAG_LIB) 
-- Unable to find apriltag library, skipping pose-apriltag example
-- Configuring incomplete, errors occurred!
See also ".../librealsense-2.43.0/build/CMakeFiles/CMakeOutput.log".
See also ".../librealsense-2.43.0/build/CMakeFiles/CMakeError.log".
```
How can I fix this?

MartyG-RealSense, 2021-12-23:
The most recent officially supported JetPack version supported by the librealsense SDK at the time of writing this is **32.5.1** rather than 32.5.2.  My understanding is that support for 32.5.1 was added in SDK version **2.45.0**.

A RealSense user recently contributed an update for JetPack **32.6.1** that was incorporated into librealsense in SDK version **2.50.0**.

https://github.com/IntelRealSense/librealsense/pull/9855

If you would prefer to continue using 2.43.0 then JetPack **32.5** was the most recently supported JetPack at that point (32.5 support was actually added in SDK 2.43.0).

Divelix, 2021-12-24:
Changing JetPack version sounds like a huge pain to me.
I just tried to install SDK 2.43.0 + ROS wrapper 2.2.23 in docker container and got error while building ROS wrapper with `catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release`:
```
/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp: In member function 'void realsense2_camera::BaseRealSenseNode::publishPointCloud(rs2::points, const ros::Time&, const rs2::frameset&)':
/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:2292:29: error: 'find_if' was not declared in this scope
         texture_frame_itr = find_if(frameset.begin(), frameset.end(), [&texture_source_id, &available_formats] (rs2::frame f)
                             ^~~~~~~
/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:2292:29: note: suggested alternatives:
In file included from /usr/include/c++/7/algorithm:62:0,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:39,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /opt/ros/noetic/include/class_loader/class_loader.hpp:36,
                 from /opt/ros/noetic/include/pluginlib/./class_list_macros.hpp:40,
                 from /opt/ros/noetic/include/pluginlib/class_list_macros.h:35,
                 from /catkin_ws/src/realsense-ros/realsense2_camera/src/../include/../include/realsense_node_factory.h:6,
                 from /catkin_ws/src/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1:
/usr/include/c++/7/bits/stl_algo.h:3923:5: note:   'std::find_if'
     find_if(_InputIterator __first, _InputIterator __last,
     ^~~~~~~
In file included from /usr/include/boost/mpl/find.hpp:17:0,
                 from /usr/include/boost/mpl/aux_/contains_impl.hpp:20,
                 from /usr/include/boost/mpl/contains.hpp:20,
                 from /usr/include/boost/math/policies/policy.hpp:10,
                 from /usr/include/boost/math/policies/error_handling.hpp:21,
                 from /usr/include/boost/math/special_functions/round.hpp:14,
                 from /opt/ros/noetic/include/ros/time.h:58,
                 from /opt/ros/noetic/include/ros/console.h:39,
                 from /opt/ros/noetic/include/nodelet/nodelet.h:40,
                 from /catkin_ws/src/realsense-ros/realsense2_camera/src/../include/../include/realsense_node_factory.h:7,
                 from /catkin_ws/src/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1:
/usr/include/boost/mpl/find_if.hpp:32:8: note:   'boost::mpl::find_if'
 struct find_if
        ^~~~~~~
realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:86: recipe for target 'realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o' failed
make[2]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:366: recipe for target 'realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/all' failed
make[1]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```
Looks like compatibility error of SDK with wrapper, according to [similar issue](https://github.com/IntelRealSense/realsense-ros/issues/548).
I also tried to install it on top of SDK 2.50.0 - the same error on wrapper build.

MartyG-RealSense, 2021-12-24:
My knowledge of Docker is limited unfortunately and the only RealSense ROS case about Docker that I know of that also features the L515 model is at https://github.com/IntelRealSense/realsense-ros/issues/1289 - my understanding is that the RealSense user in that case was using librealsense 2.42.0 or 2.43.0.  They were having similar Docker problems with D435 around the same time at https://github.com/IntelRealSense/realsense-ros/issues/1287

In regard to **2.50.0**, Intel published a Docker guide with that release.

https://github.com/IntelRealSense/librealsense/tree/master/scripts/Docker

MartyG-RealSense, 2021-12-30:
Hi @Divelix  Do you require further assistance with this case, please?  Thanks!

Divelix, 2021-12-30:
Yes, my current solution is to use latest versions of SDK and ROS wrapper + disable pointcloud filter and generate it from rgbd with [another package](http://wiki.ros.org/depth_image_proc), but I would like to use native SDK implementation.
It seems like the problem with pointcloud is caused by some bug in realsense-ros and it would be nice if you find and fix it in the next version.

MartyG-RealSense, 2021-12-30:
I will raise the issue with **Doronhi** the RealSense ROS wrapper developer.

MartyG-RealSense, 2022-01-05:
In response to discussion with Doronhi and Intel, an official Intel bug report has been filed so that the issue of pointcloud problems on Jetson boards can be investigated.  Thanks very much for your report, @Divelix

MartyG-RealSense, 2022-01-15:
Case closed due to no further comments received.  The Intel bug report about Jetson pointcloud problems that is associated with https://github.com/IntelRealSense/realsense-ros/issues/1967 will continue to be open.

closed 2022-01-15 by MartyG-RealSense