realsenseai/realsense-ros #2202: Complete OS hangs! Hardware Notification:Depth stream start failure,1.64023e+12,Error,Hardware Error

ravijo, 2021-12-23:
Hi,

Just now, I updated to new firmware and LibRealSense package too. After this upgrade, I am unable to use RealSense inside ROS. The complete OS hangs up and I have no choice but forcefully power off by pressing the power key!

```
$ roslaunch realsense2_camera rs_camera.launch enable_pointcloud:=True
... logging to /home/ravi/.ros/log/582373f2-639f-11ec-a389-44af2821c654/roslaunch-dell-10806.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dell:37617/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: True
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
[... 142 lines cut ...]
Dec 23 12:21:01 dell kernel: [  438.120281] uvcvideo 2-5.4.4:1.3: Entity type for entity Camera 6 was not initialized!
Dec 23 12:21:01 dell kernel: [  438.120633] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
Dec 23 12:21:01 dell kernel: [  438.122196] uvcvideo 2-5.4.4:1.3: Entity type for entity Processing 7 was not initialized!
Dec 23 12:21:01 dell kernel: [  438.122197] uvcvideo 2-5.4.4:1.3: Entity type for entity Extension 8 was not initialized!
Dec 23 12:21:01 dell kernel: [  438.122198] uvcvideo 2-5.4.4:1.3: Entity type for entity Camera 6 was not initialized!
Dec 23 12:21:01 dell kernel: [  438.123199] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
Dec 23 12:21:01 dell kernel: [  438.124767] uvcvideo 2-5.4.4:1.3: Entity type for entity Processing 7 was not initialized!
Dec 23 12:21:01 dell kernel: [  438.124768] uvcvideo 2-5.4.4:1.3: Entity type for entity Extension 8 was not initialized!
Dec 23 12:21:01 dell kernel: [  438.124769] uvcvideo 2-5.4.4:1.3: Entity type for entity Camera 6 was not initialized!
Dec 23 12:21:01 dell kernel: [  438.125143] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
Dec 23 12:21:01 dell kernel: [  438.126747] uvcvideo 2-5.4.4:1.3: Entity type for entity Processing 7 was not initialized!
Dec 23 12:21:01 dell kernel: [  438.126748] uvcvideo 2-5.4.4:1.3: Entity type for entity Extension 8 was not initialized!
Dec 23 12:21:01 dell kernel: [  438.126748] uvcvideo 2-5.4.4:1.3: Entity type for entity Camera 6 was not initialized!
Dec 23 12:21:02 dell kernel: [  438.233928] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
Dec 23 12:21:02 dell kernel: [  438.236462] uvcvideo 2-5.4.4:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
Dec 23 12:21:02 dell kernel: [  438.236463] uvcvideo 2-5.4.4:1.0: Entity type for entity Processing 2 was not initialized!
Dec 23 12:21:02 dell kernel: [  438.236464] uvcvideo 2-5.4.4:1.0: Entity type for entity Camera 1 was not initialized!
Dec 23 12:21:02 dell kernel: [  438.236516] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-5/2-5.4/2-5.4.4/2-5.4.4:1.0/input/input72
Dec 23 12:21:02 dell kernel: [  438.322031] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
Dec 23 12:21:02 dell kernel: [  438.324494] uvcvideo 2-5.4.4:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
Dec 23 12:21:02 dell kernel: [  438.324495] uvcvideo 2-5.4.4:1.0: Entity type for entity Processing 2 was not initialized!
Dec 23 12:21:02 dell kernel: [  438.324495] uvcvideo 2-5.4.4:1.0: Entity type for entity Camera 1 was not initialized!
Dec 23 12:21:02 dell kernel: [  438.324530] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-5/2-5.4/2-5.4.4/2-5.4.4:1.0/input/input73
Dec 23 12:21:03 dell kernel: [  439.158646] usb 2-5.4.4: Process 11856 (nodelet) called USBDEVFS_CLEAR_HALT for active endpoint 0x82
```

MartyG-RealSense, 2021-12-23:
Hi @ravijo   After the librealsense SDK version is updated, the RealSense ROS wrapper has to be built again afterwards.  Did you rebuild the wrapper, please?

ravijo, 2021-12-23:
@MartyG-RealSense 

Thanks a lot. I installed the wrapper via apt-get.

MartyG-RealSense, 2021-12-30:
Hi @ravijo  Do you require further assistance with this case, please?  Thanks!

ravijo, 2021-12-30:
Thanks a lot.

I happened to upgrade Ubuntu to 20.04 for a requirement and it automatically worked! Closing the issue now.

closed 2021-12-30 by ravijo

MartyG-RealSense, 2021-12-30:
Great to hear that you found a solution, @ravijo - thanks very much for the update   :)