realsenseai/realsense-ros #2203: Realsense-viewer unable to detect D435 but roslaunch realsense2_camera rs_camera.launch works fine. How can I make realsense-viewer work?

Gaurav37, 2021-12-26:
[image]

Gaurav37, 2021-12-26:
Im using Ros Melodic. Also it shows a warning before running D435.

[ INFO] [1640555131.198300848]: Running with LibRealSense v2.50.0
[ WARN] [1640555131.198341650]: ***************************************************
[ WARN] [1640555131.198349501]: ** running with a different librealsense version **
[ WARN] [1640555131.198355776]: ** than the one the wrapper was compiled with!   **
[ WARN] [1640555131.198363738]: ***************************************************
[ INFO] [1640555131.221124135]:  
[ INFO] [1640555131.241459814]: Device with serial number 135522075703 was found.

MartyG-RealSense, 2021-12-26:
Hi @Gaurav37  The ROS message suggests that the librealsense version has been changed since the ROS wrapper was last built.  When the librealsense version is updated, the ROS wrapper must be built again afterwards.  Then the warning should disappear.  The appropriate wrapper version to use with SDK 2.50.0 on Melodic is **2.3.2**.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.2

I note that your RealSense Viewer version in the image that you kindly provided is version 2.49.0 but you are using 2.50.0 in your RealSense ROS image.  Does the D435 camera also not appear in the 2.50.0 version of the RealSense Viewer, please?

Also, can you check whether the D435 camera appears if you try the actions below.

1.  Unplug the camera from the USB port, wait a couple of seconds and plug it back into the port. Repeat this up to 10 times if necessary before stopping if the camera still has not been detected.  If that does not work then:

2.  Check if the D435 is listed when left-clicking on the **Add Source** button at the top of the Viewer's options side-panel.

Gaurav37, 2021-12-29:
How can I upgrade realsense-viewer to 2.50?
Its 2.49 in my system and the actions suggested don't seem to work on 2.49. It doesn't show option to add D435. It just shows Load recorded sequence and add network device.

MartyG-RealSense, 2021-12-30:
If you installed the librealsense SDK on Linux using Debian packages then you can uninstall it before performing a new installation of 2.5.0 by using the instruction below:

**dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge**

If you installed librealsense from source code with CMake then you should be able to uninstall by going to the **build** folder and using this command:

**sudo make uninstall && make clean**

After librealsense 2.50.0 has been successfully installed, the matching RealSense ROS wrapper for it (2.3.2) should then be built afterwards.

Gaurav37, 2021-12-31:
Hey,
Now its **not publishing depth_registered node**.
After running:
```
roslaunch realsense2_camera rs_camera.launch filter:=pointcloud

```
**Rostopic list**:
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static

Gaurav37, 2021-12-31:
Roslaunch messages are still same.

[ WARN] [1640980641.242587664]: ***************************************************
[ WARN] [1640980641.242600566]: ** running with a different librealsense version **
[ WARN] [1640980641.242609288]: ** than the one the wrapper was compiled with!   **
[ WARN] [1640980641.242620062]: ***************************************************
[ INFO] [1640980641.261268762]:  
[ INFO] [1640980641.280801561]: Device with serial number 135522075703 was found.

[ INFO] [1640980641.280828718]: Device with physical ID 2-1-5 was found.
[ INFO] [1640980641.280836419]: Device with name Intel RealSense D435 was found.
[ INFO] [1640980641.281118449]: Device with port number 2-1 was found.
[ INFO] [1640980641.281136405]: Device USB type: 3.2
[ INFO] [1640980641.282326827]: getParameters...
[ INFO] [1640980641.317377812]: setupDevice...
[ INFO] [1640980641.317403121]: JSON file is not provided
[ INFO] [1640980641.317415543]: ROS Node Namespace: camera
[ INFO] [1640980641.317435795]: Device Name: Intel RealSense D435
[ INFO] [1640980641.317450724]: Device Serial No: 135522075703
[ INFO] [1640980641.317466570]: Device physical port: 2-1-5
[ INFO] [1640980641.317482396]: Device FW version: 05.12.15.50
[ INFO] [1640980641.317498123]: Device Product ID: 0x0B07
[ INFO] [1640980641.317512417]: Enable PointCloud: Off
[ INFO] [1640980641.317526920]: Align Depth: Off
[ INFO] [1640980641.317541386]: Sync Mode: Off
[ INFO] [1640980641.317577534]: Device Sensors: 
[ INFO] [1640980641.332914933]: Stereo Module was found.
[ INFO] [1640980641.344940238]: RGB Camera was found.
[ INFO] [1640980641.345016761]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1640980641.345057996]: num_filters: 0
[ INFO] [1640980641.345084723]: Setting Dynamic reconfig parameters.
[ INFO] [1640980641.826428479]: Done Setting Dynamic reconfig parameters.
[ INFO] [1640980641.827336869]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1640980641.828165216]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1640980641.828206840]: setupPublishers...
[ INFO] [1640980641.832429914]: Expected frequency for depth = 30.00000
[ INFO] [1640980641.882080417]: Expected frequency for color = 30.00000
[ INFO] [1640980641.911149225]: setupStreams...
 31/12 14:57:21,963 WARNING [139776808179456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[ INFO] [1640980642.013592269]: SELECTED BASE:Depth, 0
[ INFO] [1640980642.031232003]: RealSense Node Is Up!
[ WARN] [1640980642.081816290]: 
 31/12 14:57:22,082 WARNING [139776808179456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 31/12 14:57:22,133 WARNING [139776808179456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 31/12 14:57:22,184 WARNING [139776808179456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 31/12 14:57:22,234 WARNING [139776808179456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 31/12 14:57:22,409 WARNING [139776808179456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 31/12 14:57:22,459 WARNING [139776808179456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 31/12 14:57:22,612 WARNING [139776808179456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 31/12 14:57:22,662 WARNING [139776808179456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 31/12 14:57:22,814 WARNING [139776808179456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 31/12 14:57:22,865 WARNING [139776808179456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 31/12 14:57:23,016 WARNING [139776808179456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 31/12 14:57:23,066 WARNING [139776808179456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 31/12 14:57:31,828 WARNING [139776808179456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 31/12 14:57:31,878 WARNING [139776808179456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 31/12 14:57:54,948 WARNING [139776808179456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 31/12 14:57:54,998 WARNING [139776808179456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 31/12 14:59:05,102 WARNING [139776808179456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 31/12 14:59:58,512 ERROR [139776145483520] (ds5-options.cpp:88) Asic Temperature value is not valid!
 31/12 14:59:59,516 ERROR [139776145483520] (ds5-options.cpp:88) Asic Temperature value is not valid!
 31/12 15:00:00,519 ERROR [139776145483520] (ds5-options.cpp:88) Asic Temperature value is not valid!

Gaurav37, 2021-12-31:
However When I run:
```
roslaunch realsense2_camera demo_pointcloud.launch
```

I get the following as rostopic list and I can see topic for pointcloud.

/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/color/points
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/pointcloud/parameter_descriptions
/camera/pointcloud/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/clicked_point
/diagnostics
/initialpose
/move_base_simple/goal
/rosout
/rosout_agg
/tf
/tf_static

Gaurav37, 2022-01-01:
I tried dpkg command and removed catkin_ws packages of realsense and reinstalled by sudo apt-install ros-melodic-realsense2-camera and ros-melodic-realsense2-description but realsense-viewer doesn't work.

MartyG-RealSense, 2022-01-01:
So you are experiencing the same problem described originally in the title of this case - that the roslaunch works fine but the RealSense Viewer does not work?

And like the image at the top of this case, the Viewer is able to launch when using the **realsense-viewer** command but the camera is not listed in the options side-panel even though you have upgraded to librealsense version 2.50.0?

Gaurav37, 2022-01-01:
Yes exactly that.

MartyG-RealSense, 2022-01-02:
Is the camera detected in the Viewer if you unplug the micro-sized end of the USB cable from the side of the camera, turn the connector around the opposite way and re-insert it (USB-C cables are two-way insertion at the micro-sized end).  The 400 Series models are not usually affected by the direction that it is plugged into the camera, but it is worth trying in order to eliminate the possibility.

Since the RealSense ROS wrapper is able to work normally, it does not sound as though there is a problem with the camera hardware or firmware driver.

Is the camera able to be detected on the **Depth Quality Tool** program that can be launched with the command **rs-depth-quality**

closed 2022-01-14 by MartyG-RealSense