realsenseai/realsense-ros #2207: Error-"Failed to open scan_element" when using rs-ros2.3.2. with d455 librealsense_2.50 Nvidia NX

luobodan, 2021-12-31:
hello, when i was using d455 on NX,I got this:

process[camera/realsense2_camera_manager-2]: started with pid [30088]
process[camera/realsense2_camera-3]: started with pid [30089]
[ INFO] [1640924508.098420312]: Initializing nodelet with 6 worker threads.
[ INFO] [1640924509.351566683]: RealSense ROS v2.3.2
[ INFO] [1640924509.351703900]: Built with LibRealSense v2.50.0
[ INFO] [1640924509.351781469]: Running with LibRealSense v2.50.0
[ INFO] [1640924509.436313405]:  
[ INFO] [1640924509.487353286]: Device with serial number 030422250432 was found.

[ INFO] [1640924509.487558312]: Device with physical ID /sys/devices/3610000.xhci/usb2/2-3/2-3.2/2-3.2.4/2-3.2.4:1.0/video4linux/video0 was found.
[ INFO] [1640924509.487667017]: Device with name Intel RealSense D455 was found.
[ INFO] [1640924509.489561208]: Device with port number 2-3.2.4 was found.
[ INFO] [1640924509.489712570]: Device USB type: 3.2
[ INFO] [1640924509.510515687]: getParameters...
[ INFO] [1640924509.875386758]: setupDevice...
[ INFO] [1640924509.875572328]: JSON file is not provided
[ INFO] [1640924509.875663912]: ROS Node Namespace: camera
[ INFO] [1640924509.875768873]: Device Name: Intel RealSense D455
[ INFO] [1640924509.875845098]: Device Serial No: 030422250432
[ INFO] [1640924509.876575792]: Device physical port: /sys/devices/3610000.xhci/usb2/2-3/2-3.2/2-3.2.4/2-3.2.4:1.0/video4linux/video0
[ INFO] [1640924509.876739473]: Device FW version: 05.13.00.50
[ INFO] [1640924509.876894419]: Device Product ID: 0x0B5C
[ INFO] [1640924509.876999187]: Enable PointCloud: Off
[ INFO] [1640924509.877072948]: Align Depth: On
[ INFO] [1640924509.877149333]: Sync Mode: On
[ INFO] [1640924509.877542808]: Device Sensors: 
[ INFO] [1640924509.999884819]: Stereo Module was found.
[ INFO] [1640924510.073989212]: RGB Camera was found.
[ INFO] [1640924510.077203863]: Motion Module was found.
[ INFO] [1640924510.087411148]: num_filters: 1
[ INFO] [1640924510.087612238]: Setting Dynamic reconfig parameters.
[ WARN] [1640924510.348686061]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1640924510.407023603]: Done Setting Dynamic reconfig parameters.
[ INFO] [1640924510.409828779]: infra1 stream is enabled - width: 640, height: 360, fps: 15, Format: Y8
[ INFO] [1640924510.411073269]: infra2 stream is enabled - width: 640, height: 360, fps: 15, Format: Y8
[ INFO] [1640924510.413629803]: gyro stream is enabled - fps: 400
[ INFO] [1640924510.413863180]: accel stream is enabled - fps: 250
[ INFO] [1640924510.414038894]: setupPublishers...
[ INFO] [1640924510.431955107]: Expected frequency for infra1 = 15.00000
[ INFO] [1640924510.609805101]: Expected frequency for infra2 = 15.00000
[ INFO] [1640924510.731365474]: Start publisher IMU
[ INFO] [1640924510.734232122]: setupStreams...
 30/12 23:21:50,763 WARNING [547759305088] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 30/12 23:21:50,784 WARNING [547373429120] (backend-hid.h:48) write_fs_attribute with accel_3d-dev1 failed. The specified path /sys/devices/3610000.xhci/usb2/2-3/2-3.2/2-3.2.4/2-3.2.4:1.5/0003:8086:0B5C.000C/HID-SENSOR-200073.1.auto/iio:device1/trigger/current_trigger is not valid
[ERROR] [1640924510.786980049]: An exception has been thrown: Failed to open scan_element /sys/devices/3610000.xhci/usb2/2-3/2-3.2/2-3.2.4/2-3.2.4:1.5/0003:8086:0B5C.000C/HID-SENSOR-200073.1.auto/iio:device1/scan_elements/in_accel_z_en Last Error: Permission denied
[ERROR] [1640924510.787332788]: Exception: Failed to open scan_element /sys/devices/3610000.xhci/usb2/2-3/2-3.2/2-3.2.4/2-3.2.4:1.5/0003:8086:0B5C.000C/HID-SENSOR-200073.1.auto/iio:device1/scan_elements/in_accel_z_en Last Error: Permission denied
 30/12 23:21:50,785 ERROR [547759305088] (librealsense-exception.h:52) Failed to open scan_element /sys/devices/3610000.xhci/usb2/2-3/2-3.2/2-3.2.4/2-3.2.4:1.5/0003:8086:0B5C.000C/HID-SENSOR-200073.1.auto/iio:device1/scan_elements/in_accel_z_en Last Error: Permission denied
 30/12 23:21:50,786 ERROR [547759305088] (backend-hid.cpp:1003) Hid device is busy!

I am just a new user with realsense and d455. please ! I dont know how to solve this..

MartyG-RealSense, 2021-12-31:
Hi @luobodan  The **Failed to open scan_element** error can indicate that the **udev rules** are not installed or improperly installed, as described in https://github.com/IntelRealSense/librealsense/issues/3242#issuecomment-464775017

MartyG-RealSense, 2022-01-13:
Case closed due to no further tickets received.

closed 2022-01-13 by MartyG-RealSense

luobodan, 2022-01-18:
Thank you very much! I change the dev rules and make it very well

MartyG-RealSense, 2022-01-18:
That is excellent news. @luobodan - thanks very much for the update!