realsenseai/realsense-ros #2208: (roslaunch realsense2_camera rs_camera.launch) have some warning

gaoyi-2016, 2021-12-31:
[image]

MartyG-RealSense, 2021-12-31:
Hi @gaoyi-2016  You can ignore that warning as it is not something to be concerned about.

gaoyi-2016, 2021-12-31:
ok,thanks,but there is a problem i have 
sometimes /camera/aligned_depth_to_color/image_raw    and     /camera/color/image_raw  number is same,sometimes the number is different,can u help me
[image]

gaoyi-2016, 2021-12-31:
@MartyG-RealSense

gaoyi-2016, 2021-12-31:
the problem is staying when using rosbag  data
[image]
[image]

gaoyi-2016, 2021-12-31:
my  system is ubuntu18.04 , realsense2.40.0 , realsensense-ros-2.2.20
when i static d435i and reinforce usb cable,the number is same,
but when i move d435i and usb cable will be loose,the number is different,
i doubt my usb cable is not totally in the rabbet,
can u help me,thanks

MartyG-RealSense, 2021-12-31:
If you consistently get the same number when not moving but the numbers vary if the cable moves, you could consider replacing the USB cable with a 'screw lock' one.  On the side of the camera there are small holes either side of the USB port for screwing a screw-lock connector in firmly, as shown in https://github.com/IntelRealSense/librealsense/issues/3118#issuecomment-456392939

gaoyi-2016, 2021-12-31:
ok,i read your answer and think it will help me truely,thanks

MartyG-RealSense, 2021-12-31:
You are very welcome, @gaoyi-2016   :)

gaoyi-2016, 2021-12-31:
Ok,i have purchased a USB-C 3 screw-lock cable and i think it will help me.
its a good experience,i will keep supporting inter and hope tomorrow you have a lucky year.

MartyG-RealSense, 2022-01-06:
Hi @gaoyi-2016  Do you require further assistance with this case, please?  Thanks!

gaoyi-2016, 2022-01-07:
I am sorry to reply late.
I have solved this problem  in my another issue#10127,you could close this issue.
Thanks for your help.

MartyG-RealSense, 2022-01-07:
Thanks very much @gaoyi-2016 for the update.  As you suggested, I will close the issue.  Thanks again!

closed 2022-01-07 by MartyG-RealSense