realsenseai/realsense-ros #2209: rosrun pcl_ros pointcloud_to_pcd input:/camera/depth/color/points does not save files after roslaunch realsense2_camera demo_pointcloud.launch

Gaurav37, 2021-12-31:
After I run 
``` roslaunch realsense2_camera demo_pointcloud.launch ```
Rviz opens with a pointcloud image
[image]
.

Then I run 
```
rosrun pcl_ros pointcloud_to_pcd input:/camera/depth/color/points
```
Output is 
[ INFO] [1640994236.548494422]: Saving as ASCII PCD
[ INFO] [1640994236.551086164]: Listening for incoming data on topic /input
Then nothing happens
I have to press Ctrl C and then I see no files generated.

MartyG-RealSense, 2022-01-01:
Hi @Gaurav37  It looks as though your pointcloud_to_pcd command is missing an equals sign from its input.  Please try the command below.

**rosrun pcl_ros pointcloud_to_pcd input:=/camera/depth/color/points**

Gaurav37, 2022-01-01:
Hey Thanks @MartyG-RealSense Apologies for silly mistake :)

closed 2022-01-01 by Gaurav37

MartyG-RealSense, 2022-01-02:
No problem at all - thanks for the update!