realsenseai/realsense-ros #2210: Meeting warning of "Depth stream start failure" and error of "Asic Temperature value is not valid!"

J1dan, 2022-01-01:
I meet a warn and an error when I played "roslaunch realsense2_camera rs_camera.launch" which is shown below,

 [ WARN] [1641036019.228018001]: Hardware Notification:Depth stream start failure,1.64104e+12,Error,Hardware Error
[ INFO] [1641036035.980228819]: publishing unordered pointcloud.
 01/01 19:21:25,836 ERROR [546753634688] (ds5-options.cpp:88) Asic Temperature value is not valid!
 01/01 19:21:26,840 ERROR [546753634688] (ds5-options.cpp:88) Asic Temperature value is not valid!
 01/01 19:21:27,843 ERROR [546753634688] (ds5-options.cpp:88) Asic Temperature value is not valid!

The error just keeps coming out but I don't know what's wrong. Despite these, I could still view correct image in Rviz and I didn't meet other problems. Could you please help me with it? Thanks a lot!

MartyG-RealSense, 2022-01-01:
Hi @J1dan  If the **Asic Temperature value is not valid!** warnings are not negatively affecting the image in RViz then they can be ignored.  

The librealsense SDK periodically performs a check of the camera's current internal temperatures.  If librealsense is not ready to provide a temperature reading at the instant when it is requested by an application then it returns this warning.

MartyG-RealSense, 2022-01-13:
Case closed due to no further tickets received.

closed 2022-01-13 by MartyG-RealSense

joshuawillman, 2022-01-20:
Hello, 

As for this warning (as I get the same on the D435i I am using), what does it mean? Will I be unable to get accurate depth measurements? 
[ WARN] [1641036019.228018001]: Hardware Notification:Depth stream start failure,1.64104e+12,Error,Hardware Error

I ask because I am using ROS and my depth measurements are inaccurate. For example, an object I want to detect is placed roughly 80 cm in front of the camera. When using the camera outside of ROS, I get this value. However, I get roughly 13 cm in ROS. 

I can share more details if this does is not causing my issue. Thanks for the help!

MartyG-RealSense, 2022-01-20:
Hi @joshuawillman   The warning **Depth stream start failure** literally means that the depth stream was unable to be started for some reason.  It sounds as though the camera hardware is able to function though if you can receive depth information, and receive correct values outside of ROS.

Do your results in ROS improve if you add **initial_reset:=true** to your roslaunch instruction to reset the camera at launch?

joshuawillman, 2022-01-22:
Hi @MartyG-RealSense,

Thanks so much for the update! Despite the warning, I am still able to get depth info. 

As for your suggestion, adding `initial_reset:=True` to my roslaunch command produces the following error:
```
[ERROR] [1642824551.289393853]: An exception has been thrown: 
Failed to resolve the request: 
	Format: RGB8, width: 1280, height: 720

Into:
failed to set power state
[ERROR] [1642824551.289566945]: Exception: 
Failed to resolve the request: 
	Format: RGB8, width: 1280, height: 720

Into:
failed to set power state
```
After that, nothing launches. The command I use is:
`roslaunch realsense2_camera rs_camera.launch depth_width:=1280 depth_height:=720 depth_fps:=30 color_width:=1280 color_height:=720 color_fps:=30 enable_sync:=True align_depth:=True initial_reset:=True`

Any idea why reset fails? Not sure if my command is the cause of the reset failing. 
Thanks for your help!

joshuawillman, 2022-01-22:
As a side note, re-running the command in my previous reply no longer produces the error I mentioned. However, the original warning (`[ WARN] [1641036019.228018001]: Hardware Notification:Depth stream start failure,1.64104e+12,Error,Hardware Error`) still appears.

MartyG-RealSense, 2022-01-22:
It might generate the **Failed to set power state** message if it thinks that the camera is already active and busy when the reset attempts to access it.

Another possibility is that upon the reset the camera is mis-detected as being on a USB 2.1 connection (even if it is plugged into a USB 3 port), in which case it would be unable to provide your custom configuration of 30 FPS at 1280x720 because on a USB 2 connection with a D435i camera it could only provide a maximum of 6 FPS depth and 15 FPS RGB at that resolution.