realsenseai/realsense-ros #2211: intel realsense underwater calibration issue

marwanka91, 2022-01-01:
Hi
we would like to use D415 to calculate the dimensions of objects underwater.

The depth measurement doesn’t work as expected underwater so we tried to calibrate the camera but without success 

I tried to calibrate it using intel dynamic calibration tool but after finishing all the progress I got err msg, I don’t remember the error exactly but it was something like the parameters out of range  

Also, I tried calibrating it using intel custom calibration tool and got the following err 
 Error (2003): right camera intrinsics unreasonable.

 CalibrationParameters.xml :
<?xml version="1.0"?>
<Config>
    <param name = "ResolutionLeftRight">
        <value>1920</value>
        <value>1080</value>
    </param>
    <param name = "FocalLengthLeft">
        <value>1779.58</value>
        <value>1779.58</value>
    </param>
    <param name = "PrincipalPointLeft">
        <value>933.349</value>
        <value>528.637</value>
    </param>
    <param name = "DistortionLeft">
        <value>0.561952</value>
        <value>0.0523446</value>
        <value>-0.0115425</value>
        <value>-0.0173872</value>
        <value>-0.400511</value>
    </param>
    <param name = "FocalLengthRight">
        <value>1784.27</value>
        <value>1784.27</value>
    </param>
    <param name = "PrincipalPointRight">
        <value>1026.14</value>
        <value>531.436</value>
    </param>
    <param name = "DistortionRight">
        <value>0.553344</value>
        <value>0.577333</value>
        <value>-0.00462277</value>
        <value>0.0285169</value>
        <value>-3.39496</value>
    </param>
    <param name = "RotationLeftRight">
        <value>0.998673</value>
        <value>-0.000703848</value>
        <value>-0.0515042</value>
        <value>0.000638708</value>
        <value>0.999999</value>
        <value>-0.00128121</value>
        <value>0.0515051</value>
        <value>0.00124661</value>
        <value>0.998672</value>
    </param>
    <param name = "TranslationLeftRight">
        <value>-57.1089</value>
        <value>0.232227</value>
        <value>-0.87117</value>
    </param>
    <param name = "HasRGB">
        <value>1</value>
    </param>
    <param name = "ResolutionRGB">
        <value>1920</value>
        <value>1080</value>
    </param>
    <param name = "FocalLengthRGB">
        <value>1885.02</value>
        <value>1885.02</value>
    </param>
    <param name = "PrincipalPointRGB">
        <value>926.624</value>
        <value>532.308</value>
    </param>
    <param name = "DistortionRGB">
        <value>0.65894</value>
        <value>-0.749079</value>
        <value>-0.00991082</value>
        <value>-0.0188965</value>
        <value>0.0607226</value>
    </param>
    <param name = "RotationLeftRGB">
        <value>0.999998</value>
        <value>-0.000232411</value>
        <value>0.00197259</value>
        <value>0.000234087</value>
        <value>1</value>
        <value>-0.000849467</value>
        <value>-0.00197239</value>
        <value>0.000849927</value>
        <value>0.999998</value>
    </param>
    <param name = "TranslationLeftRGB">
        <value>15.3768</value>
        <value>-0.431579</value>
        <value>4.3694</value>
    </param>
</Config>

Also, we tried to buy intel OEM calibration target but it's out of stock :(

Can anyone help please

MartyG-RealSense, 2022-01-01:
Hi @marwanka91  This question has been responded to at the version of it at the link below.  Let's carry out the discussion at that location.  Thanks!

MartyG-RealSense, 2022-01-08:
Case closed due to no further comments received here and case continued at the other support forum.

closed 2022-01-08 by MartyG-RealSense