realsenseai/realsense-ros #2212: Can not get imu data (realsense D455, Jetson nano)

y-furuta-hq, 2022-01-04:
Camera model: D455
OS: Ubuntu 18.04 (Nvidia Jetpack 4.6)
Platform: Nvidia Jetson nano 4GB
realsense-ros: v2.3.2
librealsense: v2.50.0
D455 firmware: 5.13.0.50

I'm trying to obtain orientation data from the D455 using the topic camera/imu while running the launch command ros2 launch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation or opensource_tracking.launch.but I could not receive imu data.
I can receive 2D image or depth image but imu (accel, gyro).
There were some warning when I started launch file, no error.

Does anybody have any idea what this is?

MartyG-RealSense, 2022-01-04:
Hi @y-furuta-hq  May I first confirm which roslaunch instruction you are using.  You are using the ROS1 wrapper 2.3.2 and your roslaunch instruction seems to mostly be a ROS1 one.  However, the instruction starts with **ros2 launch**.  If you were using ROS1 (Kinetic, Melodic or Noetic) then the roslaunch instruction should look like the one below.

**roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation**

y-furuta-hq, 2022-01-04:
Hi @MartyG-RealSense , Thank you for your comment, but sorry I made a mistake writing ros2 launch. I use ros1 (melodic). so it is not well after your written command (roslaunch realsense2......).

MartyG-RealSense, 2022-01-04:
So you receive no messages from the /camera/imu topic if you check whether they are being published using **rostopic echo /camera/imu**

y-furuta-hq, 2022-01-04:
Hi @MartyG-RealSense, there was nothing though I tried "rostopic echo /camera/imu". but topic "/camera/imu" exists when I tried rostopic list.

MartyG-RealSense, 2022-01-04:
Are there still no IMU messages published if you use **unite_imu_method:=copy** instead of linear_interpolation

MartyG-RealSense, 2022-01-04:
Thanks very much for your patience.  Let's test if it is able to publish IMU data at all.  Leave out the **unite_imu_method** command from the roslaunch this time so that IMU gyro and accel are published as separate topics.  Then echo the gyro topic:

**rostopic echo /camera/gyro/imu_info**

y-furuta-hq, 2022-01-04:
Thanks @MartyG-RealSense , it was below

[image]

MartyG-RealSense, 2022-01-04:
Thanks very much.  Next, add **initial_reset:=true** to the roslaunch instruction to reset the camera at launch to see whether or not this positively effects IMU publishing.

If that does not work, would it be possible to run the RealSense Viewer program with the **realsense-viewer** command in the Ubuntu terminal to see whether the IMU data can be streamed by enabling the **Motion Module**

y-furuta-hq, 2022-01-04:
@MartyG-RealSense  thanks a lot.
1. initial_reset:=true adding:    same as before.
2. reasense-viewer:    motioin module now wroking (attached picture), but RGB and Stereo camera worked good.

[image]

y-furuta-hq, 2022-01-04:
@MartyG-RealSense , I made a mistake writing "motioin module now wroking". It is correct "not working".

MartyG-RealSense, 2022-01-04:
Thanks, yes I guessed.  :)

Let's confirm that the camera's IMU streaming has a problem by running **rs-motion** in the Ubuntu terminal.

y-furuta-hq, 2022-01-04:
@MartyG-RealSense Thanks.
A Realsense displayed on rs-motion but did not move though I moved the realsenseD455.

MartyG-RealSense, 2022-01-04:
Okay, thanks again for your testing.  At this point I will seek advice from Intel about your case and update you when I have feedback to provide.

y-furuta-hq, 2022-01-04:
@MartyG-RealSense Ok, Thank you for your kindness.

MartyG-RealSense, 2022-01-04:
I received feedback from Intel.  They sked whether it would be possible for you to test the camera on a Windows PC to eliminate the possibility that the lack of IMU data is a Jetson-specific issue.

y-furuta-hq, 2022-01-05:
Thank you @MartyG-RealSense,
I can watch motion module moving (IMU data) very fine by using realsense-viewer on a Windows PC, presumably it is a Jetson specific issue.

MartyG-RealSense, 2022-01-05:
On a past case where IMU data was unavailable, a RealSense user commented today that a solution that worked in their particular case was to disable secure boot.

https://github.com/IntelRealSense/librealsense/issues/9346#issuecomment-1005428886

y-furuta-hq, 2022-01-06:
@MartyG-RealSense  Sorry for  the delay.
I attached the image of the launch log.
There is a Hardware Notification:Motion Module failure error . 

[image]

MartyG-RealSense, 2022-01-06:
Hi @y-furuta-hq  Thanks very much for the update.  I received feedback from Intel about your IMU problem on Jetson.  They thought that when you built librealsense, you might not have applied the kernel patch to enable the IMU.  A patch for JetPack **4.6.1** (L4T 32.6.1)  was released as part of librealsense version 2.50.0 (you have **4.6** listed as your JetPack version at the top of this case).  

Instructions for applying the kernel patch on Jetson are provided in the **Building from Source using Native Backend** section of Intel's Jetson installation guide.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend

Jayadev22, 2022-01-06:
Hallo @y-furuta-hq and @MartyG-RealSense 
Even I faced the same issue as soon as `apt` updated `ros-melodic-librealsense2`. And it took a while for me to realize this as `realsense-ros` 2.3.1 that I had submodule-d started throwing compilation errors, which was absolutely strange because I've been using this very setup for quite some time now. Then, I went ahead and updated `realsense-ros` to 2.3.2 to get it successfully compiled but face this IMU related issue.
For now, I removed `apt`'s `ros-melodic-librealsense2` (v2.50.0) to build & install `librealsense` v2.48.0 from source. With this I can now continue using `realsense-ros` 2.3.1, which seems to be more stable and has no issues as such.
I hope even @y-furuta-hq can maybe roll back the versions to get the sensor going again.

MartyG-RealSense, 2022-01-07:
Thanks very much @Jayadev22 for your advice to @y-furuta-hq  :)

MartyG-RealSense, 2022-01-18:
Hi @y-furuta-hq  Do you require further assistance with this case, please after the advice provided by @Jayadev22 in https://github.com/IntelRealSense/realsense-ros/issues/2212#issuecomment-1007003514  

Thanks!

y-furuta-hq, 2022-01-18:
Thank you very much @MartyG-RealSense and @Jayadev22, I am so sorry late response. I will try your method and let you know after I reviewed it.

MartyG-RealSense, 2022-01-18:
It's no problem at all, @y-furuta-hq - I look forward to your test results.  Good luck!

MartyG-RealSense, 2022-02-11:
Case closed due to no further comments received.

closed 2022-02-11 by MartyG-RealSense

Juanito970, 2022-05-10:
Hello, I am having the very same problem, IMU Not publishing on Jetson, but it does publish ok on other Ubuntu PCs.

MartyG-RealSense, 2022-05-10:
Hi @Juanito970  Which RealSense ROS wrapper version are you using, please?

Juanito970, 2022-05-10:
I am using 2.3.2, followed [this](https://github.com/IntelRealSense/realsense-ros) instructions for the wrapper, and [this](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) instructions for the librealsense, both downloaded with apt.

MartyG-RealSense, 2022-05-10:
When installing with apt-get using the ROS wrapper's **Method 1**, it installs both librealsense and the ROS wrapper at the same time.  This method is not suitable for Jetson as the Method 1 packages do not include CUDA support.

If you then installed librealsense on its own afterwards with the Jetson page's packages then you have installed librealsense twice.  Even if a conflict does not arise from this then it introduces a further complication.  Each time that the librealsense SDK is updated, the ROS wrapper has to be built again afterwards.

The ideal installation approach for Jetson would therefore be to install librealsense first on its own using the Jetson page's packages, and then secondly build the ROS wrapper from **source code** after that once the librealsense installation is confirmed to be working successfully.

********

Alternatively, if you are not comfortable with building from source code then the less ideal approach would be to build librealsense and the wrapper together from packages, and do not use the Jetson page's packages after that.  This will mean that CUDA support will not be included and so you will not gain a performance advantage when using depth-color alignment or point cloud.  If you do not plan to use those features though then a Method 1 build may not be a problem.