realsenseai/realsense-ros #2213: D435i is not providing IMU data on Jetson Xavier NX

haquebd, 2022-01-05:
My goal is to use the D435i in Rtabmap with camera depth and IMU. I also tried with unite_imu_method:=linear_interpolation But still the same error. I dont see any "rostopic as /camera/imu"

The camera is connected with the supplied USB cable directly connected with the USB port in Jetson Xavier NX. 

**realsense_ros version 2.3.2 or latest and librealsense is 2.50.x**
The Jetpack and Ubuntu info as below:

ROS: Melodic
Ubuntu: 18.04
Jetson Package: nvidia-jetpack
Version: 4.6-b199
Architecture: arm64
Maintainer: NVIDIA Corporation
Installed-Size: 194
Depends: nvidia-cuda (= 4.6-b199), nvidia-opencv (= 4.6-b199), nvidia-cudnn8 (= 4.6-b199), nvidia-tensorrt (= 4.6-b199), nvidia-visionworks (= 4.6-b199), nvidia-container (= 4.6-b199), nvidia-vpi (= 4.6-b199), nvidia-l4t-jetson-multimedia-api (>> 32.6-0), nvidia-l4t-jetson-multimedia-api (<< 32.7-0)

`remote-eng@xavier-nx:~$ roslaunch realsense2_camera rs_camera.launch initial_reset:=true align_depth:=true filters:=pointcloud depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30 unite_imu_method:=copy enable_gyro:=true enable_accel:=true gyro_fps:=250 accel_fps:=200
... logging to /home/remote-eng/.ros/log/9c1d2e46-6dda-11ec-b7ba-d8c0a6b11fcf/roslaunch-xavier-nx-23024.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://xavier-nx:42683/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 200
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
[... 106 lines cut ...]
[ WARN] [1641354639.244338662]: Using default profile instead.
[ INFO] [1641354639.244465191]: gyro stream is enabled - fps: 200
[ WARN] [1641354639.244642248]: No mathcing profile found for accel with fps=200
[ WARN] [1641354639.244744328]: Using default profile instead.
[ INFO] [1641354639.244831689]: accel stream is enabled - fps: 63
[ INFO] [1641354639.244914953]: setupPublishers...
[ INFO] [1641354639.258937868]: Expected frequency for depth = 30.00000
[ INFO] [1641354639.430975739]: Expected frequency for color = 30.00000
[ INFO] [1641354639.545222777]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1641354639.653979612]: Start publisher IMU
[ INFO] [1641354639.659942809]: setupStreams...
 04/01 22:50:39,837 WARNING [547035201920] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ WARN] [1641354640.217446699]: 
[ WARN] [1641354640.320831510]: Hardware Notification:Motion Module failure,1.64135e+12,Error,Hardware Error
 04/01 22:50:40,484 WARNING [546459083136] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1641354640.535796144]: SELECTED BASE:Depth, 0
[ INFO] [1641354640.596498480]: RealSense Node Is Up!
 04/01 22:50:40,608 WARNING [546459083136] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 04/01 22:50:40,658 WARNING [546459083136] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 04/01 22:50:40,709 WARNING [546459083136] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 04/01 22:50:40,760 WARNING [546459083136] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 04/01 22:50:41,031 WARNING [546459083136] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 04/01 22:50:41,081 WARNING [546459083136] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 04/01 22:50:42,708 ERROR [546056429952] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
`

MartyG-RealSense, 2022-01-05:
Hi @haquebd  There was a similar sounding case yesterday at https://github.com/IntelRealSense/realsense-ros/issues/2212 where a Jetson user with a D455 could not access the IMU in RealSense ROS.  They also found that the IMU data was not accessible in librealsense programs such as realsense-viewer and rs-motion.  They were able to confirm the presence of the **/camera/imu** topic using rostopic list in their particular case though.  The IMU worked normally for them on a Windows computer, indicating that it was a Jetson-specific issue.

An important difference in your own case is that you are including **align_depth:=true filters:=pointcloud** in your roslaunch instruction.  Align and pointcloud instructions are known to currently have problems when used in the ROS wrapper on Jetson boards specifically (they work correctly on desktop / laptop PCs).  An official Intel bug report has been filed about this Jetson issue.

Could you try removing **align_depth:=true filters:=pointcloud** from your roslaunch instruction please in order to eliminate the possibility that your IMU access problem is being caused by these commands, please?  Thanks!

haquebd, 2022-01-06:
We are still getting Hardware errors and are not getting /camera/imu data. The IMU data is showing on the realsense-viewer.

I have already downgraded the Jetpack version from 4.6 to 4.5 thinking it might resolve the issue. Now I have Jetpack 4.5 and Realsense SDK 2.50.0 and realsense-ros version 2.3.2
I am also getting these Warning message on the realsense-viewer

Resource temporarily unavailable, number: 11
[ WARN] [1641441263.387054933]: Hardware Notification:Depth stream start failure,1.64144e+12,Error,Hardware Error

**nx-bot@expert-nx:~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch initial_reset:=true depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30 unite_imu_method:=copy enable_gyro:=true enable_accel:=true gyro_fps:=250 accel_fps:=200**
... logging to /home/nx-bot/.ros/log/49c871a8-6ea4-11ec-97b6-d8c0a6b11fcf/roslaunch-expert-nx-12624.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://expert-nx:42787/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 200
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
[... 114 lines cut ...]
 05/01 22:54:21,790 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 05/01 22:54:21,841 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 05/01 22:54:21,893 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 05/01 22:54:21,944 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 05/01 22:54:21,996 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 05/01 22:54:22,048 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 05/01 22:54:22,098 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 05/01 22:54:22,149 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 05/01 22:54:22,173 ERROR [546098360704] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
[... cut ...]
 05/01 22:54:22,822 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 05/01 22:54:22,873 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 05/01 22:54:23,025 WARNING [546727518592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: **Resource temporarily unavailable, number: 11
[ WARN] [1641441263.387054933]: Hardware Notification:Depth stream start failure,1.64144e+12,Error,Hardware Error**

MartyG-RealSense, 2022-01-06:
In the other recent case of an IMU problem on Jetson, I posted feedback that Intel provided me at https://github.com/IntelRealSense/realsense-ros/issues/2212#issuecomment-1006579320

haquebd, 2022-01-06:
As I mentioned I already downgraded the Jetpack version from 4.6.b199 to **4.5 now**. So will librealsense 2.50.0 be supportive with Jetpack 4.5 and relasense-roc version 2.3.2?

MartyG-RealSense, 2022-01-06:
JetPack 4.5 should work with librealsense 2.50.0.  Please let me know if the /camera/imu topic is now able to publish data in wrapper 3.2.2.

haquebd, 2022-01-06:
I will update the ros-wrapper to 3.2.2 and run it again.

MartyG-RealSense, 2022-01-07:
Okay, thanks very much @haquebd - please do let me know the outcome of your wrapper update.  Good luck!

haquebd, 2022-01-08:
I have managed to work it with Jetpack 4.5.1 and ros wrapper 2.3.2 @MartyG-RealSense thanks for your feedback.

closed 2022-01-08 by haquebd

MartyG-RealSense, 2022-01-08:
That's excellent news that you found a solution, @haquebd - thanks very much for the update!

AlexKaravaev, 2022-02-22:
@MartyG-RealSense I have following problem, but slightly different setup. I have jetpack 4.6 on host jetson agx and running realsense node in docker. librealsense in docker is compiled with USB flag on. Still IMU messages are not published. What I should do in this case? Applying the patch on the host machine does nothing(which makes sense).

```
[realsense2_camera_node-1] [INFO] [1645528259.641187793] [shadow.cam1]: RealSense ROS v3.2.3
[realsense2_camera_node-1] [INFO] [1645528259.641603810] [shadow.cam1]: Built with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1645528259.641764456] [shadow.cam1]: Running with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1645528260.045681218] [shadow.cam1]: Device with serial number 026322072896 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1645528260.046193143] [shadow.cam1]: Device with physical ID 2-1-14 was found.
[realsense2_camera_node-1] [INFO] [1645528260.046385535] [shadow.cam1]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1645528260.048977513] [shadow.cam1]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1645528260.049368249] [shadow.cam1]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1645528260.069402031] [shadow.cam1]: getParameters...
[realsense2_camera_node-1] [INFO] [1645528260.094866019] [shadow.cam1]: setupDevice...
[realsense2_camera_node-1] [INFO] [1645528260.095073708] [shadow.cam1]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1645528260.095256499] [shadow.cam1]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1645528260.095369144] [shadow.cam1]: Device physical port: 2-1-14
[realsense2_camera_node-1] [INFO] [1645528260.095453435] [shadow.cam1]: Device FW version: 05.13.00.50
[realsense2_camera_node-1] [INFO] [1645528260.095531231] [shadow.cam1]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1645528260.095635715] [shadow.cam1]: Enable PointCloud: Off
[realsense2_camera_node-1] [INFO] [1645528260.095715654] [shadow.cam1]: Align Depth: Off
[realsense2_camera_node-1] [INFO] [1645528260.095790665] [shadow.cam1]: Sync Mode: On
[realsense2_camera_node-1] [INFO] [1645528260.095982961] [shadow.cam1]: Device Sensors: 
[realsense2_camera_node-1] [INFO] [1645528260.232183716] [shadow.cam1]: Stereo Module was found.
[realsense2_camera_node-1] [INFO] [1645528260.273146805] [shadow.cam1]: RGB Camera was found.
[realsense2_camera_node-1] [INFO] [1645528260.274364422] [shadow.cam1]: Motion Module was found.
[realsense2_camera_node-1] [INFO] [1645528260.274597200] [shadow.cam1]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1645528260.274708245] [shadow.cam1]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1645528260.274838490] [shadow.cam1]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1645528260.274968095] [shadow.cam1]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1645528260.275126854] [shadow.cam1]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1645528260.275266635] [shadow.cam1]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1645528260.275408273] [shadow.cam1]: num_filters: 0
[realsense2_camera_node-1] [INFO] [1645528260.275538391] [shadow.cam1]: Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] set_option: stereo_module.emitter_enabled = 0
[realsense2_camera_node-1] [INFO] [1645528264.764069859] [shadow.cam1]: Done Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [INFO] [1645528264.766629996] [shadow.cam1]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[realsense2_camera_node-1] [INFO] [1645528264.769750348] [shadow.cam1]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[realsense2_camera_node-1] [INFO] [1645528264.770052760] [shadow.cam1]: gyro stream is enabled - fps: 400
[realsense2_camera_node-1] [INFO] [1645528264.770544908] [shadow.cam1]: accel stream is enabled - fps: 250
[realsense2_camera_node-1] [INFO] [1645528264.770734260] [shadow.cam1]: setupPublishers...
[realsense2_camera_node-1] [ERROR] [1645528264.818460858] [shadow.cam1]: Failed to load plugin image_transport/compressed_pub, error string: parameter 'format' has already been declared
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1645528264.822980243] [shadow.cam1]: Start publisher IMU
[realsense2_camera_node-1] [INFO] [1645528264.826280378] [shadow.cam1]: setupStreams...
[realsense2_camera_node-1]  22/02 11:11:05,541 WARNING [547129117088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[realsense2_camera_node-1] [INFO] [1645528265.593080564] [shadow.cam1]: SELECTED BASE:Depth, 0
[realsense2_camera_node-1] [INFO] [1645528265.607424256] [shadow.cam1]: Device Serial No: 026322072896
[realsense2_camera_node-1] [INFO] [1645528265.607664522] [shadow.cam1]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1645528265.663310804] [shadow.cam1]: 
[realsense2_camera_node-1]  22/02 11:11:05,665 WARNING [547129117088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[... cut ...] ss, number: 0
[realsense2_camera_node-1] [WARN] [1645528265.765844481] [shadow.cam1]: Hardware Notification:Motion Module failure,1.64553e+12,Error,Hardware Error
[realsense2_camera_node-1]  22/02 11:11:05,767 WARNING [547129117088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[realsense2_camera_node-1]  22/02 11:11:05,938 WARNING [547129117088] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
```

MartyG-RealSense, 2022-02-22:
Hi @AlexKaravaev  I note that your log has a **Motion Module Failure** error.  This error relates to a problem with starting the IMU streams.

Does the error still occur if you add the instruction **initial_reset:=true** to the roslaunch instruction to reset the camera at launch?

AlexKaravaev, 2022-02-22:
@MartyG-RealSense I can not test that, because in docker after initial_reset it can not update again device list in /dev

MartyG-RealSense, 2022-02-22:
Are you able to set gyro to 200 and accel to 63 to see whether the motion module error still occurs?

Does your launch set **unite_imu_method** to linear_interpolation

If unite_imu_method is not defined then the 'imu' topic would not be published, and two separate gyro and accel topics would be published instead.

AlexKaravaev, 2022-02-22:
Still the same. accel and gyro are not published, but imu just appears empty. Attaching full config and full rosout

```
<robot_namespace>:
  cam1:
    ros__parameters:
      device_type: 'd435i'
      serial_no: ''
      usb_port_id: ''

      # color_fps: 6.0
      # color_height: 480
      # color_width: 640

      # depth_fps: 6.0
      # depth_height: 480
      # depth_width: 640

      gyro_fps: 200.0
      accel_fps: 63.0
      # align_depth: true

      clip_distance: 3.0
 
      enable_accel: true
      enable_color: true
      enable_depth: true
      enable_gyro: true
      enable_infra1: false
      enable_infra2: false

      enable_pointcloud: false
      
      filters: ''

      pointcloud_texture_index: 0
      pointcloud_texture_stream: RS2_STREAM_COLOR

      stereo_module:
        emitter_enabled: 0
      unite_imu_method: 'linear_interpolation'
      enable_sync: true
      rosbag_filename: ''
      publish_odom_tf: false
      image_raw/compressed/format: 'jpeg'
      image_raw/compressed/jpeg_level: 1
      #depth/image_rect_raw/compressedDepth/format: "jpeg"
      aligned_depth_to_color/image_raw/compressedDepth/format: "png"
      camera1/aligned_depth_to_color/image_raw/compressedDepth: 0
      aligned_depth_to_color:
        image_raw:
          png_level: 9
```

```
[realsense2_camera_node-1] [INFO] [1645548398.972104971] [shadow.cam1]: RealSense ROS v3.2.3
[realsense2_camera_node-1] [INFO] [1645548398.972587283] [shadow.cam1]: Built with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1645548398.972718165] [shadow.cam1]: Running with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1645548399.269282763] [shadow.cam1]: Device with serial number 026322072896 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1645548399.271695635] [shadow.cam1]: Device with physical ID 2-1-2 was found.
[realsense2_camera_node-1] [INFO] [1645548399.271831157] [shadow.cam1]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1645548399.274944904] [shadow.cam1]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1645548399.275343311] [shadow.cam1]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1645548399.291874079] [shadow.cam1]: getParameters...
[realsense2_camera_node-1] [INFO] [1645548399.323592713] [shadow.cam1]: setupDevice...
[realsense2_camera_node-1] [INFO] [1645548399.323786732] [shadow.cam1]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1645548399.323898446] [shadow.cam1]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1645548399.324039088] [shadow.cam1]: Device physical port: 2-1-2
[realsense2_camera_node-1] [INFO] [1645548399.324156434] [shadow.cam1]: Device FW version: 05.13.00.50
[... 19 lines cut ...]
[realsense2_camera_node-1] [INFO] [1645548403.825567013] [shadow.cam1]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[realsense2_camera_node-1] [INFO] [1645548403.829208191] [shadow.cam1]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[realsense2_camera_node-1] [INFO] [1645548403.829638886] [shadow.cam1]: gyro stream is enabled - fps: 200
[realsense2_camera_node-1] [INFO] [1645548403.829847625] [shadow.cam1]: accel stream is enabled - fps: 63
[realsense2_camera_node-1] [INFO] [1645548403.829958315] [shadow.cam1]: setupPublishers...
[realsense2_camera_node-1] [ERROR] [1645548403.885639843] [shadow.cam1]: Failed to load plugin image_transport/compressed_pub, error string: parameter 'format' has already been declared
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1645548403.888914167] [shadow.cam1]: Start publisher IMU
[realsense2_camera_node-1] [INFO] [1645548403.891911559] [shadow.cam1]: setupStreams...
[realsense2_camera_node-1]  22/02 16:46:44,550 WARNING [547591539104] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1645548404.602547587] [shadow.cam1]: SELECTED BASE:Depth, 0
[realsense2_camera_node-1] [INFO] [1645548404.612009561] [shadow.cam1]: Device Serial No: 026322072896
[realsense2_camera_node-1] [INFO] [1645548404.612192828] [shadow.cam1]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1645548404.672396373] [shadow.cam1]: 
[realsense2_camera_node-1]  22/02 16:46:44,674 WARNING [547591539104] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  22/02 16:46:44,724 WARNING [547591539104] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  22/02 16:46:44,776 WARNING [547591539104] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [WARN] [1645548404.826658974] [shadow.cam1]: Hardware Notification:Motion Module failure,1.64555e+12,Error,Hardware Error
[realsense2_camera_node-1]  22/02 16:46:44,947 WARNING [547591539104] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  22/02 16:46:44,998 WARNING [547591539104] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
[realsense2_camera_node-1]  22/02 16:46:45,303 WARNING [547591539104] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  22/02 16:46:45,455 WARNING [547591539104] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  22/02 16:46:45,506 WARNING [547591539104] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```

AlexKaravaev, 2022-02-22:
This is what I get with 400 and 250 fps set on gyro and accel. No Motion Module failure, but still no data arrived on imu topic

```
[realsense2_camera_node-1] [INFO] [1645548937.209929988] [shadow.cam1]: RealSense ROS v3.2.3
[realsense2_camera_node-1] [INFO] [1645548937.210358571] [shadow.cam1]: Built with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1645548937.210483757] [shadow.cam1]: Running with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1645548937.575424427] [shadow.cam1]: Device with serial number 026322072896 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1645548937.575847283] [shadow.cam1]: Device with physical ID 2-1-2 was found.
[realsense2_camera_node-1] [INFO] [1645548937.575996053] [shadow.cam1]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1645548937.578038872] [shadow.cam1]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1645548937.578322909] [shadow.cam1]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1645548937.593120378] [shadow.cam1]: getParameters...
[realsense2_camera_node-1] [INFO] [1645548937.613863709] [shadow.cam1]: setupDevice...
[realsense2_camera_node-1] [INFO] [1645548937.614028735] [shadow.cam1]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1645548937.614148609] [shadow.cam1]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1645548937.614236867] [shadow.cam1]: Device physical port: 2-1-2
[realsense2_camera_node-1] [INFO] [1645548937.614318404] [shadow.cam1]: Device FW version: 05.13.00.50
[realsense2_camera_node-1] [INFO] [1645548937.614397190] [shadow.cam1]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1645548937.614530696] [shadow.cam1]: Enable PointCloud: Off
[realsense2_camera_node-1] [INFO] [1645548937.614643370] [shadow.cam1]: Align Depth: Off
[realsense2_camera_node-1] [INFO] [1645548937.614726763] [shadow.cam1]: Sync Mode: On
[realsense2_camera_node-1] [INFO] [1645548937.614908494] [shadow.cam1]: Device Sensors: 
[realsense2_camera_node-1] [INFO] [1645548937.691488587] [shadow.cam1]: Stereo Module was found.
[realsense2_camera_node-1] [INFO] [1645548937.728470404] [shadow.cam1]: RGB Camera was found.
[realsense2_camera_node-1] [INFO] [1645548937.729485877] [shadow.cam1]: Motion Module was found.
[realsense2_camera_node-1] [INFO] [1645548937.729693433] [shadow.cam1]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1645548937.729888700] [shadow.cam1]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1645548937.730069151] [shadow.cam1]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1645548937.730167873] [shadow.cam1]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1645548937.730255714] [shadow.cam1]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1645548937.730360324] [shadow.cam1]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1645548937.730452326] [shadow.cam1]: num_filters: 0
[realsense2_camera_node-1] [INFO] [1645548937.730531335] [shadow.cam1]: Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [INFO] [1645548941.845409770] [shadow.cam1]: Done Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [INFO] [1645548941.849426511] [shadow.cam1]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[realsense2_camera_node-1] [INFO] [1645548941.853072205] [shadow.cam1]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[realsense2_camera_node-1] [INFO] [1645548941.853644343] [shadow.cam1]: gyro stream is enabled - fps: 400
[realsense2_camera_node-1] [INFO] [1645548941.853823962] [shadow.cam1]: accel stream is enabled - fps: 250
[realsense2_camera_node-1] [INFO] [1645548941.853910747] [shadow.cam1]: setupPublishers...
[realsense2_camera_node-1] [ERROR] [1645548941.900732700] [shadow.cam1]: Failed to load plugin image_transport/compressed_pub, error string: parameter 'format' has already been declared
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1645548941.905442156] [shadow.cam1]: Start publisher IMU
[realsense2_camera_node-1] [INFO] [1645548941.908279997] [shadow.cam1]: setupStreams...
[realsense2_camera_node-1]  22/02 16:55:42,561 WARNING [547197278624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1645548942.612753319] [shadow.cam1]: SELECTED BASE:Depth, 0
[realsense2_camera_node-1] [INFO] [1645548942.621225239] [shadow.cam1]: Device Serial No: 026322072896
[realsense2_camera_node-1] [INFO] [1645548942.621423451] [shadow.cam1]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1645548942.683518784] [shadow.cam1]: 
[realsense2_camera_node-1]  22/02 16:55:42,685 WARNING [547197278624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  22/02 16:55:42,736 WARNING [547197278624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
[realsense2_camera_node-1]  22/02 16:55:43,527 WARNING [547197278624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  22/02 16:55:43,780 WARNING [547197278624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  22/02 16:55:43,830 WARNING [547197278624] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```

AlexKaravaev, 2022-02-22:
oh yes and important thing is that I am using ros2, latest commit on ros2 branch

AlexKaravaev, 2022-02-22:
And also I have tested on another pc with x86 arch in realsense-viewer imu is displayed correctly with the same camera

MartyG-RealSense, 2022-02-22:
The warning **control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11**, if it is generating continuously in the ROS terminal, would indicate a communications issue where frames are not being reliably received from the camera on your Jetson AGX + Docker setup.   In fact, you said earlier that IMU messages are not being published.

Noting that you have aig_depth set to True in a log above, can you confirm that you still have no IMU messages if align_depth is False.  I know that it is not directly related to the IMU messages, but having align_depth enabled has been known to cause issues on Jetson specifically.

AlexKaravaev, 2022-02-22:
Unfortunately, no( Stil the same

AlexKaravaev, 2022-02-22:
Also output of ```rs-data-collect -c config -t 30```

config
``` 
#Video streams
DEPTH,640,480,30,Z16,0
INFRARED,640,480,30,Y8,1
INFRARED,640,480,30,Y8,2
COLOR,640,480,30,RGB8,0
#Lines starting with non-alpha character (#@!...) will be skipped
# IMU streams will produce data in m/sec^2 & rad/sec
ACCEL,1,1,63,MOTION_XYZ32F
GYRO,1,1,200,MOTION_XYZ32F

```

Output
```
Configuration:
Stream Type,Stream Name,Format,FPS,Width,Height
Infrared 1,Infrared,Y8,30,640,480
Infrared 2,Infrared,Y8,30,640,480
Depth,Depth,Z16,30,640,480
Color,Color,RGB8,30,640,480
Accel,Accel,MOTION_XYZ32F,63
Gyro,Gyro,MOTION_XYZ32F,200

Stream Type,Index,F#,HW Timestamp (ms),Host Timestamp(ms)

Depth,0,0,1645552764980.670,242.018
Depth,0,1,1645552765015.549,274.926
Depth,0,0,1645552765440.052,674.509
Depth,0,1,1645552765475.689,708.155

Stream Type,Index,F#,HW Timestamp (ms),Host Timestamp(ms)

Color,0,0,1645552764911.586,180.973
Color,0,1,1645552764946.055,210.813
Color,0,2,1645552764980.730,243.380
Color,0,3,1645552765015.610,276.196
Color,0,4,1645552765050.694,310.469
Color,0,5,1645552765085.646,342.558
Color,0,6,1645552765120.745,376.007
Color,0,7,1645552765155.990,409.406
Color,0,8,1645552765191.382,442.433
Color,0,9,1645552765226.878,475.831
Color,0,10,1645552765262.511,509.096
Color,0,11,1645552765298.279,542.630
Color,0,12,1645552765333.631,575.908
Color,0,13,1645552765369.054,609.180
Color,0,14,1645552765404.548,642.670
Color,0,15,1645552765440.114,678.158
Color,0,16,1645552765475.751,710.215
Color,0,17,1645552765511.459,742.752
Color,0,18,1645552765547.239,775.957
Color,0,19,1645552765583.090,809.441
Color,0,20,1645552765618.959,842.797
Color,0,21,1645552765654.894,876.187
Color,0,22,1645552765690.893,909.483
Color,0,23,1645552765726.216,942.972
Color,0,24,1645552765761.544,976.242
Color,0,25,1645552765796.878,1009.557
Color,0,26,1645552765831.819,1042.943
Color,0,27,1645552765866.736,1076.267
Color,0,28,1645552765901.629,1109.878
Color,0,29,1645552765936.241,1143.075
Color,0,30,1645552765970.810,1176.407
Color,0,31,1645552766005.338,1209.805
Color,0,32,1645552766039.668,1243.049
Color,0,33,1645552766073.947,1276.202
[... 855 lines cut ...]
Color,0,889,1645552794584.905,29830.731
Color,0,890,1645552794618.264,29864.079
Color,0,891,1645552794651.624,29897.475
Color,0,892,1645552794684.984,29930.945
Color,0,893,1645552794718.344,29964.364
Color,0,894,1645552794751.704,29997.523
Color,0,895,1645552794785.063,30031.038
Color,0,896,1645552794818.423,30064.404
Color,0,897,1645552794851.783,30097.664
Color,0,898,1645552794885.143,30130.918

Stream Type,Index,F#,HW Timestamp (ms),Host Timestamp(ms)

Infrared,1,0,1645552764666.219,15.175
Infrared,1,2,1645552764767.240,36.055
Infrared,1,3,1645552764801.016,67.298
Infrared,1,17,1645552766108.113,1307.291

Stream Type,Index,F#,HW Timestamp (ms),Host Timestamp(ms)

Infrared,2,0,1645552764666.219,15.233
Infrared,2,2,1645552764767.240,36.111
Infrared,2,3,1645552764801.016,67.388
Infrared,2,17,1645552766108.113,1307.324
```

MartyG-RealSense, 2022-02-22:
@AlexKaravaev  Which installation instructions did you use to install the librealsense SDK and the RealSense ROS wrapper on Jetson, please?

Have you also tested without Docker to determine whether Docker may be a factor in the lack of IMU data?

AlexKaravaev, 2022-02-22:
@MartyG-RealSense This is the command:
```
cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=true -DBUILD_WITH_CUDA=false -DBUILD_GRAPHICAL_EXAMPLES=false -DBUILD_PYTHON_BINDINGS=true && make -j$(($(nproc)-0)) && sudo make install 
```

The wrapper is built just with usual ```colcon build --symlink-install```.

Though if rs-data-collect report no imu too, I guess it's issue with librealsense, not the wrapper.

I haven't tried on host, but I guess it's possible that it will be working because it's installable with apt-get . In docker it is not, because distrubution is focal I guess? 

But I don't see problem why it should not work with a source build

MartyG-RealSense, 2022-02-22:
Because you are using a Jetson that has an Nvidia graphics GPU, DBUILD_WITH_CUDA can be set to true.  The CUDA support in librealsense offloads processing work from the CPU to the GPU when performing color conversion, depth-color alignment and point cloud operations.  As you are using align_depth = true, enabling CUDA support in the librealsense build could considerably reduce CPU percentage utilization.

AlexKaravaev, 2022-02-22:
@MartyG-RealSense It's true, thank you for the tip! I had some problems with cuda in docker, but I will do that off course in the future. But yes, unfortunetely problem with imu lies somewhere else

MartyG-RealSense, 2022-02-23:
In your opening message in this case, you describe using **rostopic as /camera/imu**.  Are you able to echo the topic to check if messages are being published?

**rostopic echo /camera/imu**

AlexKaravaev, 2022-02-24:
It wasn't showing anything, as well as ros2 hz

MartyG-RealSense, 2022-02-24:
@AlexKaravaev  I researched your problem from the beginning in search of fresh insights.  I recalled reading that Nvidia's Jetson **JetPack** was compatible with Ubuntu 16.04 and 18.04 but not designed for 20.04 (Focal), which you are using in your Docker system.  Nvidia's official JetPack installation instructions seemed to confirm my recollection of JetPack support only being for 16.04 and 18.04.

https://docs.nvidia.com/jetson/jetpack/install-jetpack/index.html#jetpack-debian-packages

I did observe in my research though that some progress was being made in developing workarounds for 20.04 JetPack compatibility, such as the JetsonHacks guide in the link below.

https://www.jetsonhacks.com/2021/08/25/native-boot-for-jetson-xaviers/

AlexKaravaev, 2022-02-24:
@MartyG-RealSense But I am using JetPack on 18.04. 18.04 is installed on my host machine, in my docker it's 20.04(focal),which should not cause a problem. Also given that imu was working in docker previously(though I last tested it very-very long time ago, so I can not  tell exactly which update causes this error with imu not published is happened)

MartyG-RealSense, 2022-02-25:
In librealsense 2.50.0 Intel released a tutorial guide for **Librealsense Docker**.  The guide states that Librealsense Docker does not currently support use of IMU devices.

https://github.com/IntelRealSense/librealsense/tree/master/scripts/Docker

Is your own Docker system using Librealsense Docker?

blue-ring-octopus, 2022-06-11:
I have the exact same issue in jetson xavier, the imu is working in realsense viewer, but nothing is publishing in the rostopic, not even the individual message from gyro and accel. I've checked my realsense2-camera version, librealsense and jetpack version are exactly the same as the one that op managed to get working. Any more idea on what do i need to do

MartyG-RealSense, 2022-06-11:
Hi @blue-ring-octopus  What is the roslaunch instruction that you are using, please?  The creator of this case used the following launch instruction:

`roslaunch realsense2_camera rs_camera.launch initial_reset:=true align_depth:=true filters:=pointcloud depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30 unite_imu_method:=copy enable_gyro:=true enable_accel:=true gyro_fps:=250 accel_fps:=200`

blue-ring-octopus, 2022-06-11:
> 
i used both 
`roslaunch realsense2_camera rs_camera.launch align_depth:=true  unite_imu_method:=copy enable_gyro:=true enable_accel:=true 
``roslaunch realsense2_camera rs_camera.launch align_depth:=true enable_gyro:=true enable_accel:=true`
neither published anything to the imu topic or the gyro/accel topic, i tried the initial reset and setting align_depth to false too as suggested but it was not helping, I left the other launch params to default

MartyG-RealSense, 2022-06-11:
Does it publish any depth or color topic data if you use the most basic launch instruction, **roslaunch realsense2_camera rs_camera.launch**

blue-ring-octopus, 2022-06-11:
> Does it publish any depth or color topic data if you use the most basic launch instruction, **roslaunch realsense2_camera rs_camera.launch**

yes, it's working perfectly

MartyG-RealSense, 2022-06-12:
If the camera works normally on your Windows computer then we can conclude that the camera hardware, firmware driver and USB cable are likely fine and that it is a problem related to Linux (such as the kernel) or to the Linux version of the librealsense SDK on the particular computer that you are using.

What method do you use to install librealsense on your Jetson Xavier, please?  For example, if you were using **Method 1** of the ROS wrapper instructions to install librealsense and the wrapper together at the same time from packages then that method would not be suitable for Jetson's Arm architecture and it would be better to install librealsense and the wrapper separately (librealsense first and then the wrapper secondly after that).

blue-ring-octopus, 2022-06-12:
> If the camera works normally on your Windows computer then we can conclude that the camera hardware, firmware driver and USB cable are likely fine and that it is a problem related to Linux (such as [...]

i was using method 1. i purged it and switched to RSUSB method and installed realsense-camera from source this morning. It worked the first time I launched it, then the same problem is happening again after i relaunched everything

MartyG-RealSense, 2022-06-12:
If you are using RSUSB then it is likely not a kernel related problem, as an RSUSB build bypasses the kernel.  

Does the IMU behave correctly again if you do a complete shutdown and reboot of the Jetson when it stops working after the first launch?

blue-ring-octopus, 2022-06-12:
> If you are using RSUSB then it is likely not a kernel related problem, as an RSUSB build bypasses the kernel.
> 
> Does the IMU behave correctly again if you do a complete shutdown and reboot of the Jetson when it stops working after the first launch?

no, it worked exactly once after the first launch, and neither rebooting or resetting can get it to work again

MartyG-RealSense, 2022-06-12:
Are you able to reload the **udev** device handling rules by inputting the commands below into the Ubuntu terminal?

**udevadm control --reload-rules
udevadm trigger**

blue-ring-octopus, 2022-06-12:
> Are you able to reload the **udev** device handling rules by inputting the commands below into the Ubuntu terminal?
> 
> **udevadm control --reload-rules udevadm trigger**
it didn't help, but now you mentioned it i remember when it was working there was a warning like "more then one udev rule found" , i thought something was left over from method 1 so i re-purged the apt-get version and relaunch and it stopped working since

MartyG-RealSense, 2022-06-12:
A purge will not clear that particular problem (multiple udev rules detected) but it has a simple solution.  If librealsense is installed with both packages and source code on the same computer then you can end up with two different udev rules that conflict, **etc/udev/rules.d/99-realsense-libusb.rules** and **/lib/udev/rules.d/60-librealsense2-udev-rules.rules**

The rule at the **etc** location is the one that should be deleted.  The multiple udev rules warning and problems associated with it should then automatically disappear.

blue-ring-octopus, 2022-06-12:
> A purge will not clear that particular problem (multiple udev rules detected) but it has a simple solution. If librealsense is installed with both packages and source code on the same computer then [...]

just did, and ran the two line for reloading too, still not fixing the imu problem,

MartyG-RealSense, 2022-06-12:
Having eliminated a camera hardware / firmware problem or kernel problem as possibilities and tried most other solutions, the remaining unpleasant option would be a complete wipe and reinstall of the Jetson (including the Ubuntu OS) to remove any hidden problems that might be affecting the IMU.  When everything else has been tried, a total wipe and reinstall does sometimes correct a problem, even if you never find out what the cause of the problem actually was.

blue-ring-octopus, 2022-06-12:
> Having eliminated a camera hardware / firmware problem or kernel problem as possibilities and tried most other solutions, the remaining unpleasant option would be a complete wipe and reinstall of [...]

very unpleasant indeed, many things were setup by my teammates long time ago so it could be a lot of work to retrace every step to get it back to the current state. I will think about it and try to hunt for more ways to check what's wrong with it, thanks for your help

tugbakara, 2022-08-01:
Hi @blue-ring-octopus we have the exact same issue, we reinstall Jetson , but when we run ros examples we caouldn't get IMU datas, did you solve ur problem? If it is can you write down here one by one.