realsenseai/realsense-ros #2214: RGB color distortion (D435i, python, opencv)

LightwishWONG, 2022-01-08:
Recently I have been using realsense D435i camera in ROS to capture RGB images. I notice the RGB images received by D435i are generally more brighter and blurry than original ones, which leads to noticeable RGB distortion. RGBs of some pixels may even severely changed. But if I use my phone camera, this problem disappears. Below are codes, transmitted RGB images and received RGB images.

Codes:
`import rospy 
import message_filters 
import cv2, cv_bridge   #cv_bridge: bridge between ROS image and OpenCV image
import numpy as np
from sensor_msgs.msg import Image   # ROS image message format

def rxCallback(RGB, depth):
    bridge = cv_bridge.CvBridge() 
    RGB_frame = bridge.imgmsg_to_cv2(RGB, desired_encoding='bgr8')  # ROS image -> OpenCV image
    cv2.imshow('RGB_frame', RGB_frame)
    cv2.waitKey(1)

def rx_ros(): 
    # read image
    im_sub = message_filters.Subscriber('/camera/color/image_raw', Image)
    dep_sub = message_filters.Subscriber('/camera/depth/image_rect_raw', Image)

    timeSynchronizer = message_filters.ApproximateTimeSynchronizer([im_sub, dep_sub], 10, 0.5)
    timeSynchronizer.registerCallback(rxCallback)

if __name__ == '__main__': 
    rospy.init_node("rx_ros")    # create a node
    rx_ros()
    rospy.spin()`

Transmitted image:
[image]

Received image:
[image]

Should I change some configurations of D435i, like white-balancing or auto-explosure? Or how to do image processing for this problem? Could anyone help me solve it? Thanks!

MartyG-RealSense, 2022-01-08:
Hi @LightwishWONG  That kind of change in color tint makes me think of white balance too.  Increasing the white balance makes the tinting become progressively more blue, whilst increasing it makes the colors brighter and warmer.

By default the white balance value should be **4600**.  You can check it by launching the ROS wrapper and then opening the **dynamic reconfigure** interface after launch to check the camera values during runtime.  The white balance should be under the **RGB** section of the dynamic reconfigure interface, as described in the wrapper's documentation about it in the link below.

https://github.com/IntelRealSense/realsense-ros#set-camera-controls-using-dynamic-reconfigure-params

MartyG-RealSense, 2022-01-15:
Hi @LightwishWONG  Do you require further assistance with this case, please?  Thanks!

LightwishWONG, 2022-01-16:
@MartyG-RealSense Sorry for my late reply. I don't require further assistance and I think this issue can be closed. Thank you for your kind attention!

closed 2022-01-16 by LightwishWONG